espressif/esp-idf #8146: Not able to communicate HCI messages through UART1

99002473, 2021-12-27:
After flasing my ESP32 with the controller_hci_uart_esp32 code, I am getting the desired logs as mentioned in README, but after that I'm not able to communicate the HCI messages through UART1

BetterJincheng, 2021-12-28:
@99002473 Please attach the whole logs here to help debug.

99002473, 2021-12-28:
Log from UART1

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7588
load:0x40078000,len:14912
ho 0 tail 12 room 4
load:0x40080400,len:3688
entry 0x4008067c
I (28) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader
I (28) boot: compile time 17:51:09
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0b81ch ( 47132) map
I (128) esp_image: segment 1: paddr=0001b844 vaddr=3ffbdb60 size=03b5ch ( 15196) load
I (134) esp_image: segment 2: paddr=0001f3a8 vaddr=40080000 size=00c70h (  3184) load
I (136) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=356fch (218876) map
I (222) esp_image: segment 4: paddr=00055724 vaddr=40080c70 size=1649ch ( 91292) load
I (259) esp_image: segment 5: paddr=0006bbc8 vaddr=50000000 size=00010h (    16) load
I (271) boot: Loaded app from partition at offset 0x10000
I (271) boot: Disabling RNG early entropy source...
I (283) cpu_start: Pro cpu up.
I (283) cpu_start: Starting app cpu, entry point is 0x40081210
I (0) cpu_start: App cpu up.
I (297) cpu_start: Pro cpu start user code
I (297) cpu_start: cpu freq: 160000000
I (297) cpu_start: Application information:
I (301) cpu_start: Project name:     controller_hci_uart
I (307) cpu_start: App version:      1
I (312) cpu_start: Compile time:     Dec 27 2021 15:42:14
I (318) cpu_start: ELF file SHA256:  5f352a50dab60d78...
I (324) cpu_start: ESP-IDF:          v4.3.2-dirty
I (330) heap_init: Initializing. RAM available for dynamic allocation:
I (337) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (343) heap_init: At 3FFB7CD8 len 00000328 (0 KiB): DRAM
I (349) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (355) heap_init: At 3FFC2E68 len 0001D198 (116 KiB): DRAM
I (361) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (367) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (374) heap_init: At 4009710C len 00008EF4 (35 KiB): IRAM
I (381) spi_flash: detected chip: generic
I (385) spi_flash: flash io: dio
W (389) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (403) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (431) CONTROLLER_UART_HCI: HCI UART1 Pin select: TX 5, RX 18, CTS 23, RTS 19
I (431) BTDM_INIT: BT controller compile version [6a07b06]
I (431) system_api: Base MAC address is not set
I (441) system_api: read default base MAC address from EFUSE
I (451) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07

#(After this LOG I'm not able to communicate any HCI messages with the board, According to the README file, I should be able to communicate with the board after the last log file)

99002473, 2021-12-28:
Following is the LOG from IDF-monitor

Executing action: monitor
Running idf_monitor in directory c:\users\amat2\eclipse-workspace-1\controller_hci_uart_esp32
Executing "C:\Users\AMAT2\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe C:\Users\AMAT2\Desktop\esp-idf-v4.3.2\esp-idf-v4.3.2\tools/idf_monitor.py -p \\.\COM4 -b 115200 --toolchain-prefix xtensa-esp32-elf- --print_filter  c:\users\amat2\eclipse-workspace-1\controller_hci_uart_esp32\build\controller_hci_uart.elf -m 'C:\Users\AMAT2\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe' 'C:\Users\AMAT2\Desktop\esp-idf-v4.3.2\esp-idf-v4.3.2\tools\idf.py'"...
--- idf_monitor on \\.\COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7588
load:0x40078000,len:14912
ho 0 tail 12 room 4
load:0x40080400,len:3688
0x40080400: _init at ??:?

entry 0x4008067c
I (28) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader
I (28) boot: compile time 17:51:09
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0b81ch ( 47132) map
I (128) esp_image: segment 1: paddr=0001b844 vaddr=3ffbdb60 size=03b5ch ( 15196) load
I (134) esp_image: segment 2: paddr=0001f3a8 vaddr=40080000 size=00c70h (  3184) load
I (136) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=356fch (218876) map
I (222) esp_image: segment 4: paddr=00055724 vaddr=40080c70 size=1649ch ( 91292) load
I (259) esp_image: segment 5: paddr=0006bbc8 vaddr=50000000 size=00010h (    16) load
I (271) boot: Loaded app from partition at offset 0x10000
I (271) boot: Disabling RNG early entropy source...
I (283) cpu_start: Pro cpu up.
I (283) cpu_start: Starting app cpu, entry point is 0x40081210
0x40081210: call_start_cpu1 at C:/Users/AMAT2/Desktop/esp-idf-v4.3.2/esp-idf-v4.3.2/components/esp_system/port/cpu_start.c:150

I (0) cpu_start: App cpu up.
I (297) cpu_start: Pro cpu start user code
I (297) cpu_start: cpu freq: 160000000
I (297) cpu_start: Application information:
I (301) cpu_start: Project name:     controller_hci_uart
I (307) cpu_start: App version:      1
I (312) cpu_start: Compile time:     Dec 27 2021 15:42:14
I (318) cpu_start: ELF file SHA256:  5f352a50dab60d78...
I (324) cpu_start: ESP-IDF:          v4.3.2-dirty
I (330) heap_init: Initializing. RAM available for dynamic allocation:
I (337) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (343) heap_init: At 3FFB7CD8 len 00000328 (0 KiB): DRAM
I (349) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (355) heap_init: At 3FFC2E68 len 0001D198 (116 KiB): DRAM
I (361) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (367) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (374) heap_init: At 4009710C len 00008EF4 (35 KiB): IRAM
I (381) spi_flash: detected chip: generic
I (385) spi_flash: flash io: dio
W (389) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (403) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (431) CONTROLLER_UART_HCI: HCI UART1 Pin select: TX 5, RX 18, CTS 23, RTS 19
I (431) BTDM_INIT: BT controller compile version [6a07b06]
I (431) system_api: Base MAC address is not set
I (441) system_api: read default base MAC address from EFUSE
I (451) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07

BetterJincheng, 2021-12-28:
@99002473 Please note the README.md at Hardware Required.

> This example should be able to run on any commonly available ESP32 development board. To connect UART to PC, another board such as ESP_Test Board or FT232 USB UART board is usually needed.

99002473, 2021-12-28:
I am using the following configuration
One PC is connected to ESP32 developmment board through USRT0 Micro usb cable.
The other PC is connected to UART1 using the "RS232 to TTL SP3232 UART Transceiver Communication Serial Module" 
Link of the product [https://www.waveshare.com/rs232-board.htm]

99002473, 2021-12-28:
> @99002473 Please note the README.md at Hardware Required.
> 
> > This example should be able to run on any commonly available ESP32 development board. To connect UART to PC, another board such as ESP_Test Board or FT232 USB UART board is usually needed.

Can I get the Link for steps to prepare the ESP_Test Board for the following project. We will try this as a parallel solution in case our method doesn't work

BetterJincheng, 2021-12-28:
@99002473  In my environment, I use an FT232 USB UART board.

```
FT232 ------ ESP32

GND          GND
TXD          18
RXD          5
RTS          23
CTS          19
```

Plug FT232 into the computer (linux) with BT-HOST. UART can not be recognized as a Bluetooth device. I use `hciattach` to attach the UART to the Bluetooth stack as HCI transport interface. Then it works.

99002473, 2021-12-29:
@BetterJincheng Thanks for the solution, 
I have got FT232 USB UART board from lab, connected FT232---ESP232 with the mentioned pinouts and plugged the board to my Linux computer.
Used the following  hciattach command 

"sudo hciattach -n -p /dev/ttyUSB0 any 115200 CRTSCTS"

Hope this command configuration is correct? (baud rate, flow control, vendor ID).
If not can we get the correct hciattach command that you have used which works.

BetterJincheng, 2021-12-29:
@99002473 I locally used `hciattach -s 921600 /dev/ttyUSB0 any 921600 flow nosleep`. And it works.

99002473, 2021-12-29:
@BetterJincheng I used the command just now `hciattach -s 921600 /dev/ttyUSB0 any 921600 flow nosleep` . It shows the message `Device Setup Completed` . But not able to enter the HCI commands there. Do I need to use minicom after that?

BetterJincheng, 2021-12-29:
1. `hciattach -s 921600 /dev/ttyUSB0 any 921600 flow nosleep`
2. `hcitool dev
And you will see
```
Devices:
	hci0	XX:XX:XX:XX:XX:XX
```
3. `sudo hciconfig hci0 up`

99002473, 2021-12-29:
@BetterJincheng 

```
user@BLTSP06365:~$ sudo hciattach -s 921600 /dev/ttyUSB2 any 921600 flow nosleep
Device setup complete
user@BLTSP06365:~$ hcitool dev
Devices:
user@BLTSP06365:~$ sudo hciconfig hci0 up
```

- dmesg logs
```
[434552.454219] usb 1-1: new full-speed USB device number 55 using xhci_hcd
[434552.633719] usb 1-1: New USB device found, idVendor=10c4, idProduct=ea60
[434552.633725] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[434552.633728] usb 1-1: Product: CP2102 USB to UART Bridge Controller
[434552.633732] usb 1-1: Manufacturer: Silicon Labs
[434552.633734] usb 1-1: SerialNumber: 0001
[434552.642529] cp210x 1-1:1.0: cp210x converter detected
[434552.644597] usb 1-1: cp210x converter now attached to ttyUSB2
[434568.782359] Bluetooth: hci0: command 0x1003 tx timeout
[434570.798230] Bluetooth: hci0: command 0x1001 tx timeout
[434572.814353] Bluetooth: hci0: command 0x1009 tx timeout
[434578.958214] Bluetooth: hci0: command 0x1003 tx timeout
[434580.974225] Bluetooth: hci0: command 0x1001 tx timeout
[434582.990167] Bluetooth: hci0: command 0x1009 tx timeout
[434589.198344] Bluetooth: hci0: command tx timeout
```

for me in Devices , it is showing empty

BetterJincheng, 2021-12-29:
@99002473 Try this:
1. Unplug the USB-UART board
2. plug the USB-UART board
3. `ls /dev/ttyUSB*`
4. `sudo chmod 777 /dev/ttyUSB0` (at your condition)
5. `hciattach -s 921600 /dev/ttyUSB0 any 921600 flow nosleep`
6. `hcitool dev`
7. `sudo hciconfig hci0 up`

BetterJincheng, 2021-12-29:
@99002473 Then, Host communicates with Controller through HCI. 
e.g. Read BD_ADDR command
```
➜ ~ sudo hcitool -i hci0 cmd 0x04 0x0009
< HCI Command: ogf 0x04, ocf 0x0009, plen 0
> HCI Event: 0x0e plen 10
  05 09 10 00 XX XX XX XX XX XX
```

99002473, 2021-12-29:
> @BetterJincheng
> 
> ```
> user@BLTSP06365:~$ sudo hciattach -s 921600 /dev/ttyUSB2 any 921600 flow nosleep
> Device setup complete
> user@BLTSP06365:~$ hcitool dev
> Devices:
> user@BLTSP06365:~$ sudo hciconfig hci0 up
> ```
> 
> * dmesg logs
> 
> ```
> [434552.454219] usb 1-1: new full-speed USB device number 55 using xhci_hcd
> [434552.633719] usb 1-1: New USB device found, idVendor=10c4, idProduct=ea60
> [434552.633725] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [434552.633728] usb 1-1: Product: CP2102 USB to UART Bridge Controller
> [434552.633732] usb 1-1: Manufacturer: Silicon Labs
> [434552.633734] usb 1-1: SerialNumber: 0001
> [434552.642529] cp210x 1-1:1.0: cp210x converter detected
> [434552.644597] usb 1-1: cp210x converter now attached to ttyUSB2
> [434568.782359] Bluetooth: hci0: command 0x1003 tx timeout
> [434570.798230] Bluetooth: hci0: command 0x1001 tx timeout
> [434572.814353] Bluetooth: hci0: command 0x1009 tx timeout
> [434578.958214] Bluetooth: hci0: command 0x1003 tx timeout
> [434580.974225] Bluetooth: hci0: command 0x1001 tx timeout
> [434582.990167] Bluetooth: hci0: command 0x1009 tx timeout
> [434589.198344] Bluetooth: hci0: command tx timeout
> ```
> 
> for me in Devices , it is showing empty. That's why I guess `hciconfig hci0 up` is timing out.

 @BetterJincheng 
Thanks for the help 
I am getting the message now 

```
Devices:
             hci0 30:AE:A4:98:50:72
```
after reflashing the code onto ESP32.

Will try the basic HCI commands now and let you know if any issue occurs.
As per my understanding basic HCI commands would work with this (e.g.- Reset/ADV_PARAM/ADV_DATA/ADV_ENABLE ) ?

BetterJincheng, 2021-12-29:
@99002473

If it works, please help closed the issue. Thanks!

Turn to ESP any time

99002473, 2021-12-29:
> @99002473 Then, Host communicates with Controller through HCI. e.g. Read BD_ADDR command
> 
> ```
> ➜ ~ sudo hcitool -i hci0 cmd 0x04 0x0009
> < HCI Command: ogf 0x04, ocf 0x0009, plen 0
> > HCI Event: 0x0e plen 10
>   05 09 10 00 XX XX XX XX XX XX
> ```

@BetterJincheng Ohkay, Thanks v. much. 

I am able to enter these commands now and getting similar messages.

To test this BT-HOST, I'm using the following android application [https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en_IN&gl=US](url) 
Hope this is a good one. 
Thanks for the continuous support.

99002473, 2021-12-29:
> @99002473 Then, Host communicates with Controller through HCI. e.g. Read BD_ADDR command
> 
> ```
> ➜ ~ sudo hcitool -i hci0 cmd 0x04 0x0009
> < HCI Command: ogf 0x04, ocf 0x0009, plen 0
> > HCI Event: 0x0e plen 10
>   05 09 10 00 XX XX XX XX XX XX
> ```

Hi @BetterJincheng 
Can we get the OGF and OCF codes of the ESP32 development board for modes like scanning, advertising etc. (the same which you gave for "Read BD_ADDR command")
We wanted to use CBT Bluetooth tester for RF analyzer
It would be very much helpful for us.
Thanks

BetterJincheng, 2021-12-29:
@99002473 ref to Bluetooth Core Spec

99002473, 2021-12-29:
@BetterJincheng One more query was there. Can we change the baud rate to 115200 ? If we change directl in sysconfig.defaults, will it reflect?

BetterJincheng, 2021-12-29:
@99002473 Just have a try. Usually it works.

99002473, 2021-12-29:
Ohkay thankyou v much @BetterJincheng for the support. Will follow the inputs.

closed 2021-12-29 by 99002473