espressif/esp-idf #8147: ESP32C3 advertsing issue with iPhone bluetooth scanning.

jigneshpatel47, 2021-12-27:
Hi,

We are using ESP32C3 with ESP-IDF v4.3.2. 
I have built the BLE server example using [gatt server example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server) for esp32c3.

My issue is that ESP32C3 is not showing in the Bluetooth scanning list on iPhone. 
I am using iPhone12 with software version 15.0.

The same thing, It works in android. I can able show esp32c3 in the android mobile Bluetooth scanning list.

What may be the issue ESP32c3 and iPhone are facing for advertising?

Thank you in advance.
Jignesh Patel

xiewenxiang, 2021-12-29:
I have retested the problem you mentioned on iPhone 12, and the name can be displayed normally.

xiewenxiang, 2021-12-29:
[image]
[image]

jigneshpatel47, 2021-12-29:
Thank you @xiewenxiang  for your reply,

I can able to see ESP32C3 in the NRF BLE Application, But I can not discover in Bluetooth scanning that comes under Setting -> Bluetooth in iPhone.

What is the method is used by Bluetooth native and NRF scanning mechanisms?

Best Regards,
Jignesh Patel

xiewenxiang, 2021-12-30:
From the perspective of Bluetooth, the device name has been included in the broadcast, but why not in the system Settings of the mobile phone? You may need to consult Apple.

Alvin1Zhang, 2022-10-27:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2022-10-27 by Alvin1Zhang

skv6631, 2023-07-11:
Got any resolution for this issue?? @jigneshpatel47  @xiewenxiang