espressif/esp-idf #8149: CMake Issue in esp_build_process (Fix in Problem Description)

AndruePeters, 2021-12-27:
## Environment

- IDF version (run ``git describe --tags`` to find it):
    // v5.0-dev-810-gb886dc6998
- Build System:         [CMake, idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux|]
- Using an IDE?: [No]

## Problem Description
There are two issues, the first I have solved. The second I am still figuring out.
Look at github.com/AndruePeters/pt_controller with the branch cmake_issue

The issue is from ${IDF_PATH}/esp-idf/tools/cmake/build.cmake
`build_dir` is never set. On line 429, add `idf_build_get_property(build_dir BUILD_DIR`
I'm not getting another issue with it failing, but at least the build_dir is being set properly.

### Expected Behavior
The tool should work.

### Actual Behavior
The tool does not work because build_dir is not being set.

### Steps to reproduce

1. Run idf.py build

AndruePeters, 2021-12-27:
Looks like for some reason `local_components_list_file` is not being written. I'm looking into it now.

zikalino, 2021-12-29:
@AndruePeters thank you for reporting! We will wait for your updates

AndruePeters, 2021-12-29:
Hey. I'll look into this more over the nest couple days. Got a few other build environment things worked out. 

I've pieced together a lot of information from diving into the build scripts, but better and easier to find documentation on the CMake build system (without using idf.py) would be great. But like I said, I fixed some other issues and will look into this more in the next few days.

ESP-Marius, 2022-08-31:
Sorry for the delay @AndruePeters, do you end up figuring this out on your own?

I see that you found the example script which shows how to build (`build-esp32.sh`) so that's good.

I managed to build your project using that one plus some minor changes:
-  Removed `COMPONENTS` from `idf_build_process()`, this ensures all IDF components are built instead of only the ones listed (you could also solve this by listing all the components you need of course)
- Add any extra libraries that need to be linked in `target_link_libraries()`, e.g. `idf::esp_wifi`.

closed 2022-09-08 by ESP-Marius