espressif/esp-idf #8150: uart_isr_register doesnt match uart_isr_free steps

frini37, 2021-12-27:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v4.3
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [Yes] VS Code
- Power Supply:         [USB]

## Problem Description
When working with custom uart isr function and there is a requirement to free the isr call at some stage,
uart_isr_register doesn't match uart_isr_free steps, causing error when attempting to free the custom isr.

The functionality of uart_isr_register doesn't do:
p_uart_obj[uart_num]->intr_handle=*handle;

While uart_isr_free does the following:
UART_CHECK((p_uart_obj[uart_num]->intr_handle != NULL), "uart driver error", ESP_ERR_INVALID_ARG);
p_uart_obj[uart_num]->intr_handle=NULL;

### Expected Behavior
I would expect the uart_isr_register to look as follows:
esp_err_t uart_isr_register(uart_port_t uart_num, void (*fn)(void*), void * arg, int intr_alloc_flags,  uart_isr_handle_t *handle)
{
    esp_err_t ret;
    UART_CHECK((uart_num < UART_NUM_MAX), "uart_num error", ESP_FAIL);
    UART_CHECK((p_uart_obj[uart_num]->intr_handle == NULL), "uart driver error", ESP_FAIL);
    UART_CHECK((handle != NULL), "uart driver error", ESP_ERR_INVALID_ARG);
    UART_ENTER_CRITICAL(&(uart_context[uart_num].spinlock));
    ret=esp_intr_alloc(uart_periph_signal[uart_num].irq, intr_alloc_flags, fn, arg, handle);
    p_uart_obj[uart_num]->intr_handle=*handle;
    UART_EXIT_CRITICAL(&(uart_context[uart_num].spinlock));
    return ret;
}
### Actual Behavior

### Steps to reproduce

1. install uart
2. free isr
3. register a custom new isr func
4. free the custom isr will lead to an error

Thank for all your wonderful work!

o-marshmallow, 2021-12-29:
Hi @frini37 ,
Thank you for reporting this, I was able to reproduce the issue, I am going to post a fix for it

AxelLin, 2022-02-19:
It's strange this fix is in v4.4 but not in master.
BTW, v4.3 also needs the fix.

o-marshmallow, 2022-02-21:
@AxelLin This fix is not in master because this feature will be dropped in the next version.
The patch is on the way for v4.3

Alvin1Zhang, 2022-03-17:
Thanks for reporting, fix on release/4.3 is available, feel free to reopen if the issue still happens.

closed 2022-03-17 by Alvin1Zhang