espressif/esp-idf #8151: coredump broken?

chegewara, 2021-12-28:
I am following simplest possible example on the world:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/core_dump.html

Logs are looking good, core dump partition is found, crash info seems to be stored on it with 1 exception. There is no info about `global_var` dumped at all:
```
(gdb) p global_var
$1 = 0 '\000'
```
With `espcoredump.py -p /dev/ttyUSB1 info_corefile ./build/core-dump.elf` i see nice logs printed, and when i am trying to search for any info about `global_var` i find nothing.

Helpful logs:
```
I (256) cpu_start: ESP-IDF:          v5.0-dev-869-g3bd5ae6b78-dirty
...
assert failed: app_main core-dump.c:12 (0)

Backtrace:0x40081a72:0x3ffb79a00x40085841:0x3ffb79c0 0x4008a9b9:0x3ffb79e0 0x400d665a:0x3ffb7b00 0x400e63f8:0x3ffb7b20 0x40088331:0x3ffb7b40 
0x40081a72: panic_abort at /home/chegewara/esp/master/components/esp_system/panic.c:403

0x40085841: esp_system_abort at /home/chegewara/esp/master/components/esp_system/esp_system.c:130

0x4008a9b9: __assert_func at /home/chegewara/esp/master/components/newlib/assert.c:85

0x400d665a: app_main at /home/chegewara/upwork/components/tests/core-dump/build/../main/core-dump.c:12

0x400e63f8: main_task at /home/chegewara/esp/master/components/freertos/FreeRTOS-Kernel/portable/port_common.c:127 (discriminator 2)

0x40088331: vPortTaskWrapper at /home/chegewara/esp/master/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:132

ELF file SHA256: 7b07d4e440407ddb

I (385) esp_core_dump_flash: Save core dump to flash...
I (391) esp_core_dump_flash: Erase flash 12288 bytes @ 0x110000
I (560) esp_core_dump_flash: Write end offset 0x2964, check sum length 4
I (560) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP�ets Jul 29 2019 12:21:46

chegewara@ubuntu:~/upwork/components/tests/core-dump$ espcoredump.py -p /dev/ttyUSB1 dbg_corefile ./build/core-dump.elf 
espcoredump.py v0.4-dev
```

Tested with master branch on esp32 and esp32 S3 with the same result.
```
#include <stdio.h>
#include "esp_attr.h"

// uint8_t global_var;
COREDUMP_DRAM_ATTR uint8_t global_var;

void app_main(void)
{

    global_var = 25;
    printf("%d\n", global_var); // use variable to avoid linker optimization
    assert(0);

}
```
Thanks

igrr, 2021-12-28:
@chegewara that looks correct, core dump only contains task stacks (hence local variables) and _doesn't_ contain .data or .bss sections or the heap.

chegewara, 2021-12-28:
@igrr 
```
COREDUMP_DRAM_ATTR places variable into DRAM area which will be included into dump.

COREDUMP_RTC_ATTR places variable into RTC area which will be included into dump.

COREDUMP_RTC_FAST_ATTR places variable into RTC_FAST area which will be included into dump.
```
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/core_dump.html#supported-notations-and-ram-regions

igrr, 2021-12-28:
@chegewara of course, you are right, I have overlooked the attribute in your issue report. We will investigate.

closed 2022-01-11 by espressif-bot