espressif/esp-idf #8152: DFS + CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y causes reading ADC hang

AxelLin, 2021-12-28:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v4.3.2-106-g8a7b2f326b
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

DFS + CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y causes reading ADC hang

### Steps to reproduce

Use examples/wifi/power_save snip code to reproduce the issue:
1. Set ESP32_RTC_CLK_SRC_INT_8MD256=y
2. Add below code for reading adc1

```
diff --git a/examples/wifi/power_save/main/power_save.c b/examples/wifi/power_save/main/power_save.c
index 4ad924665c..2d4a985165 100644
--- a/examples/wifi/power_save/main/power_save.c
+++ b/examples/wifi/power_save/main/power_save.c
@@ -19,6 +19,7 @@
 #include "esp_event.h"
 #include "esp_pm.h"
 #include "nvs_flash.h"
+#include "driver/adc.h"
 
 /*set the ssid and password via "idf.py menuconfig"*/
 #define DEFAULT_SSID CONFIG_EXAMPLE_WIFI_SSID
@@ -81,6 +82,19 @@ static void wifi_power_save(void)
     esp_wifi_set_ps(DEFAULT_PS_MODE);
 }
 
+static void test_adc(void)
+{
+    int adc_val;
+    adc1_config_width(ADC_WIDTH_BIT_12);
+    adc_gpio_init(ADC_UNIT_1, ADC1_CHANNEL_0);
+    adc1_config_channel_atten(ADC1_CHANNEL_0, ADC_ATTEN_DB_11);
+
+    vTaskDelay(pdMS_TO_TICKS(50));
+
+    adc_val = adc1_get_raw(ADC1_CHANNEL_0);
+    ESP_LOGI(TAG, "ADC val: %d", adc_val);
+}
+
 void app_main(void)
 {
     // Initialize NVS
@@ -111,5 +125,7 @@ void app_main(void)
     ESP_ERROR_CHECK( esp_pm_configure(&pm_config) );
 #endif // CONFIG_PM_ENABLE
 
+// Test reading adc
+     test_adc();
     wifi_power_save();
 }
```

3. build the power_save and run
Then observe device hang at below:
...
I (590) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (877) pm: Frequency switching co

Also note comment out below code can make CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y work.
https://github.com/espressif/esp-idf/blob/release/v4.3/components/esp_hw_support/port/esp32/rtc_sleep.c#L196-L202
Looks like 73384adfbe3a4365fb8fefdae9b1f9df4e509635 is the bad commit.

AxelLin, 2021-12-28:
BTW, when I test DFS + wifi, sometimes I got below error message:

E (19377) wifi_apb80m_request: WiFi needs 80MHz APB frequency to work, but got 40000000Hz
E (28126) wifi_apb80m_request: WiFi needs 80MHz APB frequency to work, but got 40000000Hz

Does that mean something wrong in my configuration?
Or I can just ignore it? (The device seems still working)

negativekelvin, 2021-12-28:
https://github.com/espressif/esp-idf/pull/8089

AxelLin, 2021-12-28:
> #8089

Above PR fixes the issue, thanks.

AxelLin, 2021-12-29:
> BTW, when I test DFS + wifi, sometimes I got below error message:
>
> E (19377) wifi_apb80m_request: WiFi needs 80MHz APB frequency to work, but got 40000000Hz E (28126) wifi_apb80m_request: WiFi needs [...]

I found the device shows below messages today:
E (36759743) wifi_apb80m_request: WiFi needs 80MHz APB frequency to work, but got 80000000Hz
E (37138888) wifi_apb80m_request: WiFi needs 80MHz APB frequency to work, but got 80000000Hz
E (37233375) wifi_apb80m_request: WiFi needs 80MHz APB frequency to work, but got 80000000Hz
E (37354657) wifi_apb80m_request: WiFi needs 80MHz APB frequency to work, but got 40000000Hz
E (37577531) wifi_apb80m_request: WiFi needs 80MHz APB frequency to work, but got 80000000Hz

I'm confused, isn't 80MHz == 80000000Hz?

AxelLin, 2021-12-29:
> BTW, when I test DFS + wifi, sometimes I got below error message:
> 
> E (19377) wifi_apb80m_request: WiFi needs 80MHz APB frequency to work, but got 40000000Hz E (28126) wifi_apb80m_request: WiFi needs 80MHz APB frequency to work, but got 40000000Hz

Above message only happens on v4.3 branch, it does not happen on v4.4 or master tree.

esp-lis, 2022-03-24:
> BTW, when I test DFS + wifi, sometimes I got below error message:
>
> E (19377) wifi_apb80m_request: WiFi needs 80MHz APB frequency to work, but got 40000000Hz E (28126) wifi_apb80m_request: WiFi needs [...]

I summarize the commits that fix this issue on 4.1, 4.2, 4.3, 4.4 as follows:
4.4 or master: [`75a1942`](https://github.com/espressif/esp-idf/commit/75a1942d5c7af4540ed34bc6941f959644c8b54d)
4.3 [`ae84265`](https://github.com/espressif/esp-idf/commit/ae842651280678b1a3c69f463603c08263103b25)
4.2 [`ebee6f3`](https://github.com/espressif/esp-idf/commit/ebee6f3c3f250006f9837b387c4b8a0da8f5fef3)
4.1 [`2ed9685`](https://github.com/espressif/esp-idf/commit/2ed96858a5cc97c9cbcd603fcb4d8a839873345)

Alvin1Zhang, 2022-03-25:
Thanks for reporting, feel free to reopen.

closed 2022-03-25 by Alvin1Zhang

AxelLin, 2022-03-25:
The https://github.com/espressif/esp-idf/issues/8152#issuecomment-1077490871 fixes the error mesage.
However it still hang unless applying https://github.com/espressif/esp-idf/pull/8089 .

AxelLin, 2022-03-25:
@Alvin1Zhang @esp-lis 
It's fine to close this one and trace https://github.com/espressif/esp-idf/issues/8007 instead.
However, nobody responses with https://github.com/espressif/esp-idf/issues/8007 .