espressif/esp-idf #8153: Unable to set AES-128 input string more than 16 char

tsctrl, 2021-12-28:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version DevKitC v1
- Module or chip used:  ESP32-WROOM-32
- IDF version : 3.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt.
- Using an IDE?: eclipse
- Power Supply:        external 5V

## Problem Description

Hi Esperessif team i am successfully encrypt plaintext to aes128 encrypted hex but unable to 
create longer AES encrypted text from MBED TLS. Is this mbed library limitation or the approach i did to input and output the encyption are wrong?

### Expected Behavior
Able to generate more that 16 character encrypted text.

### Actual Behavior
Only able to generate 16 character encrypted text.

### Steps to reproduce

1. Run the below code in setup in IDF init.
it will encrypt to 32 char long hex string successfully. 567a3b23b683d8488d5d40d2a56e31d2a
2. change the input char to more than 16 character: 
char *input = "abcdefghijklmnopabcdefghijklmnop";
3. change the output to unsigned char output[32];
4. increase the output logs array to 128
5. observe the serial output which is not correct: 
"567a3b23b683d8488d5d40d2a56e31d2a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5"

### Code to reproduce this issue

```
mbedtls_aes_context aes;

char * key = "abcdefghijklmnop";
char *input = "abcdefghijklmnop";
unsigned char output[16];

mbedtls_aes_init(&aes);
mbedtls_aes_setkey_enc(&aes, (const unsigned char*) key, strlen(key) * 8);
mbedtls_aes_crypt_ecb(&aes, MBEDTLS_AES_ENCRYPT, (const unsigned char*)input, output);
mbedtls_aes_free(&aes);

char result[50];
result[0] = '\0';
for (int i = 0; i < 32; i++) {
    char str[3];
    sprintf(str, "%02x", (int)output[i]);
    strlcat(result, str, sizeof(result));
    Serial.print(str);
}
Serial.print(result);
```

## Debug Logs
```
567a3b23b683d8488d5d40d2a56e31d2a5
567a3b23b683d8488d5d40d2a56e31d2a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5
```

ESP-Marius, 2021-12-28:
Hi,

You are using ECB which only encrypts 16 bytes at a time (and using ECB for anything is not really recommended, see e.g. https://en.wikipedia.org/wiki/Block_cipher_mode_of_operation#Electronic_codebook_(ECB))

For encrypting a string of arbitrary length I'd suggest looking at some of the other mode of operation, e.g. aes_ctr.

As always with cryptography, if you are planning to use this for anything serious then I'd highly recommend reading up on the subject or you are very likely make mistakes.

For more general "how to" questions like this I'd recommend our forum: `www.esp32.com`.
Closing the issue as this is not really related to esp-idf. Feel free to open another one if you encounter any bugs :smile:

closed 2021-12-28 by ESP-Marius

tsctrl, 2021-12-29:
i have no reference for this. it is an issue when there is no examples and there is no proof people successfully doing it. all question regarding this in esp32 forum or anywhere are left unanswered.

this is important to secure the esp devices communication and look like this is not an expressif concern to do it.

@ESP-Marius , @mahavirj

igrr, 2021-12-29:
Hi @tsctrl, security of data in storage and transit between ESP based devices is very important for us, and we try to achieve this by implementing a number of standard security protocols in ESP-IDF. You will find AES-XTS based hardware flash encryption, RSA-PSS based secure boot, HTTPS, MQTT over TLS implemented in ESP hardware or software. Generally these high level building blocks should be used when creating your application.

There is a famous (and somewhat ambiguous) saying, "don't roll your own crypto". What it means is that applying textbook cryptographic algorithms is more difficult and more risky than finding a proven high level protocol which meets your application needs.

Take this case as an example. Suppose there was an example in esp-idf demonstrating how to use AES to encrypt data of arbitrary length. You might think, that's great, let's take this example and use it in the application. But in fact there is a scary number of ways in which AES can be applied incorrectly, resulting in little or no security — the linked Wikipedia article mentions a few.

If you think there's a common application security requirement which isn't addressed yet by esp-idf, please let us know and we'll either suggest some existing solution or implement something to address such a requirement.