espressif/esp-idf #8154: esp32s3 build error in idf4.4-beta1

heculess, 2021-12-28:
## Environment
- Module or chip used:  [ESP32-S3]
- IDF version (v4.4-beta1):
- Build System: [idf.py]
- Operating System:     [Windows]
- Using an IDE?: [No)]
- Power Supply:  [USB]

E:/esp-idf/components/esp_rom/include/esp32s3/rom/secure_boot.h:30:42: error: conflicting declaration 'typedef struct ets_secure_boot_signature ets_secure_boot_signature_t'
 typedef struct ets_secure_boot_signature ets_secure_boot_signature_t;
                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
E:/esp-idf/components/esp_rom/include/esp32s3/rom/secure_boot.h:27:8: note: previous declaration as 'struct ets_secure_boot_signature_t'
 struct ets_secure_boot_signature_t;
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\users\heculess\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\time.h:11,
                 from E:/esp-idf/components/newlib/platform_include/time.h:22,
                 from ../components/posture/src/aliyun.cpp:1:
E:/esp-idf/components/esp_rom/include/esp32s3/rom/secure_boot.h:104:50: error: invalid application of 'sizeof' to incomplete type 'ets_secure_boot_signature_t'
 _Static_assert(sizeof(ets_secure_boot_signature_t) == 4096, "ets_secure_boot_signature_t should occupy 4096 Bytes in memory");

E:/esp-idf/components/esp_rom/include/esp32s3/rom/secure_boot.h:27:
"struct ets_secure_boot_signature_t;" should be " struct ets_secure_boot_signature;"

mahavirj, 2021-12-28:
@heculess Could you please try fix from https://github.com/espressif/esp-idf/commit/cb91b51c52c2dbdb3e063238293c41425b739d41#diff-e1f081fdbd902e92b9083d52c21ad99f95e121de17b3327e52c2999045fceb98? We will backport this fix to `release/v4.4` branch.

heculess, 2021-12-29:
@mahavirj  Yes, it can be worked

mahavirj, 2022-01-06:
@heculess Thanks for confirming, fix is now available at https://github.com/espressif/esp-idf/commit/530a06cf613ed2d9816163287d7323abf9834575

closed 2022-01-06 by mahavirj