espressif/esp-idf #8155: ESP_NETIF_TCPIP_ADAPTER_COMPATIBLE_LAYER cause build issue with time

chegewara, 2021-12-28:
Why disabling `ESP_NETIF_TCPIP_ADAPTER_COMPATIBLE_LAYER` in menuconfig is causing problems with standard time functions?
`error: 'tm' does not name a type; did you mean 'tms'?`
`strftime(buffer, 80, "%d-%m-%Y %H:%M:%S GMT", timeinfo);`

With option enabled all works fine.

Thanks

AxelLin, 2021-12-28:
Which esp-idf version?

chegewara, 2021-12-28:
master branch, sorry

ESP-YJM, 2021-12-28:
@chegewara Which example you used?

chegewara, 2021-12-28:
Sorry, it is not espressif example, it is from my library test code. It is based on sntp example.

chegewara, 2021-12-28:
I checked SNTP example now and its working. Looks like with this config enabled some headers are auto-magically included and with disabled i have to add:
```
#include <time.h>
#include <sys/time.h>
```

Sorry for confusion

closed 2021-12-28 by chegewara