espressif/esp-idf #8156: macro ESP_ERROR_CHECK_WITHOUT_ABORT with return

chegewara, 2021-12-28:
Can we have a macro like this or similar in `esp_err.t`, please, or maybe small modification of `ESP_ERROR_CHECK_WITHOUT_ABORT`?

```
#define ESP_ERROR_CHECK_WITH_RETURN(x) ({                                         \
        esp_err_t err_rc_ = (x);                                                    \
        if (unlikely(err_rc_ != ESP_OK)) {                                          \
            _esp_error_check_failed_without_abort(err_rc_, __FILE__, __LINE__,      \
                                                  __ASSERT_FUNC, #x);               \
            return err_rc_;                                                         \
        }                                                                           \
        err_rc_;                                                                    \
    })
```

dobairoland, 2021-12-30:
Hi @chegewara. Can you check please if `ESP_RETURN_ON_ERROR` (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/error-handling.html#esp-return-on-error-macro, https://github.com/espressif/esp-idf/blob/master/components/esp_common/include/esp_check.h#L27) is sufficient for you?

`ESP_ERROR_CHECK` is with abort, `ESP_ERROR_CHECK_WITHOUT_ABORT` is without, so I think your function would have to be called `ESP_ERROR_CHECK_WITHOUT_ABORT_WITH_RETURN` which is getting too complicated.

I would combine `ESP_ERROR_CHECK_WITHOUT_ABORT` and `ESP_RETURN_ON_ERROR` and define my custom one in my project if I had to.

chegewara, 2021-12-30:
Hi
yes, it looks good, i didnt see that and othe macros from `esp_common.h`.

Thanks

closed 2021-12-30 by chegewara