espressif/esp-idf #8157: Facing Problem in Connecting ESP32 with MicroSD Card Module and PDM Mic

JugalAdhyaru17, 2021-12-28:
**PDM Breakout** : (https://cdn-learn.adafruit.com/downloads/pdf/adafruit-pdm-microphone-breakout.pdf)

I have ESP32-V1 Module and I'm Trying to connect MicroSD Card Module and PDM Microphone Breakout With ESP32 but Unable to Connect in ESP-IDF Tool but In Arduino IDE I'm able Access SD Card Modules With Same Pinouts and It is able to record the Audio but Unable to Unable to Connect PDM Microphone Breakout in Arduino IDE.

Hardware I'm Using : 

1. ESP32-V1 Board
2. PDM Microphone Breakout
3. MicroSD Card Module

Can Anyone Suggest What To do in this Situation?

Thanks,
In Advance.

Sunlu2333, 2022-11-17:
> **PDM Breakout** : (https://cdn-learn.adafruit.com/downloads/pdf/adafruit-pdm-microphone-breakout.pdf)
>
> I have ESP32-V1 Module and I'm Trying to connect MicroSD Card Module and PDM Microphone [...]

Hello! I have encountered a similar problem, can you share me with the solution? Thank you!

omnitex, 2026-11-02:
Hi @JugalAdhyaru17 @Sunlu2333 , sorry for the late reply. Your log shows two separate problems.

**1. SD init fails at CMD8** (`send_if_cond (1) returned 0x108`). The card answered, but the reply was corrupted, which usually points to wiring or signal problems. Please share:
- The GPIOs for SD MOSI, MISO, SCK and CS, and your `sdspi_device_config_t` / `spi_bus_config_t` code.
- Which microSD module you use, and whether it's a 5V module with a level shifter.
- Whether the SD lines have 10k pull-ups to 3.3V (see [SD pull-up requirements](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html)).

**2. Console output turns to garbage after the SD error.** This usually means a PDM mic pin is on the UART0 console pins (GPIO1/GPIO3). Please share your PDM CLK and DATA GPIOs.

The pin list in the Adafruit PDM guide (PA10/PB11/PA20, D1/TX, D9...) is for SAMD21 boards and doesn't apply to ESP32. On ESP32, PDM RX works only on `I2S_NUM_0`, and CLK/DATA can go on most free GPIOs. Avoid GPIO1/3, the flash pins (6-11), strapping pins and the SD pins.

v4.4-beta1 is no longer supported. Please retry on a supported release: first `examples/storage/sd_card/sdspi` (SD only, mic disconnected), then add `examples/peripherals/i2s/i2s_basic/i2s_pdm`.

If we don't hear back, we'll close this issue. Feel free to reopen it with the details above.