espressif/esp-idf #8159: USB HOST compiler errors, too many errors

ChienYangChang, 2021-12-28:
## Environment

- Development Kit:  esp32s3
- Build System:        idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     windows 10
- Using an IDE?: No
- Power Supply:  USB

## Problem Description: just compiler the example of esp-idf\examples\peripherals\usb\host\cdc\cdc_acm_host, there're many errors even i solve it one by one, but that it appears many. Please help to test it for fully coverage. thanks ~

### Expected Behavior

### Actual Behavior

### Steps to reproduce

zikalino, 2021-12-29:
thank you for reporting @ChienYangChang ! I will try to reproduce

ChienYangChang, 2021-12-29:
Hi Zikalino

Thanks for your help, btw, the code dependency are very high, that's why compiler errors rapidly by including path, maybe modify code structure to CMSIS, to split HAL and APP layer clean. 

Another way, i have already fixed all of compiler errors in previous version of ESP-IDF for USB HOST function, but the task didn't recognize USB CDC from cdc_acm_host_open, i'm debugging for this. if you have any suggestion, please inform me.  

In Last, does any document to describes the chip register for programming used ? i wanna to solve the usb host problem. 

Many thanks

ChienYangChang, 2021-12-29:
HI zikalino
    In v4.4, it's okay in my testing, so i think that is easy to narrow down. Thanks 
B.R.

Dazza0, 2021-12-29:
@ChienYangChang We currently don't support USB Host in TinyUSB. If you need USB Host support, use the USB Host Library instead.

ChienYangChang, 2021-12-29:
Hi Dazza0
    Thanks for your response, i already done for that, i use the usb host library to communicate with the device of usb cdc in v4.4, the function is work properly after testing today. 
B.R.

closed 2021-12-29 by ChienYangChang