espressif/esp-idf #8160: Facing pthread errors while adding the Bootloader Hook(.cpp files) in ESP32 2nd stage bootloader

vim-git, 2021-12-28:
## Environment
- Development Kit: ESP32-Mini-1
- Kit version :
- Module or chip used: ESP32
- IDF version : v4.2.1
- Build System: Make|CMake|idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Windows.
- (Windows only) environment type:ESP Command Prompt.
- Using an IDE?: Yes, Eclipse
- Power Supply: USB

## Problem Description

Facing issue related to pthread while adding the **Bootloader Hook as .cpp files in ESP32 2nd stage bootloader after bootloader init**. 
It started giving the pthread issue at linking time. 

If I am adding the pthread module and freertos module in CMakeList(components/Bootloader/subproject/main/CMakelist.txt) then I am started facing in pthread module related to undefined reference for freertos files. 

### Expected Behavior

I wanted to add the **bootloader hook in .cpp file** in ESP32 2nd stage bootloader after bootloader init in bootloader_start.c file. I want t**o include and create and delete the cpp object in bootloader hook**.

### Actual Behavior
Observation 1:
While adding the bootloader hook as cpp files in ESP32 2nd stage bootloader, it produces the pthread as undefined at linking time. 
Please find the attached log below
[pthread errros while addign in 2nd stage bootloader.txt](https://github.com/espressif/esp-idf/files/7785781/pthread.errros.while.addign.in.2nd.stage.bootloader.txt)

Observation 2: 
To solve pthread errors in linker files, if I am adding **set(pre_requires pthread freertos)** in **components/Bootloader/subproject/main/CMakelist.txt** file then I started facing errors in pthread module for undefined reference for freertos. Please find the error log below
[Errors with Pthread freertos in 2nd stage bootloader.txt](https://github.com/espressif/esp-idf/files/7785783/Errors.with.Pthread.freertos.in.2nd.stage.bootloader.txt)

### Steps to reproduce

1. Add the bootloader hook in bootloader_start.c file file 
2. In .cpp hook file, create the object and performed the read/write the memory operation using bootloader flash API's.
3. delete the object.

### Code to reproduce this issue

I have taken reference from ESP-IDF v4.3.1 for adding the bootloader hook, I have added the code same as by create .cpp files and class object etc.

Also wanted to know, Is it possible to add the cpp code in ESP32 bootloader as hook or not?

igrr, 2021-12-28:
@vim-git at the moment, nontrivial use of C++ is supported in the context of the 2nd stage bootloader. `pthread` is not the only dependency, we also don't call global contructors, don't support exceptions or RTTI, don't implement static init guards in the 2nd stage bootloader. You'll have to use C to implement bootloader hooks.

zikalino, 2021-12-29:
@0xjakob please take a look

0xjakob, 2022-03-17:
@vim-git As @igrr stated, we currently don't support C++ in the 2nd stage bootlaoder and we also don't know if we ever will. Hence, I'm closing this issue for now. If you have a particular problem that is still pending, please let us know and feel free to re-open the issue.

closed 2022-03-17 by 0xjakob