espressif/esp-idf #8161: function "tu_fifo_peek_at_n" return vlaue incompatible

dreamtrek, 2021-12-29:
## Environment

- IDF version:
v4.4-beta1

## Problem Description
it returned a bool value instead of the data length

### Code to reproduce this issue

```cpp
/******************************************************************************/
/*!
    @brief Read n items without removing it from the FIFO
    This function checks for an overflow and corrects read pointer if required.

    @param[in]  f
                Pointer to the FIFO buffer to manipulate
    @param[in]  offset
                Position to read from in the FIFO buffer with respect to read pointer
    @param[in]  p_buffer
                Pointer to the place holder for data read from the buffer
    @param[in]  n
                Number of items to peek

    @returns Number of bytes written to p_buffer
 */
/******************************************************************************/
uint16_t tu_fifo_peek_at_n(tu_fifo_t* f, uint16_t offset, void * p_buffer, uint16_t n)
{
  _ff_lock(f->mutex_rd);
  bool ret = _tu_fifo_peek_at_n(f, offset, p_buffer, n, f->wr_idx, f->rd_idx, TU_FIFO_COPY_INC);
  _ff_unlock(f->mutex_rd);
  return ret;
}

```

zikalino, 2021-12-29:
Just checked, indeed an issue in Espressif's TinyUSB fork here: https://github.com/espressif/tinyusb/blob/075334af80303488a8c874255d725cb4ca15a7de/src/common/tusb_fifo.c#L486

@dreamtrek we will fix it and keep you updated

alisitsyn, 2022-02-07:
@dreamtrek,
The fix was merged in commit 6ddf2ea05eaeb7ffb9ea50414b2b7986a4fa8bdd. Will be synchronized with github soon.

closed 2022-09-06 by Alvin1Zhang