espressif/esp-idf #8167: Flash only by specifying .elf/.bin?

silardg, 2021-12-29:
Is there a command with idf.py or any other that will allow me to flash my device by passing the .elf/.bin file to it? I don't want to rebuild the whole project and then flash.

silardg, 2021-12-29:
What I'm trying to achieve is to build the project on a separate device and then transfer the build folder and flash it locally.

igrr, 2021-12-29:
If you have the `build` directory, you can run the following (adjust port and baud rate as needed):
```
cd build
esptool.py --chip esp32 --port=/dev/ttyUSB0 --baud=921600 write_flash @flash_args
```
Here `flash_args` is a file inside `build` directory, containing addresses and names of files to flash. `@` sign tells esptool to read arguments from that file.
For this you only need esptool installed, not the whole esp-idf. You can install esptool using `pip`.

dobairoland, 2021-12-30:
We have also a related documentation page which might help: https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flashing-firmware.html

Here is a page about how to install standalone `esptool` without ESP-IDF: https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html

silardg, 2021-12-30:
> esptool.py --chip esp32 --port=/dev/ttyUSB0 --baud=921600 write_flash @flash_args

Yepp this works fine. Thanks!

closed 2021-12-30 by silardg