espressif/esp-idf #8170: Add interrupt support to lan8720 component

romatetemadze, 2021-12-30:
**General Idea**
In some designs we want LAN8720 to wakeup ESP32 based on the interrupt pin accertion. My requirement was to have following:
- Enable ethernet
- Once ethernet event "link down" is generated, send esp32 to light or deep sleep
- Once etherner cable is plugged back, pull a wakeup pin so that esp32 wakes up

LAN8720 has a specific PIN to allow such behavior (nINT). If registers are set, it will assert low on every configured event until status register is read.

**Desired additions**

Add methods to lan8720 to:
- Configure interrupt ETH_PHY_ISR_REG_ADDR
- Add clear interrupt call

**My custom solution**
I have modified esp_eth_phy_lan8720.c and added following custom calls:

`esp_err_t lan8720_set_ISR_Reg(esp_eth_phy_t *phy) {
	phy_lan8720_t *lan8720 = __containerof(phy, phy_lan8720_t, parent);
	esp_eth_mediator_t *eth = lan8720->eth;
	imr_reg_t imrdd;

	imrdd.auto_nego_page_received=0;  		/* Auto-Negotiation Page Received */
	imrdd.parallel_detect_falut=0;    			/* Parallel Detection Fault */
	imrdd.auto_nego_lp_acknowledge=0; 		/* Auto-Negotiation LP Acknowledge */
	imrdd.link_down=0;                			/* Link Down */
	imrdd.remote_fault_detect=0;      			/* Remote Fault Detect */
	imrdd.auto_nego_complete=0;       		/* Auto-Negotiation Complete */
	imrdd.energy_on_generate=1;			/* ENERYON generated (cable connected with another host) */
	
    PHY_CHECK(eth->phy_reg_write(eth, lan8720->addr, ETH_PHY_IMR_REG_ADDR, imrdd.val) == ESP_OK,
              "Write ISR_Reg failed", err);

	PHY_CHECK(eth->phy_reg_read(eth, lan8720->addr, ETH_PHY_IMR_REG_ADDR, &(imrdd.val)) == ESP_OK,
	              "Read IMR_Reg failed", err);
	ESP_LOGE(TAG,"ID7>>0x%X", imrdd.energy_on_generate);
	return ESP_OK;
err:
    return ESP_FAIL;	
}

esp_err_t lan8720_read_ISR_Reg(esp_eth_phy_t *phy) {
	phy_lan8720_t *lan8720 = __containerof(phy, phy_lan8720_t, parent);
	esp_eth_mediator_t *eth = lan8720->eth;
	imr_reg_t imrdd;
	PHY_CHECK(eth->phy_reg_read(eth, lan8720->addr, ETH_PHY_IMR_REG_ADDR, &(imrdd.val)) == ESP_OK,
	              "Read IMR_Reg failed", err);	
	ESP_LOGE(TAG,"ID7=0x%X", imrdd.energy_on_generate);
    return ESP_OK;
err:
    return ESP_FAIL;
}
`

Then, I have wired the nINT pin to IO34 of ESP32-Wrover-E. Once phy_lan8720_t generates an event of "LINK DOWN", I am setting IO34 as wakeup source and configure the registers to enable nINT accertion. Solution proved working well. Solutyion can be modded to allow the notification of the event caused the hardware wakeup.

**Nice to have**
- Add forcing power mode to the driver (ETH_PHY_MCSR_REG_ADDR). In this mode lan8720 goes to sleep while ENERGYON event is till being monitored.
`esp_err_t lan8720_set_PWR_Reg(esp_eth_phy_t *phy) {
	phy_lan8720_t *lan8720 = __containerof(phy, phy_lan8720_t, parent);
	esp_eth_mediator_t *eth = lan8720->eth;
	mcsr_reg_t pwrdt;
	
	pwrdt.en_energy_detect_powerdown=1;
    PHY_CHECK(eth->phy_reg_write(eth, lan8720->addr, ETH_PHY_MCSR_REG_ADDR, pwrdt.val) == ESP_OK,
              "Write EDPWRDOWN failed", err);
			  
	PHY_CHECK(eth->phy_reg_read(eth, lan8720->addr, ETH_PHY_MCSR_REG_ADDR, &(pwrdt.val)) == ESP_OK,
	              "Read EDPWRDOWN failed", err);
	ESP_LOGE(TAG,"EDPWRDOWN>>0x%X", pwrdt.en_energy_detect_powerdown);
    return ESP_OK;
err:
    return ESP_FAIL;	`

kostaond, 2022-01-10:
Hi @romatetemadze, thank you for your suggestion. It's definitely something what some users could find useful. However, I am generally a little bit hesitant to implement specific PHY chip features into IDF. From my perspective, IDF's goal is to provide basic support with functionality aligned among the PHY chips so users could quickly start their Ethernet project and to keep maintainability at some acceptable level for us at the same time. When users require advanced functionality they can implement it by themselves as you did so.

However, I think I have good news at the end! :) I am planning to refactor PHY chips code since majority of the code is common. Therefore I would like to create a file with common functionality like speed/duplex/power/etc. and separate chip specific files with chip specific functionality. This would make maintainability of generic PHY functionality simpler and clearly separate chip specific functionality. I am also planning to make an option to extend the driver with user specific implementation which could be accessed via `eth_ioctl`. Your proposed feature could be good candidate to prove that concept. But please be patient since this is not high priority task.

romatetemadze, 2022-01-15:
> Hi @romatetemadze, thank you for your suggestion. It's definitely something what some users could find useful. However, I am generally a little bit hesitant to implement specific PHY chip features [...]

Thanks for your reply. I udnerstand all of your pointa and those are fair enough. Definetely I am not in a rush, just thought it could be a valuable feature to some project (especially those multiprotocol), so I decided to make a request.

When you're up to it and need any tester, kindly consider me (especially with the LAN87xx series).

Best.