espressif/esp-idf #8171: Tool to create img file for FATFS

MaazSk, 2021-12-30:
**Is your feature request related to a problem? Please describe.**

I am using FATFS in my code and I want to load some of the files from the computer to it. So, if we have a tool as we have for spiffs like spiffsgen or mkspiffs which are used to generate .img would be great as I am unable to create a fatfs image.

**Describe the solution you'd like**

As I have used mkspiffs to create a .bin file so a tool like mkspiffs or spiffsgen would be of great help

**Describe alternatives you've considered**

I have considered these 2 alternatives as follows
- [ESP32-mkfatfs](https://github.com/jkearins/ESP32_mkfatfs)
- [Mk-esp32fat](https://github.com/TobleMiner/mk_esp32fat)

But I am unable to build both the files because they are not supported for ESP-IDF v4.3

I know there is a utility in the master branch in example/storage/fatfsgen but I am not sure whether it will support my version of esp-idf or not

dobairoland, 2021-12-30:
I'm sorry @MaazSk but we are not adding new features to already released versions. The FATFS generating tool is available starting from v5.0.

The tool relies on some CMake-level project integration but it not using anything else from ESP-IDF. You can easily generate your image using the `master` branch and include it into your v4.3.x-based project.

MaazSk, 2021-12-30:
@dobairoland Thank you for the quick reply

Do I need to install complete esp-idf or just downloading fatfsgen.py will do the work for me?

MaazSk, 2021-12-30:
@dobairoland Thank you for the reply!

As you said I was trying to generate the img file but I got the following error

> maaz@maaz:~/esp/esp_master_version/esp-idf/components/fatfs$ python3 wl_fatfsgen.py /home/maaz/Downloads/Telegram_Desktop/TS-Dolphin-SMC3P-0000001/Required_Cerificates
> Traceback (most recent call last):
>   File "/home/maaz/esp/esp_master_version/esp-idf/components/fatfs/wl_fatfsgen.py", line 196, in <module>
>     wl_fatfs.wl_generate(input_dir)
>   File "/home/maaz/esp/esp_master_version/esp-idf/components/fatfs/wl_fatfsgen.py", line 175, in wl_generate
>     self.plain_fatfs.generate(input_directory=input_directory)
>   File "/home/maaz/esp/esp_master_version/esp-idf/components/fatfs/fatfsgen.py", line 190, in generate
>     self._generate_partition_from_folder(folder_name, folder_path=path_to_folder, is_dir=True)
>   File "/home/maaz/esp/esp_master_version/esp-idf/components/fatfs/fatfsgen.py", line 183, in _generate_partition_from_folder
>     self._generate_partition_from_folder(os.path.join(smaller_path, path), folder_path=folder_path)
>   File "/home/maaz/esp/esp_master_version/esp-idf/components/fatfs/fatfsgen.py", line 174, in _generate_partition_from_folder
>     self.create_file(name=file_name, extension=extension, path_from_root=split_path[1:-1] or None)
>   File "/home/maaz/esp/esp_master_version/esp-idf/components/fatfs/fatfsgen.py", line 101, in create_file
>     self.root_directory.new_file(name=name, extension=extension, path_from_root=path_from_root)
>   File "/home/maaz/esp/esp_master_version/esp-idf/components/fatfs/fatfsgen_utils/fs_object.py", line 190, in new_file
>     free_cluster, free_entry, target_dir = self.allocate_object(name=name,
>   File "/home/maaz/esp/esp_master_version/esp-idf/components/fatfs/fatfsgen_utils/fs_object.py", line 183, in allocate_object
>     free_entry.allocate_entry(first_cluster_id=free_cluster.id,
>   File "/home/maaz/esp/esp_master_version/esp-idf/components/fatfs/fatfsgen_utils/entry.py", line 111, in allocate_entry
>     raise TooLongNameException(
> fatfsgen_utils.exceptions.TooLongNameException: Maximal length of the object name is 8 characters and 3 characters for extension!
>

dobairoland, 2021-12-30:
The tool is under development and the initial version was published only recently. The support for long file names and FAT16 is coming in the next months.

MaazSk, 2021-12-30:
@dobairoland, Thank you for the update.

Can you please let us know whenever you release the newer version.

Thanks in advance.

dobairoland, 2021-12-30:
Alright, lets keep this open. It will be closed by the commit adding support for long filenames.

MaazSk, 2021-12-30:
Okay, Thank you!

closed 2022-03-12 by espressif-bot