espressif/esp-idf #8173: I2C - PSRAM error when SPIRAM_USE_MALLOC and ESP_INTR_FLAG_IRAM

SunnyRuss, 2021-12-30:
in the file I2C.C at line 1380 the code reads:

       if (cmd_link->cmd.hw_cmd.op_code == I2C_LL_CMD_WRITE || cmd_link->cmd.hw_cmd.op_code == I2C_LL_CMD_READ) {
            if (cmd_link->cmd.data != NULL && !esp_ptr_internal(cmd_link->cmd.data)) {
                return false;
            }
        }

I think it should be like this or similar:

       if ((cmd_link->cmd.hw_cmd.op_code == I2C_LL_CMD_WRITE && !i2c_cmd_is_single_byte(&cmd_link->cmd)) || 
            cmd_link->cmd.hw_cmd.op_code == I2C_LL_CMD_READ) {
            if (cmd_link->cmd.data != NULL && !esp_ptr_internal(cmd_link->cmd.data)) {
                return false;
            }
        }

I believe this because when you write just one byte, the value of cmd.data is not a pointer, but the actual byte to be sent out over the I2C bus and that value will never be a valid esp_ptr_internal() address.  As such, you always get a PSRAM buffer error.

This is in release/v4.4

o-marshmallow, 2021-12-31:
Hi @SunnyRuss ,
Thanks for reporting this, I am going to manage this issue

closed 2022-01-04 by espressif-bot

AxelLin, 2022-02-04:
@o-marshmallow 
v4.4 needs this fix.

o-marshmallow, 2022-02-15:
Hi @AxelLin ,
A backport to v4.4 is on the way. 
However, v4.3 doesn't need this fix since it doesn't have this "single-byte-in-pointer" feature.