espressif/esp-idf #8175: esp_image: Image hash failed - image is corrupt

v-3n0m, 2021-12-31:
## Environment

- Module or chip used:  ESP32-S3-WROOM-1
- IDF version :                v4.4-dev-3703-gddc44956bf
- Build System:              idf.py
- Compiler version :      xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Windows 11
-  environment type:    ESP Command Prompt
- Using an IDE?:            Vscode 1.63.2 using IDF extension
- Power Supply:            USB

## Problem Description

I was testing the IP5360 i2c IC on my custom build esp32s3 board. it was working fine yesterday but  when i tried to flash and run the program again i am getting the bellow error. The same program runs fine on my other boards. 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403bb452
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1710
load:0x403b6000,len:0x8c8
load:0x403ba000,len:0x2eb4
entry 0x403b61c0
I (29) boot: ESP-IDF v4.4-dev-3703-gddc44956bf 2nd stage bootloader
I (29) boot: compile time 14:34:40
I (29) boot: chip revision: 0
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
W (50) bootloader_random: RNG for ESP32-S3 not currently supported
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08918h ( 35096) map
I (103) esp_image: segment 1: paddr=00018940 vaddr=3fc91ae0 size=0269ch (  9884) load
I (107) esp_image: segment 2: paddr=0001afe4 vaddr=40374000 size=05034h ( 20532) load
I (117) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=18ef8h (102136) map
I (138) esp_image: segment 4: paddr=00038f20 vaddr=40379034 size=08aa0h ( 35488) load
I (146) esp_image: segment 5: paddr=000419c8 vaddr=50000000 size=00010h (    16) load
E (146) esp_image: Image hash failed - image is corrupt
E (149) boot: Factory app partition is not bootable
E (154) boot: No bootable app partitions in the partition table

i tried running the helloworld program it also gives same error. also done erase_flash still no change.

Any help would be appreciated.

Thanks

o-marshmallow, 2022-01-13:
Hi @v-3n0m ,
As you've recompiled hello world and reflashed it from scratch, I would tend to say that this is a hardware issue, corrupted flash maybe. Can you have a try flashing the whole flash with random data, read them back and compare the two files?

To generate a 2MB random file, you can use (on Linux):
```
dd if=/dev/random of=flash.img bs=2M count=1
```

Then flash it to your board:
```
esptool.py write_flash 0x0 flash.img
```

Read the flash back, store this in another file"
```
esptool.py read_flash 0 2097152 dump.img
```

Finally, compared them:
```
md5sum flash.img dump.img
```
If the given MD5 are different, then the flash is defective

mythbuster5, 2022-02-18:
@v-3n0m  I have a question. Have you tried deep-sleep or ULP something else?

Alvin1Zhang, 2022-02-24:
@v-3n0m Thanks for reporting, would you please help share if any further updates on the issue? Thanks.

AxelLin, 2022-02-24:
> @v-3n0m I have a question. Have you tried deep-sleep or ULP something else?

@mythbuster5 
I don't understand why this issue is related to deep-sleep or ULP?
Any known issue?

@v-3n0m 
Please provide your sdkconfig

sudeep-mohanty, 2022-09-06:
@v-3n0m Apologies for the delayed follow up!
We have had some power setting adjustments for the esp32s3 that could help explain the behavior you observed. These changes were merged into v4.4 under a2e1b6756e2bb053c3699856d3b902bebe5f5912 and into the mainline under ab69df3ea762b74f0747517018cfc631c55a9dbc. Would you mind trying out with these fixes and let us know your observations?

sudeep-mohanty, 2022-10-05:
Closing this as there is a lack of activity. Please feel free to reopen in case the issue is not resolved.

closed 2022-10-05 by sudeep-mohanty