espressif/esp-idf #8177: ESP MESH  - Poor mesh performance when Wifi continuously scanning

vinicius-smartme, 2021-12-31:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: v4.3.1-1-g2ddc02a4df
- Build System:         [CMake]
- Compiler version: 8.4.0
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [Visual Studio + VisualGDB]
- Power Supply:         [USB]

## Problem Description
When using the mesh in manual networking mode and with the router turned off, the continuous Wifi scanning causes the mesh to have a poor performance. 

My setup is a mesh network that can operate while the router is turned off. When the router is on, the root connects to it. I am using 4 devices close to each other (same room and no obstacles between them). Under a moderate traffic and no router there are lots of drops and highly delayed packets (both TX and RX). I can observe that the `esp_mesh_send()` returns always `ESP_OK` (mesh_tos set to `MESH_TOS_P2P`), but frequently I cannot see the packet been received in the other side (both root to node and node to root). I am using duty cycle of 100%, WiFi AMPDU RX/TX is off and rssi is at a good level.

The same test done with the router on, this problem doesn't happen. There are still some packet losses and some delay spikes, but at an acceptable level.

As a workaround, I am stopping the Wifi scanning periodically and then restarting it. This mitigates partially the problem, but when the Wifi is scanning again, the mesh performance is decreased. This is not acceptable.

_**PS.: due confidentiality I cannot post neither the code nor the logs.**_

### Expected Behavior
- The usage of `esp_mesh_send()` with `MESH_TOS_P2P` to deliver reliably the packets between the devices.
- Mesh performance to be acceptable (few losses and small delays) while Wifi is scanning.

### Actual Behavior
- The usage of `esp_mesh_send()` presents no reliability.
- Mesh performance is not acceptable (lost of losses and high delays) while Wifi is scanning.

### Steps to reproduce
As a closer way to reproduce the problem, a modified version from the example [manual_networking](https://github.com/espressif/esp-idf/tree/master/examples/mesh/manual_networking) can be used.

1. Modify the manual_networking to increase the network traffic (including some timed packet response in order to observe the packet losses/delay)
2. Setup 4 devices (1 pre-defined root)
3. Start the devices and keep the router off
4. Observe the network traffic and the packets TX/RX
5. Turn on the router and wait the router to connect
6. Observe the network traffic and the packets TX/RX

shenjun7, 2022-07-27:
If the manual networking node cannot find the parent, it will always scan. Setting up a network without a router can be achieved by setting fix-root esp_mesh_fix_root(true) esp_mesh_set_type().

vinicius-smartme, 2022-08-02:
The `esp_mesh_fix_root(true)` is already called for the pre-defined root. The `esp_mesh_set_type()` has no relation with the network performance.

shenjun7, 2022-08-09:
The logic of manual networking example makes it difficult to avoid this problem.
You can implement a no-router network in https://github.com/espressif/esp-idf/tree/master/examples/mesh/internal_communication, by `esp_mesh_fix_root() esp_mesh_set_type`. 
You can also refer to https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/no_router.

vinicius-smartme, 2022-08-16:
The user case for this problem is simple and a common situation:
- To have a good network performance for a manual mesh network while the router is temporally off.

I tried many things and it seems this is currently not possible. Maybe @shenjun7 could let me know if this will be set as an issue and fixed or just a TBD question.

Thanks.

shenjun7, 2022-08-17:
The logic of manual networking example makes it impossible to avoid this problem.
You can implement a no-router network in https://github.com/espressif/esp-idf/tree/master/examples/mesh/internal_communication, by esp_mesh_fix_root() esp_mesh_set_type.
You can also refer to https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/no_router.

vinicius-smartme, 2022-08-17:
So this issue can be closed as `won't fix` or `cannot be fixed`.

closed 2022-08-17 by vinicius-smartme

dnlmsr, 2024-06-27:
Sorry for reviving a dead issue, but I'm currently facing the same exact problem.
In my implementation I don't have a router for the root node to connect, in fact I only need a stable Wi-Fi mesh without internet connectivity so that the nodes can talk to each other.
@vinicius-smartme You mentioned a workaround by stopping the Wi-Fi scan, how did you manage to do it? Have you found another workaround for this problem?

vinicius-smartme, 2024-06-28:
> Sorry for reviving a dead issue, but I'm currently facing the same exact problem. In my implementation I don't have a router for the root node to connect, in fact I only need a stable Wi-Fi mesh [...]

Hey @dnlmsr 

If you don't have a router, you should use the examples cited above by @shenjun7. 

My problem was different. Usually I have a router to connect to, but could happen this router is OFF for some time. In this case, the WiFi will continuously scan and degrade the mesh performance significantly.

My solution was to manually control the scan by enabling and disabling the scanning. You can enable the scanning inside some timer and after the scan is finished, you disable the scanning and restart the timer.