espressif/esp-idf #8178: WiFi promiscuous mode crash / apparent heap corruption

cmsedore, 2021-12-31:
I believe there is a bug in the WiFi promiscuous mode packet receiving code in IDF v4.3.2.  I get random crashes with the example code below, which does not itself directly use the heap. The crash can be triggered almost immediately or may take a few minutes to occur.  It is possible that a specific WiFi packet is triggering this crash. (This is actually likely, in my opinion, given the random times at which it happens.)

Console output and then code at the end.

Console output:
```
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7412
load:0x40078000,len:14912
ho 0 tail 12 room 4
load:0x40080400,len:3688
0x40080400: _init at ??:?

entry 0x4008067c
[[[[...snip...]]]]
I (463) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (473) MAIN: Initializing WiFi
I (503) MAIN: drop 0, free heap 260460
I (513) wifi:wifi driver task: 3ffbf5c0, prio:23, stack:6656, core=0
I (513) system_api: Base MAC address is not set
I (513) system_api: read default base MAC address from EFUSE
I (533) wifi:wifi firmware version: eb52264
I (533) wifi:wifi certification version: v7.0
I (533) wifi:config NVS flash: enabled
I (533) wifi:config nano formating: disabled
I (533) wifi:Init data frame dynamic rx buffer num: 32
I (543) wifi:Init management frame dynamic rx buffer num: 32
I (543) wifi:Init management short buffer num: 32
I (553) wifi:Init dynamic tx buffer num: 32
I (553) wifi:Init static rx buffer size: 1600
I (553) wifi:Init static rx buffer num: 10
I (563) wifi:Init dynamic rx buffer num: 32
I (563) wifi_init: rx ba win: 6
I (573) wifi_init: tcpip mbox: 32
I (573) wifi_init: udp mbox: 6
I (573) wifi_init: tcp mbox: 6
I (583) wifi_init: tcp tx win: 5744
I (583) wifi_init: tcp rx win: 5744
I (593) wifi_init: tcp mss: 1440
I (593) wifi_init: WiFi IRAM OP enabled
I (593) wifi_init: WiFi RX IRAM OP enabled
I (603) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (703) wifi:mode : sta (1c:9d:c2:c3:8b:b4)
I (703) wifi:enable tsf
I (703) wifi:ic_enable_sniffer
I (703) MAIN: Waiting for packets...
I (1503) MAIN: drop 135, free heap 239148
I (2503) MAIN: drop 308, free heap 239148
I (3503) MAIN: drop 470, free heap 239148
[[[[...snip...]]]]
I (63503) MAIN: drop 12919, free heap 239148
I (64503) MAIN: drop 13071, free heap 239148
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008d478  PS      : 0x00060c33  A0      : 0x8008df22  A1      : 0x3ffbf380  
0x4008d478: remove_free_block at /home/esp4/esp/esp-idf-v4.3.2/components/heap/heap_tlsf.c:214
 (inlined by) block_locate_free at /home/esp4/esp/esp-idf-v4.3.2/components/heap/heap_tlsf.c:448
 (inlined by) tlsf_malloc at /home/esp4/esp/esp-idf-v4.3.2/components/heap/heap_tlsf.c:757

A2      : 0x3ffb7374  A3      : 0x0000003c  A4      : 0x3ffb2224  A5      : 0x3ffb4d94  
A6      : 0x3ffb2220  A7      : 0x3ffb8e94  A8      : 0x3ffc4758  A9      : 0x00000017  
A10     : 0xebdc8346  A11     : 0x0000000a  A12     : 0x40edea84  A13     : 0x00060c23  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xebdc834e  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4008d475:0x3ffbf380 0x4008df1f:0x3ffbf3a0 0x40081771:0x3ffbf3c0 0x400817a5:0x3ffbf3e0 0x4008f6c9:0x3ffbf400 0x400849c9:0x3ffbf420 0x400fe8e9:0x3ffbf440 0x40093657:0x3ffbf480 0x400938e3:0x3ffbf4c0 0x400910de:0x3ffbf500 0x4008beb9:0x3ffbf530
0x4008d475: remove_free_block at /home/esp4/esp/esp-idf-v4.3.2/components/heap/heap_tlsf.c:212
 (inlined by) block_locate_free at /home/esp4/esp/esp-idf-v4.3.2/components/heap/heap_tlsf.c:448
 (inlined by) tlsf_malloc at /home/esp4/esp/esp-idf-v4.3.2/components/heap/heap_tlsf.c:757

0x4008df1f: multi_heap_malloc_impl at /home/esp4/esp/esp-idf-v4.3.2/components/heap/multi_heap.c:197

0x40081771: heap_caps_malloc at /home/esp4/esp/esp-idf-v4.3.2/components/heap/heap_caps.c:145

0x400817a5: heap_caps_malloc_default at /home/esp4/esp/esp-idf-v4.3.2/components/heap/heap_caps.c:177

0x4008f6c9: malloc at /home/esp4/esp/esp-idf-v4.3.2/components/newlib/heap.c:31

0x400849c9: wifi_malloc at /home/esp4/esp/esp-idf-v4.3.2/components/esp_wifi/esp32/esp_adapter.c:79

0x400fe8e9: wDev_SnifferRxData at ??:?

0x40093657: wDev_ProcessRxSucData at ??:?

0x400938e3: wdevProcessRxSucDataAll at ??:?

0x400910de: ppTask at ??:?

0x4008beb9: vPortTaskWrapper at /home/esp4/esp/esp-idf-v4.3.2/components/freertos/port/xtensa/port.c:168

ELF file SHA256: ccb2d1f1ecbef3b4

Rebooting...
```

With comprehensive heap poisoning on, the crash looks like this:

```
I (14504) MAIN: drop 2443, free heap 237356
I (15504) MAIN: drop 2622, free heap 237356
I (16504) MAIN: drop 2801, free heap 237356
CORRUPT HEAP: Bad tail at 0x3ffc4fea. Expected 0xbaad5678 got 0x2dc84124

assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)

Backtrace:0x400815af:0x3ffbf800 0x40088a89:0x3ffbf820 0x4008f57d:0x3ffbf840 0x4008ddeb:0x3ffbf960 0x40081832:0x3ffbf980 0x4008f5ad:0x3ffbf9a0 0x40091d23:0x3ffbf9c0 0x40090f73:0x3ffbf9e0 0x4008bee1:0x3ffbfa10
0x400815af: panic_abort at /home/esp4/esp/esp-idf-v4.3.2/components/esp_system/panic.c:393

0x40088a89: esp_system_abort at /home/esp4/esp/esp-idf-v4.3.2/components/esp_system/system_api.c:112

0x4008f57d: __assert_func at /home/esp4/esp/esp-idf-v4.3.2/components/newlib/assert.c:85

0x4008ddeb: multi_heap_free at /home/esp4/esp/esp-idf-v4.3.2/components/heap/multi_heap_poisoning.c:253 (discriminator 1)

0x40081832: heap_caps_free at /home/esp4/esp/esp-idf-v4.3.2/components/heap/heap_caps.c:305

0x4008f5ad: free at /home/esp4/esp/esp-idf-v4.3.2/components/newlib/heap.c:46

0x40091d23: ppProcessRxPktHdr at ??:?

0x40090f73: ppTask at ??:?

0x4008bee1: vPortTaskWrapper at /home/esp4/esp/esp-idf-v4.3.2/components/freertos/port/xtensa/port.c:168

```

Example code:

```
#include <sys/cdefs.h>
#include "freertos/FreeRTOS.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include "string.h"
#include "freertos/ringbuf.h"
#include "esp_err.h"
#include "esp_now.h"
#define LOG_LOCAL_LEVEL ESP_LOG_VERBOSE
#include "esp_log.h"

#define TAG "MAIN"

int p_drop=0;

void wifi_raw_recv_cb(void *buf,wifi_promiscuous_pkt_type_t type) {
    p_drop++;
}

_Noreturn void wifi_raw_print_stats(void *unused)
{
    while (1) {
        ESP_LOGI(TAG,"drop %i, free heap %i",p_drop,esp_get_free_heap_size());
        vTaskDelay(1000/portTICK_PERIOD_MS);
    }
}

void app_main() {
    const wifi_promiscuous_filter_t filt = {
            .filter_mask=WIFI_PROMIS_FILTER_MASK_ALL
    };
    ESP_LOGI(TAG,"Initializing WiFi");

    nvs_flash_init();

    xTaskCreate(wifi_raw_print_stats,"wifi_p_stats",3000,NULL,0,NULL);

    esp_log_level_set(TAG,ESP_LOG_INFO);

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_FLASH));

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    wifi_config_t sta_config = {
            .sta = {
                    .channel = 11,
            }
    };

    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &sta_config));
    ESP_ERROR_CHECK(esp_wifi_start());
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous(true));
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous_filter(&filt));
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous_rx_cb(&wifi_raw_recv_cb));
    ESP_ERROR_CHECK(esp_wifi_set_channel(11,WIFI_SECOND_CHAN_NONE));

    ESP_LOGI(TAG,"Waiting for packets...");

    vTaskDelay(portMAX_DELAY);

}
```

cmsedore, 2022-01-02:
Just a follow-on that this can be triggered by high packet loads, so it may be a race condition or resource exhaustion rather than a 'special' packet.

AxelLin, 2022-01-02:
I'm wondering if this commit helps 09db8845c692ee5002810744fd1ce889d4d69c9d ?

cmsedore, 2022-01-02:
Thanks for the idea.  I just applied that patch -- Unfortunately, it does **not** resolve this issue.  I still see the same crash with the patch applied.

cmsedore, 2022-01-02:
A little more information -- while the crash still does occur, the patch has changed it.  Now, I can't make it crash with heap poisoning turned on.  The fix I was using to mitigate this was adding:

        busywait=0;
        while (busywait++<5000) {}

inside of a tight loop of request/reply transactions, providing some additional delay between the processing of the last reply and sending out a packet with the next request.  Without this busy wait, the crash is pretty reliable.  I suspect that heap poisoning is providing a similar delay which mitigates the problem.

My production code is basically sending a packet, receiving a reply packet (via a promiscuous listen) repeatedly in a tight loop (about a 5-10ms transaction).  This causes a panic pretty reliably  (note that it can occur almost immediately or downstream a bit -- my longest runs on this loop are about 90 seconds and it makes it through that about 25% of the time without the busy wait).

xueyunfei998, 2022-01-05:
hi @cmsedore 

This problem has been fixed in master and version 4.4, but it has not been backport to version 4.3, so you can test it based on this point, and then this problem will be fixed in version 4.3.

e104dd7f27d2e73ab0e9b614dd7b9295099069bf

AxelLin, 2022-02-15:
@xueyunfei998 
How is the fix status for v4.3 now? (The fix in 4.4 was committed 2 months ago, I think it should not take such long time to fix v4.3)
I cannot understand how the changes in e104dd7, it's a merge commit and I think most of the changes are not related to this issue. Could you point out the real fix commit for this issue?

AxelLin, 2022-02-18:
@cmsedore 
Can you verify this issue with v4.4 release?

xueyunfei998, 2022-02-24:
Crash reason: sniffer received some packets with check errors, and the actual length of received packets was inconsistent with the length of wifi hardware feedback, so there was an error in memcopy, which was solved in 4.4.

AxelLin, 2022-02-24:
@xueyunfei998 
I'm aware it's fixed in v4.4.
My question is if v4.3 needs  fix? (As I said in https://github.com/espressif/esp-idf/issues/8178#issuecomment-1039793902, I don't know if it's fixed in v4.3 or not)

xueyunfei998, 2022-02-25:
hi  @cmsedore 
v4.3 commit :
4b95b8a5c4b3c9d138642c97e7f68e7e0d392a24

AxelLin, 2022-02-25:
> hi @cmsedore v4.3 commit : [4b95b8a](https://github.com/espressif/esp-idf/commit/4b95b8a5c4b3c9d138642c97e7f68e7e0d392a24)

Are you sure it's correct commit?
In v4.4, the fix is https://github.com/espressif/esp32-wifi-lib/commit/657dd399cff195341fd536d503ce99f0d46876a6
The commit log is different from the one you mentioned in v4.3.

cmsedore, 2022-02-25:
@AxelLin This issue seems to be resolved in v4.4.  Thanks!

ToveRumar, 2022-03-03:
> > hi @cmsedore v4.3 commit : [4b95b8a](https://github.com/espressif/esp-idf/commit/4b95b8a5c4b3c9d138642c97e7f68e7e0d392a24)
> 
> Are you sure it's correct commit? In v4.4, the fix is [espressif/esp32-wifi-lib@657dd39](https://github.com/espressif/esp32-wifi-lib/commit/657dd399cff195341fd536d503ce99f0d46876a6) The commit log is different from the one you mentioned in v4.3.

I agree. It looks to be the wrong commit. Could it be confirmed that there is a fix in 4.3, and in that case what commit?

xueyunfei998, 2022-03-03:
hi  @ToveRumar 

I'll check ，If there is no backPort, it will be backported to 4.3 as soon as possible.

ToveRumar, 2022-03-07:
> hi @ToveRumar
> 
> I'll check ，If there is no backPort, it will be backported to 4.3 as soon as possible.

Thank you!

jack0c, 2022-03-10:
The code is backport to IDFv4.3, a20b6c4d95d1029b93b465d4512649b5db65e21c include this fix.

closed 2022-03-10 by Alvin1Zhang