espressif/esp-idf #8181: BLE Mesh node name

MiladMohammadi72, 2022-01-02:
Hi All
according to 
issue #3196 on https://github.com/espressif/esp-idf/issues/3196 and 
issue #6619 on https://github.com/espressif/esp-idf/issues/6619
in ble mesh device we can set name for unprovisioned device by function **esp_ble_mesh_set_unprovisioned_device_name();**
BUT
when a device is provisioned and you want to connect to that (in connect menu of nRF-Mesh app) you can see default node name (ESP-BLE-MESH). show it for device name.
and this mean (I think!) the device uprov and proved have different name 
and there is another function that you don't explain about that in documents
bt_mesh_gatts_set_local_device_name();
this function set default name for nodes -> (ESP-BLE-MESH)
so I can not use **bt_mesh_gatts_set_local_device_name()** this function I think I should do something else
anyone know ???!!!

InfiniteYuan, 2022-06-07:
Could you share the code to set the name?

This [comment](https://github.com/espressif/esp-idf/issues/3196#issuecomment-546322393) should be noted.

hipal31, 2022-08-24:
Try `esp_ble_mesh_set_unprovisioned_device_name()`  after esp_ble_mesh_init() . 
this worked for me

closed 2025-03-15 by Alvin1Zhang