espressif/esp-idf #8182: issue with using PSRAM after HIMEM API

asafs932, 2022-01-02:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): v4.1
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
- Operating System:     [Linux]
- (Windows only) environment type: [].
- Using an IDE?: [clion]
- Power Supply:         [USB

## Problem Description

Hi, my use case is that im doing the following 3 times in a row (without shutting down the esp) -  i acquire & save new image to the PSRAM and then extract it from there (using regular PSRAM or HIMEM operations) and send it to a server
I encountered an issue of corrupted image after using HIMEM API (for images bigger than 4mb) successfully.
for example:
image index one - 3mb, using regular extraction from psram and image looks good
image index two - **5mb**, using HIMEM and image looks good
image index three - 3mb, using regular extraction from psram and **image is corrupted**

in cases where i dont need to use the HIMEM (if all images < 4mb) everything is ok.

i saw this Note about possible issue related to it here - 
https://github.com/arkhipenko/ESP32Himem

[image]

### Expected Behavior
i expected the image to be ok, but for some reason after using the HIMEM API it seems that the PSRAM location are messed up (maybe offset or something)
### Actual Behavior

the bottom part of the image is grayed and the top part is ok - which leads me to believe that either the deallocations after the HIMEM API operation didnt went well or something needs to be initialized in the PSRAM again.

### Steps to reproduce

1. calling the himem operations
2. try to extract data from psram - some data will be missing

## Debug Logs

**boot logs**:

I (31) boot: ESP-IDF v4.1 2nd stage bootloader
I (31) boot: compile time 14:49:30
I (31) boot: chip revision: 1
I (34) qio_mode: Enabling QIO for flash chip ISSI
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : QIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
..
..
..
I (624) esp_image: segment 5: paddr=0x0016b53c vaddr=0x40080404 size=0x1cd80 (118144) load
I (700) boot: Loaded app from partition at offset 0x30000
I (700) boot: Disabling RNG early entropy source...
I (701) psram: This chip is ESP32-D0WD
I (705) spiram: Found 64MBit SPI RAM device
I (710) spiram: SPI RAM mode: flash 40m sram 40m
I (715) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
...
..
I (0) cpu_start: App cpu up.
I (1219) spiram: SPI SRAM memory test OK
I (1220) heap_init: Initializing. RAM available for dynamic allocation:
I (1220) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1226) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1232) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1239) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (1245) heap_init: At 3FFCEE58 len 000111A8 (68 KiB): DRAM
I (1251) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1257) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1264) heap_init: At 4009D184 len 00002E7C (11 KiB): IRAM
I (1270) cpu_start: Pro cpu start user code
I (1275) spiram: Adding pool of 2112K of external SPI memory to heap allocator
I (1295) esp_himem: Initialized. Using last 62 32KB address blocks for bank switching on 6080 KB of physical memory.
I (1297) spi_flash: detected chip: issi
I (1301) spi_flash: flash io: qio
I (1305) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1315) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations

**printing the following during runtime**:

printf("spiram size %u\n", esp_spiram_get_size());
printf("himem free %u\n", esp_himem_get_free_size());
printf("himem phys %u\n", esp_himem_get_phys_size());
printf("himem reserved %u\n", esp_himem_reserved_area_size());
    
**output**:
spiram size 8388608
himem free 6225920
himem phys 6225920
himem reserved 2031616

negativekelvin, 2022-01-02:
>I (1275) spiram: Adding pool of 2112K of external SPI memory to heap allocator

You only have 2mb of non-himem

asafs932, 2022-01-03:
> > I (1275) spiram: Adding pool of 2112K of external SPI memory to heap allocator
> 
> You only have 2mb of non-himem

so you mean when i have data larger than 2m i should use the himem API?
then how can i increase the PSRAM memory usage to 4mb? i understood it is by default...

negativekelvin, 2022-01-03:
https://github.com/espressif/esp-idf/blob/23a03a3bdbaab1fe0ba427dcfcc82ef391cc06f4/components/esp32/Kconfig#L317-L328

asafs932, 2022-01-04:
problem was indeed related to this noted:
```
         Note that this reservation is only actually done if your program actually uses the himem API. Without 
         any himem calls, the reservation is not done and the original amount of memory will be available 
         to malloc/esp_heap_alloc_caps.
```
so i set my psram to only direct access 2mb and the rest via Himem.
thanks @negativekelvin

closed 2022-01-07 by igrr