espressif/esp-idf #8183: arp and nslookup equivalent commands in esp-idf

twlawrenceko, 2022-01-02:
Hello,

I have looked through the provided examples but have not seen one that is equivalent to the command `arp -a` and `nslookup` command to print out all devices connected to the same Wi-Fi as the ESP device. Is there any way to retrieve local devices IP address, MAC address, and hostname of each device?

Thank you.

mhdong, 2022-01-24:
hi @twlawrenceko  You can use the following two APIs to get MAC address and IP.

esp_wifi.h
/**
  * @brief     Get STAs associated with soft-AP
  *
  * @attention SSC only API
  *
  * @param[out] sta  station list
  *             ap can get the connected sta's phy mode info through the struct member
  *             phy_11b，phy_11g，phy_11n，phy_lr in the wifi_sta_info_t struct.
  *             For example, phy_11b = 1 imply that sta support 802.11b mode
  *
  * @return
  *    - ESP_OK: succeed
  *    - ESP_ERR_WIFI_NOT_INIT: WiFi is not initialized by esp_wifi_init
  *    - ESP_ERR_INVALID_ARG: invalid argument
  *    - ESP_ERR_WIFI_MODE: WiFi mode is wrong
  *    - ESP_ERR_WIFI_CONN: WiFi internal error, the station/soft-AP control block is invalid
  */
esp_err_t esp_wifi_ap_get_sta_list(wifi_sta_list_t *sta);

tcpip_adapter.h
/**
 * @brief Compatible version of former tcpip_adapter API of esp_netif_get_sta_list
*/
esp_err_t tcpip_adapter_get_sta_list(const wifi_sta_list_t *wifi_sta_list, tcpip_adapter_sta_list_t *tcpip_sta_list);

negativekelvin, 2022-01-24:
@mhdong I think he wants to do subnet scan of the router when esp is station so those apis won't help

twlawrenceko, 2022-01-28:
Hello @mhdong ,
Yes, @negativekelvin is right, I would like to do a subnet scan of the router when esp is station and to check whether a phone device is in the subnet or not.

xueyunfei998, 2022-03-21:
[esp_arp.txt](https://github.com/espressif/esp-idf/files/8313614/esp_arp.txt)
[lwip_arp.txt](https://github.com/espressif/esp-idf/files/8313615/lwip_arp.txt)

xueyunfei998, 2022-03-21:
hi @twlawrenceko 

I added an interface to query arp table records.

nslookup is used to query the records of DNS and whether the domain name resolution is normal.

getaddrinfo  this interface can be used in idf, this interface queries the ip address corresponding to the domain name.

xueyunfei998, 2022-03-22:
@negativekelvin 

His demand should be to query the records of local arp table.
There is also a domain name query to see if the network is unobstructed.
Right? @twlawrenceko

twlawrenceko, 2022-03-22:
@xueyunfei998 
What we would like to achieve is the detect whether a device is now connected or not. For example, if the computer wakes up and connects to the same Wi-Fi as ESP, then the ESP can detect that. 
With that idea, I think the local arp table can be first scanned and then how to trigger ESP to detect whether a device becomes online or not is the question we would like to ask.
Of course we can continuous using arp scan, but as @negativekelvin  mentioned, it may take too long.

xueyunfei998, 2022-03-24:
hi @twlawrenceko 

The meaning expressed in all your statements above：

If esp32 is a sta mode connection router, then you think esp32 can scan all the devices in the same network segment under this router.

Is that what you mean?

twlawrenceko, 2022-03-25:
@xueyunfei998 
I don’t mean by 100%. I mean by if a device was originally detected (via arp or such), then it can detect whether it is currently online or not. 

Or the other way, if the device IP is provided, then the ESP can detect whether it’s online now or not by either arp scan to see if it’s in table or ping.

xueyunfei998, 2022-03-25:
@twlawrenceko 

ok，Now that I know what you need.

Whether the mode of esp32 is ap or sta, esp32 wants to know which devices are in a network segment and whether these devices are online.

First, if you want to know whether the device is online,  need the device to push online messages in real time.

If this is the case, it is better to make a heartbeat packet real-time broadcast device online in the application layer, instead of scanning arp table.

The application layer heartbeat packet, preferably uses udp communication to broadcast the information of the device.

It is difficult to know whether the device is online from wifi level or lwip level, because although arp table knows the information of other nodes, it is not effective, and arp table will not keep the information if it does not communicate with other nodes.

Therefore, it is recommended to redefine a private protocol in the application layer to complete this function.

negativekelvin, 2022-03-25:
@xueyunfei998 sure mdns or custom protocol could work but user does not always control the target devices.  So you are saying there is no way to send and receive arp packets from application layer?

xueyunfei998, 2022-03-28:
@negativekelvin 

Because arp is an ip layer protocol, the application layer is not easy to operate through socket.

However, esp32 has a function Garp, which is that esp32 broadcasts address information through ARP every 60 seconds (this time can be configured)

You can reduce this time, let esp32 devices frequently broadcast their address information, and then check arp table to complete this function.

negativekelvin, 2022-03-28:
@xueyunfei998 as mentioned before other devices may not be esp32 based or configurable by user.  Arp table size can be limiting factor too.  What about installing hook at netif layer to inject and capture arp packets?

xueyunfei998, 2022-04-25:
hi @negativekelvin 

Check whether the device is online through arp function, that is, receive the arp requset packet of the other device.

If other devices are not developed based on esp32, they may not have free arp function. If the device does not send data, it may not send arp requset.

You mentioned above to add an arp hook, which is a new function. I can add a function when receiving arp packets, and return the ip address and mac address of the other device. You can judge whether the other device is online by the returned address. Do you think it's ok to do so?

negativekelvin, 2022-04-25:
Need to be able to send arp requests from esp32 ("who has 192.168.0.2") and receive the arp replies with Mac address.

xueyunfei998, 2022-05-24:
hi @negativekelvin 
Need to be able to send arp requests from esp32 ("who has 192.168.0.2"):

- Esp32 has a free arp function, which can send arp_request periodically.

receive the arp replies with Mac address:

- I can add a callback interface when lwip handles arp request, and you can implement this callback yourself. Do you think it's ok?

negativekelvin, 2022-05-24:
@xueyunfei998 yes call back is good. Can you add example or unit test code for sending arp request and receiving response from callback?

featherfly, 2022-10-17:
@xueyunfei998 the feature arp_request is ok now