espressif/esp-idf #8184: ESP32 reset by RTC watchdog although RTX WDT is disabled in sdkconfig

MikitaMinau, 2022-01-03:
## Environment

- Development Kit: [none]: custom board
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Module or chip used: [ESP32]
- IDF version: v4.3, v4.0
- Build System: [idf.py]
- Compiler version:  8.4.0
- Operating System: [Windows]
- (Windows only) environment type: [Plain Command Prompt].
- Using an IDE?: [No]
- Power Supply: [USB]

## Problem Description

Sometimes, I see that ESP32 is reset without any reason. No stack trace is observed during the reset. After the reset, esp_reset_reason() returns 7(ESP_RST_WDT) and rtc_get_reset_reason(PRO_CPU_NUM) returns 16(RTCWDT_RTC_RESET):
```
I (13471) KernelInterface.c: ESP reset reason is 7
I (13471) KernelInterface.c: RTC reset reason is 16
```

Looking into esp-idf source code I came to a conclusion that the reason for the RTC watchdog. However, RTC watchdog is enabled only for bootloader and disabled in the application code(look at attached sdkconfig file). 

I suspect automatic light sleep mode because I see that RTC WTD is enabled before going to light sleep but I did find information on haw RTC watchdog is fed in light sleep. Light sleep code says that it enables watchdog in case exit from light sleep fails but did not describe how may it fail:
```
// Safety net: enable WDT in case exit from light sleep fails
    wdt_hal_context_t rtc_wdt_ctx = {.inst = WDT_RWDT, .rwdt_dev = &RTCCNTL};
    bool wdt_was_enabled = wdt_hal_is_enabled(&rtc_wdt_ctx);    // If WDT was enabled in the user code, then do not change it here.
    if (!wdt_was_enabled) {
        wdt_hal_init(&rtc_wdt_ctx, WDT_RWDT, 0, false);
        uint32_t stage_timeout_ticks = (uint32_t)(1000ULL * rtc_clk_slow_freq_get_hz() / 1000ULL);
        wdt_hal_write_protect_disable(&rtc_wdt_ctx);
        wdt_hal_config_stage(&rtc_wdt_ctx, WDT_STAGE0, stage_timeout_ticks, WDT_STAGE_ACTION_RESET_RTC);
        wdt_hal_enable(&rtc_wdt_ctx);
        wdt_hal_write_protect_enable(&rtc_wdt_ctx);
    }
```

### Expected Behavior

ESP32 is not reset by RTC watchdog when it is disabled.

### Actual Behavior

ESP32 is reset by RTC watchdog when it is disabled.

### Steps to reproduce

Usually, I see this kind of reset within 12-24 hours of work. BLE and WiFi are enabled. Automatic light sleep and DFS are also enabled.

### Code to reproduce this issue

Unfortunately, I cannot share the code (also, it requires custom HW and won't run on devkits).

## Debug Logs

```

I (72871) BatteryDriver.c: Battery level is 100% (5.352V)
I (73431) ThingShadowService.c: Shadow update accepted
I (74231) ThingShadowService.c: Shadow update completed
I (75231) ThingShadowService.c: Updating the reported section
I (75231) ThingShadowService.c: Update Shadow
I (78031) ThingShadowService.c: Shadow update accepted
I (78841) ThingShadowService.c: Shadow update completed
I (85001) wagz_main.c: Timer event!
I (85001) wagz_main.c: Current date is Thu Dec 16 11:41:16 2021
I (85001) wagz_main.c: Assigned Pet ID: 3077
I (85001) KernelInterface.c: Heap size CAP 8bit 3913510
I (85011) KernelInterface.c: Heap size CAP 32bit 3917574
[... 221 lines cut ...]
I (18461) GSM_Commands.c: Get the modem FW version command...
I (18581) GSM_States.c: The modem FW version is: BG77LAR02A04_01.005.01.005
I (18581) HwInfoStorage.c: Modem FW version is: BG77LAR02A04_01.005.01.005
I (18581) GSM_Commands.c: Cell Modem tests gpsOneXTRA Assistance function...
I (18711) GSM_Commands.c: Cell Modem tests injecting of gpsOneXTRA time...
I (18841) GSM_Commands.c: Cell Modem tests injecting of gpsOneXTRA data...
I (18971) GSM_Commands.c: Cell Modem query the status of gpsOneXTRA Assistance function...
I (19091) GSM_States.c: Cell modem GPS has one xtra feature enabled
I (19091) GSM_Commands.c: Query GNSS constellation configuration...
I (19151) GSM_Commands.c: +QGPSCFG: "gnssconfig",1
I (19151) GSM_Commands.c: Configured GNSS constellation is 1
I (19211) GSM_States.c: GNSS is already configured for GPS+GLONASS constellation
I (19211) GSM_Commands.c: Cell Modem enable IoT mode...
I (19271) GSM_Commands.c: eMTC IoT mode is enabled
I (19331) GSM_Commands.c: Disabling GNSS NMEA output port...
I (19451) GSM_Commands.c: Enabling NMEA acquisition using AT commands ...
I (19571) GSM_Director.c: GSM Modem configured.
I (19571) GSM_Commands.c: Get the modem FW version command...
I (19691) CellOta.c: Running LTE modem FW version is: BG77LAR02A04_01.005.01.005
I (19691) GSM_States.c: Check SIM Card...
I (19691) GSM_Commands.c: Check pin code command...
```

## Other items if possible
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7803028/sdkconfig.txt)

stopthatcow, 2023-01-20:
I am also seeing this occasionally on a ESP32-WROVER using automatic light sleep.

bearsh, 2023-04-05:
this also happens in my application on some modules. I configure pm and enable light sleep in app_main then I start various tasks and also enable wifi in station mode connecting to a remote.
it seems like when the wifi connection is established very fast, the reset occurs. if it takes some time for the wifi connection to get up, everything just works fine.

if I disable the wdt in https://github.com/espressif/esp-idf/blob/4c7d97e2bdbd26b1ad6adc6de8051888e1feec10/components/esp_hw_support/sleep_modes.c#L776-L779
by setting `wdt_was_enabled = pdTRUE` everything seems to work fine.

expresspotato, 2023-04-28:
If its running on a custom board, you might want to check your matching circuit or configure  `CONFIG_ESP32_RTC_CLK_CAL_CYCLES` to a higher value, say 8192. I had an issue with a matching circuit and hand solder job that was difficult to pin down. It's what led me to this thread with seemingly random `RTCWDT_RTC_RESET` errors.