espressif/esp-idf #8185: coded phy connectable mode

iamcalledusman, 2022-01-03:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [esp32-c3-mini-1 development board]
- Kit version (esp32-c3-mini-1 development board): [v1]
- Module or chip used:  [esp32-c3-mini-1 development board]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell(ESP-IDF 4.3)].
- Using an IDE?: [Yes (VS Code Studio)]
- Power Supply:         [USB]

## Problem Description

I am  using coded PHY using esp-idf examples (Periodic_adv/ Multi-adv as a Transmitter) & (Periodic_sync /BLE50_Security_Client as a Receiver).

The advertising type used for coded PHY:   "ESP_BLE_GAP_SET_EXT_ADV_PROP_SCANNABLE".

This is non-connectable and scannable. But when we set advertising type as per our requirement which is connectable

"ESP_BLE_GAP_SET_EXT_ADV_PROP_CONNECTABLE" generates errors at run time with coded PHY.
 
[image]

//Detailed problem description goes here.

### Expected Behavior

It should not generate Run time error as shown in image above "cmd_err =0x12" . It should connect with the Andriod App (nRF Connect)

### Actual Behavior

Generating Run time error as shown in image above "cmd_err =0x12" .And unable to connect with the Andriod App (nRF Connect). Although when i set primary phy and Secondary phy as "ESP_BLE_GAP_PHY_1M". It get connected with the andriod app.

### Steps to reproduce

1. Need two esp32c3-Mini development kits
2. Load the period_adv example code in first kit with some changes.
a) Configure the ext_adv_params_coded 
b) Change the .type = ESP_BLE_GAP_SET_EXT_ADV_PROP_CONNECTABLE
c)Primary and secondary phy both set to "ESP_BLE_GAP_PHY_CODED"
d)In app_main () Comment all the advertsings only uncomment/paste these lines 
   
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_params(0, &ext_adv_params_coded), test_sem);

    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_rand_addr(0, addr_coded), test_sem);

    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_ext_scan_rsp_data_raw(0, sizeof(raw_scan_rsp_data_coded), &raw_scan_rsp_data_coded[0]), test_sem);

    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_start(1, &ext_adv[0]), test_sem);
e)Compile and load the code into the first development esp32c3 kit.
3. Load the ble50_security_client example code in Second kit.
4. You will see the expected output on ESP-IDF monitor Device.

// If possible, attach a picture of your setup/wiring here.

No wiring required just to connect the two development boards with the PC using the USB cable 

### Code to reproduce this issue

// Replace the following configuration .
esp_ble_gap_ext_adv_params_t ext_adv_params_coded = {
    .type = ESP_BLE_GAP_SET_EXT_ADV_PROP_CONNECTABLE,
    .interval_min = 0x50,
    .interval_max = 0x50,
    .channel_map = ADV_CHNL_ALL,
    .filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
    .primary_phy = ESP_BLE_GAP_PHY_CODED,
    .max_skip = 0,
    .secondary_phy = ESP_BLE_GAP_PHY_CODED,
    .sid = 3,
    .scan_req_notif = false,
    .own_addr_type = BLE_ADDR_TYPE_RANDOM,
};
Add these lines to in app_main (){}

/* FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_params(0, &ext_adv_params_1M), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_rand_addr(0, addr_1m), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_ext_adv_data_raw(0, sizeof(raw_adv_data_1m), &raw_adv_data_1m[0]), test_sem);

    // 2M phy extend adv, Scannable advertising
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_params(1, &ext_adv_params_2M), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_rand_addr(1, addr_2m), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_ext_scan_rsp_data_raw(1, sizeof(raw_scan_rsp_data_2m), raw_scan_rsp_data_2m), test_sem);

    // 1M phy legacy adv, ADV_IND
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_params(2, &legacy_adv_params), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_rand_addr(2, addr_legacy), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_ext_adv_data_raw(2, sizeof(legacy_adv_data), &legacy_adv_data[0]), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_ext_scan_rsp_data_raw(2, sizeof(legacy_scan_rsp_data), &legacy_scan_rsp_data[0]), test_sem);
   */
    // coded phy extend adv, Connectable advertising
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_params(0, &ext_adv_params_coded), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_rand_addr(0, addr_coded), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_ext_scan_rsp_data_raw(0, sizeof(raw_scan_rsp_data_coded), &raw_scan_rsp_data_coded[0]), test_sem);

    // start all adv
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_start(1, &ext_adv[0]), test_sem);

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

darkmark88, 2023-09-20:
I had the exact same problem as above, the only difference was that I used an s3 chip

esp-cjh, 2024-02-27:
Hi, @iamcalledusman @darkmark88 
For extended connectable advertising, it's not supported to set the scan response data. So you should invoke `esp_ble_gap_config_ext_adv_data_raw()` to set advertising data instead of `esp_ble_gap_config_ext_scan_rsp_data_raw()`.

closed 2024-12-17 by Alvin1Zhang