espressif/esp-idf #8186: How do I include azure IOT over ethernet on ESP32 ?

enthunilu, 2022-01-03:
Development Kit: ESP32-Ethernet kit_A_V1.1
Kit version
Module or chip used: ESP32
IDF version :v4.2.1
Build System: Make|CMake|idf.py
Compiler version : xtensa-esp32-elf-gcc (crosstool NCG esp-2020r3)8.4.0
Operating System: Windows
(Windows only) environment type: Command Prompt
Using an IDE?: Yes (Eclipse)

Power Supply: USB

Problem Description
## Problem Description
I am trying for IOT over ethernet using ethernet example code and IOT azure code.
I could not get any direct example code for the same so I tried to use existing ethernet basic example code and IOT example code.
However functionality point of view it is not working( no connection or device tree logs are getting printed ). 
So, can anyone suggest link or documentation or any example code for IOT over ethernet?

Thanks !!

0xjakob, 2022-01-05:
@enthunilu Do you have an IOT Azure example working over WiFi already?

enthunilu, 2022-01-05:
Hi @0xjakob yes it is working with wifi. 
Looks like something is missing in ethernet part which I am unable to figure out what it is.

enthunilu, 2022-02-04:
I got the example code for reference.
ethernet_example code available in the esp-idf\examples\ethernet\basic\main.

So, now closing this ticket.

Thanks!!

closed 2022-02-04 by enthunilu