espressif/esp-idf #8187: ESP32 BLE micropython

josmet52, 2022-01-03:
HI,
I have one issue with micropython-ble between 2 ESP32-WROOM-32

## Environment

- Development Kit:      WEMOS D1 Mini
- Module or chip used:  ESP32-WROOM-32
- Firmware : v1.17 (2021-09-02) .bin
- Operating System:     Windows
- environment type: laptop
- IDE: Thonny
- Power Supply: USB

## Problem Description
Error message : "lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)"

### Detailed problem 
I have one issue with micropython-ble between 2 ESP32-WROOM-32 :
the bluetooth connection is ok, all run well but after n good cycles (n is variable) the error mesage appears and the system is down.
### Steps to reproduce

1. run the script micropython *jmb_uart_central.py* on one ESP32
2. run the script micropython *jmb_uart_sensor.py* on other ESP32
3. after n cycles the error message will apears on the *jmb_uart_sensor.py* running script --> "lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)"
4. the system is down

### Files
[micropython files.zip](https://github.com/espressif/esp-idf/files/7804412/micropython.files.zip)

josmet52, 2022-01-03:
please use the new zip, a file name error in the othe
[micropython files.zip](https://github.com/espressif/esp-idf/files/7804470/micropython.files.zip)

Weijian-Espressif, 2022-03-30:
@rahult-github  Please take a look at this

ocESP, 2022-06-29:
@josmet52, 

Could you let us know "n good cycles" of what? I mean are you doing connect/disconnect for multiple times, or are you performing read/write for multiple times or something else? 

Also, please let us know the reproduction rate, i.e. how often are you able to see the issue. Is it rarely seen or frequently seen?

Moreover, please share the console/terminal logs when the issue occurs.

rahult-github, 2023-08-22:
Closing issue due to no response. Please feel free to reopen with logs.

closed 2023-08-22 by rahult-github