espressif/esp-idf #8189: smartconfig: setsockopt failed,Unable to notify phone

ForgetCSX, 2022-01-04:
- Development Kit: NULL
- Kit version : master
- Module or chip used:  ESP32module and esp32s2module
- IDF version : ESP-IDF v5.0-dev-961-g733fbd9ecc
- Build System:  idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:   Linux
- Power Supply:   USB

## Problem Description
Using the "smartconfig" process, the "E (29386) smartconfig: setsockopt failed" error report appears, and the mobile phone cannot be notified of successful network distribution
### Expected Behavior
After obtaining the IP, espTouch will prompt the network to be successful and display the mac address of the device
### Actual Behavior
The network configuration failed, but in fact the device has acquired the IP
### Steps to reproduce

1. copy smartconfig instance
2. cd smartconfig 
3.  get_idf
4. idf.py set-target esp32s2
5. idf.py build
6.  idf.py flash monitor

### Code to reproduce this issue

smartconfig instance

## Debug Logs

```
xiang@ubuntu:~/esp32/work/smart_config$ idf.py flash monitor 
Executing action: flash
Serial port /dev/ttyUSB0
Connecting..........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Running ninja in directory /home/xiang/esp32/work/smart_config/build
Executing "ninja flash"...
[1/5] cd /home/xiang/esp32/work/smart_config/build/esp-idf/esptool_py &&...ion-table.bin /home/xiang/esp32/work/smart_config/build/smart_config.bin
smart_config.bin binary size 0x9c770 bytes. Smallest app partition is 0x100000 bytes. 0x63890 bytes (39%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /home/xiang/esp32/work/smart_config/build/bootloader/esp-idf/esptool_py && /home/xiang/.espressif/python_env/idf5.0_py3.6_env/bin/python /home/xiang/esp32/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/xiang/esp32/work/smart_config/build/bootloader/bootloader.bin
Bootloader binary size 0x50f0 bytes. 0x1f10 bytes (28%) free.
[2/3] cd /home/xiang/esp32/esp-idf/components/esptool_py && /snap/cmake/...-P /home/xiang/esp32/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py esp32s2 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 smart_config.bin 0x8000 partition_table/partition-table.bin
esptool.py v3.3-dev
Serial port /dev/ttyUSB0
Connecting..........
Chip is ESP32-S2
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 84:f7:03:71:63:d6
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00006fff...
Flash will be erased from 0x00010000 to 0x000acfff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 20720 bytes to 12949...
Writing at 0x00001000... (100 %)
Wrote 20720 bytes (12949 compressed) at 0x00001000 in 0.6 seconds (effective 260.1 kbit/s)...
Hash of data verified.
Compressed 640880 bytes to 411108...
Writing at 0x00010000... (3 %)
Writing at 0x0001c082... (7 %)
Writing at 0x00027181... (11 %)
[... 119 lines cut ...]
I (14436) smartconfig: TYPE: ESPTOUCH
I (14436) smartconfig: T|AP MAC: 20:76:93:4c:0c:86
I (14436) smartconfig: Found channel on 12-2. Start to get ssid and password...
I (14436) smartconfig_example: Found channel
I (28356) smartconfig: T|pswd: 123456789
I (28356) smartconfig: T|ssid: wifi-YF
I (28366) smartconfig: T|bssid: 20:76:93:4c:0c:86
I (28366) wifi:ic_disable_sniffer
I (28366) smartconfig_example: Got SSID and password
I (28366) smartconfig_example: SSID:wifi-YF
I (28376) smartconfig_example: PASSWORD:123456789
I (28466) wifi:new:<12,2>, old:<12,2>, ap:<255,255>, sta:<12,2>, prof:6
I (28466) wifi:state: init -> auth (b0)
I (28466) wifi:state: auth -> assoc (0)
I (28466) wifi:state: assoc -> run (10)
I (28576) wifi:connected with wifi-YF, aid = 3, channel 12, 40D, bssid = 20:76:93:4c:0c:86
I (28586) wifi:security: WPA2-PSK, phy: bgn, rssi: -64
I (28586) wifi:pm start, type: 1

W (28596) wifi:<ba-add>idx:0 (ifx:0, 20:76:93:4c:0c:86), tid:0, ssn:3, winSize:64
I (28666) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (29356) esp_netif_handlers: sta ip: 192.168.123.232, mask: 255.255.255.0, gw: 192.168.123.1
I (29356) smartconfig_example: WiFi Connected to ap
E (29386) smartconfig: setsockopt failed
```

ForgetCSX, 2022-01-04:
Just tested it, switching to the v4.4 branch is the same phenomenon

ESP-YJM, 2022-01-04:
@ForgetCSX Could you please change https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/src/smartconfig_ack.c#L124-L127 like bleow.
```
if (setsockopt(send_sock, SOL_SOCKET, SO_BROADCAST, &optval, sizeof(int)) < 0) {
    ESP_LOGE(TAG,  "setsockopt SO_BROADCAST failed");
    goto _end;
}
if (setsockopt(send_sock, SOL_SOCKET, SO_REUSEADDR, &optval, sizeof(int)) < 0) {
    ESP_LOGE(TAG,  "setsockopt SO_REUSEADDR failed");
    goto _end;
}
```

ForgetCSX, 2022-01-04:
Hi@ESP
[image]
-YJM Thanks, the problem has been solved

closed 2022-01-10 by ForgetCSX

AxelLin, 2022-01-25:
@ESP-YJM smartconfig is still broken in v4.3.

ESP-YJM, 2022-01-25:
@AxelLin Thanks for reminding me. We have already created backport MR internal. Maybe need some time to  sync to github.

mbratch, 2022-03-10:
Was this fixed in the latest v4.4 release? I did not have this issue in v4.3, but I recently upgraded to the v4.4 stable release and now have this problem.

ESP-YJM, 2022-03-10:
@mbratch Yes, it has been fixed in release/v4.4 https://github.com/espressif/esp-idf/commit/9cf57a86a0cb5d03bf37915cce113ddfcdcb61a1

mbratch, 2022-03-10:
@ESP-YJM  ok thank you.

ESP-YJM, 2022-03-10:
@mbratch The release/v4.4 branch will always have new features and fixes merged in. And the v4.4 tag was stable and will not be add any features and fixed. The v4.4 tag depends on release/v4.4. Now release/v4.4 has some new commits, we can develiver a new stable tag version, maybe named v4.4.1. And after than, some new commits megered in release/v4.4, we can develiver a new stable tag version, named v4.4.2.

mbratch, 2022-03-10:
@ESP-YJM thank you, that makes a lot of sense.