espressif/esp-idf #8191: Allow custom mock-components instead of built-in Cmock

nathanRamaNoodles, 2022-01-04:
**Is your feature request related to a problem? Please describe.**
&emsp; Regarding unit testing on Linux, I liked the feature in the [linux_host_app](https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/linux_host_app) when it built my own mock components.  However, now the app builds all components that I don't need to be mocked.

**Describe the solution you'd like**
&emsp; Use an environmental variable like `DISABLE_BUILT_IN_MOCK` and setting it to `0` would keep the same settings as in the current esp-idf.  Setting it to `1` will comment out the following line in [esp-idf/tools/cmake/build.cmake](https://github.com/espressif/esp-idf/blob/master/tools/cmake/build.cmake#L164):
```
    if("${target}" STREQUAL "linux")
        set(requires_common freertos log esp_rom esp_common)
        # idf_build_set_property(__COMPONENT_REQUIRES_COMMON "${requires_common}")
    else()
```

**Describe alternatives you've considered**
&emsp; I would manually comment out the `idf_build_set_property` line, which works for now.

0xjakob, 2022-01-24:
@nathanRamaNoodles Any further comment? Any reason why you can't leverage the component overriding mechanism?

To further clarify the statements above: `log`, `esp_rom`, `esp_common` all have a Linux implementation. They are not mocked unless you create your own mock implementation. `freertos` has to be mocked unless you provide your own mock implementation. For `freertos`, there is no Linux implementation at the moment.

nathanRamaNoodles, 2022-01-25:
I have created a repo which includes my custom component: [nathanRamaNoodles/linux_freertos_test](https://github.com/nathanRamaNoodles/linux_freertos_test).
You can build the code with my [modification](https://github.com/espressif/esp-idf/issues/8191#issue-1093474228), and it will compile and run.  
You should see this in the terminal:
```Timer Reset
Timer Reset
Timer Reset
Timer Reset
Timer Reset
Heartbeat
...
```
Without the modification, you will run into issues because KCONFIG will raise an error:
```
line 2970, in _parse_block
    raise KconfigError
```
So, I know `freertos` is not implemented at the moment, but I don't know why it won't allow me to use my own implementation called `freertos_linux`.

0xjakob, 2022-01-25:
@nathanRamaNoodles Now that sounds interesting! Nice work regarding the porting of FreeRTOS on Linux. Did you use the "official" FreeRTOS Linux simulator?

I created a PR to illustrate what I mean by "overriding" the component as stated above. It works fine on my machine. Would it be OK for you to use it that way? If you need to access the original directory, it's still possible via IDF Cmake build commands.

nathanRamaNoodles, 2022-01-25:
The name change to `freertos` fixed it, but now I need to avoid the component from being compiled when `set-target = esp32`.  Is there a way around this?
> Did you use the "official" FreeRTOS Linux simulator?

I used the [FreeRTOS-Kernel](https://github.com/FreeRTOS/FreeRTOS-Kernel), and then I manually adjusted it to fit the esp32 port: [ThirdParty/Posix](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/ac2c383bc14b2577101cad238c8779f4d9c14d6c/portable/ThirdParty/GCC/Posix)

0xjakob, 2022-01-26:
> now I need to avoid the component from being compiled when set-target = esp32

I see. The component overrides are mainly good for unit tests or permanent overrides, but not for cases where you need to seamlessly switch between targets. The latter is a use case, though, we'll work on that and I'll keep you updated here.

The original idea we had was to simply enable FreeRTOS for Linux in IDF itself to seamlessly allow switching between Linux and esp-idf targets for high-level applications. We just haven't done that so far.

0xjakob, 2022-10-11:
@nathanRamaNoodles Sorry for the slow turn-around! We finally integrated the FreeRTOS on Linux/Posix simulator into IDF in current master. It should work with the main FreeRTOS features, but many IDF-specific features such as esp_timer aren't implemented yet.

Regarding your problem:
> but now I need to avoid the component from being compiled when set-target = esp32

The way we usually test on host in IDF is creating a separate test application that overrides components that should be mocked with their respective mock. The test application would only ever run on the `linux` target. This is done in the [`esp_event` host test](https://github.com/espressif/esp-idf/tree/master/components/esp_event/host_test/esp_event_unit_test), e.g. This application overrides the `freertos` component, but the original `esp_event` component has relatively few adjustments and these adjustments are temporary until we have a better integration of the linux target. Hence, the component `esp_event` can be compiled for both `esp32` or the `linux` target.

Would this way of testing be suitable for you?

nathanRamaNoodles, 2022-10-11:
Hi @0xjakob,

Hearing that seamless mock unit-testing is working is good to hear. However, integration tests are important so that I can catch any bugs when my program has thread locks or memory leaks.  I think making a shell script to simply hide the component might be okay in this case?

0xjakob, 2022-11-01:
Hello @nathanRamaNoodles, sorry for the slow turnaround!

Doing integration tests should work on the chip as well as on linux now, with the limitation that only some IDF components are supported on Linux (hence cmake setting a different common component set, as described above). E.g., if you have an application that only uses FreeRTOS or the other supported components, you should be able to do that. We have a [linux compatible hello world example](https://github.com/espressif/esp-idf/tree/master/tools/test_apps/linux_compatible/hello_world_linux_compatible).

0xjakob, 2023-04-06:
@nathanRamaNoodles Please let us know if and where the current `linux` target does not suit your needs. The `linux`-target independent of mocking, it rather allows developers to write and test high-level IDF applications on a Linux machine, then switch to an ESP chip.

0xjakob, 2023-04-11:
We are closing the issue for now. If you find that the [POSIX/Linux simulator](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/host-apps.html#posix-linux-simulator-approach) does not work for you, feel free to re-open it again.

closed 2023-04-11 by 0xjakob