espressif/esp-idf #8192: ESP32 STA deauth frame causing reconnect issues

tablatronix, 2022-01-04:
## Environment

- Development Kit:   Custom
- Kit version NA
- Module or chip used:  ESP32-WROOM-32D

I am not yet up and running with an IDF example, as this was discovered downstream in arduino.
I finally had time to look into this long time issue and am trying to identify/resolve the 2-3 issues this is encompassing.
https://github.com/espressif/arduino-esp32/issues/2501

## Problem Description

### The Frame in question

This change was made sometime in late 3.x SDK I think. I am assuming this is maybe from a CVE patch , but I cannot find the exact documentation for this, maybe someone can tip me off, most of the dos/injections are against enterprise WPA2, not WPA2 PSK shrug.. I am not wifi expert.

I am hoping PMF/WPA3 solves this.. I have not been able to confirm that yet.

The immediate issue is that this causes some APs to (within a timeout period of last assoc, so device resets mostly!) respond to these sta deauths with a 

[image]
[image]

### The router response
This is my unifi nanohd ap sending a 

```php
Nov 20 22:24:01 nanohd f492bf7f354a,UAP-nanoHD-5.60.9+12980: stahtd: stahtd[2269]: [STA-TRACKER].stahtd_dump_event(): {"message_type":"STA_ASSOC_TRACKER","assoc_delta":"0","wpa_auth_failures":"1","mac":"7c:df:a1:00:77:ec","vap":"ra0","event_type":"failure","assoc_status":"0","event_id":"12","auth_delta":"0","auth_ts":"2360096.826707"}

Nov 20 22:49:10 nanohd f492bf7f354a,UAP-nanoHD-5.60.9+12980: hostapd: ra0: STA 7c:df:a1:00:77:ec IEEE 802.11: disassociated
Nov 20 22:49:10 nanohd f492bf7f354a,UAP-nanoHD-5.60.9+12980: kernel: ra0: AUTH - receive DE-AUTH(seq-3796) from 7c:df:a1:00:77:ec, reason=3
Nov 20 22:49:10 nanohd f492bf7f354a,UAP-nanoHD-5.60.9+12980: kernel: ra0: (MlmeDeAuthAction)Send DEAUTH frame with ReasonCode(2) to 7c:df:a1:00:77:ec
Nov 20 22:49:10 nanohd f492bf7f354a,UAP-nanoHD-5.60.9+12980: : wevent[2268]: wevent.ubnt_custom_event(): EVENT_STA_LEAVE ra0: 7c:df:a1:00:77:ec / 9
Nov 20 22:49:10 nanohd f492bf7f354a,UAP-nanoHD-5.60.9+12980: hostapd: ra0: STA 7c:df:a1:00:77:ec IEEE 802.11: disassociated
```

[image]

```php
[  1479][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[WIFI] Connecting to wifi... [20000 ms]

.....................E (7394) wifi:Association refused temporarily, comeback time 1048 mSec
.....................[  5611][V][WiFiGeneric.cpp:289] _arduino_event_cb(): STA Disconnected: SSID: ssid, BSSID: f4:92:bf:7f:35:4b, Reason: 203
[  5612][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  5619][W][WiFiGeneric.cpp:852] _eventCallback(): Reason: 203 - ASSOC_FAIL
```

Now there are 2 things going on here.

1. Why is this behavior happening? De-auth protection mechanism on both sides maybe? Can we turn it off, should it be turned off for some 80211 protos? or something ? Is it a security mechanism and thats it, we are stuck with it?

1.b Even with a workaround and reconnect, we are looking at connection times of 8-16 seconds over 1.6s in previous SDKs. Making bssid, and channel caching an absolute requirement now and YET STILL having double the connect times of 3-5s. Battery power devices be damned.. lol

2. Missing Event reason/codes to properly identify these in arduino and defer a reconnect to handle this independently from autoreconnect maybe? As of now this usually fails with no useful reason other than auth fail which is confusing. Again still looking at 8s reconnects.

refs#
https://github.com/espressif/arduino-esp32/issues/2501#issuecomment-977947879

tablatronix, 2022-01-04:
If you want to test this on arduino, be sure to turn off autoreconnect or you will be fighting 2 things as there is a async state issue in there. ` WiFi.setAutoReconnect(false);`

nishanth-radja, 2022-01-05:
Hi @tablatronix , can we have the complete capture to understand the issue better?

tablatronix, 2022-01-06:
First assoc succeeds, second fails

[esp32_deauthissue_reset.pcapng.zip](https://github.com/espressif/esp-idf/files/7819205/esp32_deauthissue_reset.pcapng.zip)

jack0c, 2022-01-06:
Reason code 2 means: `Previous authentication no longer valid`, no communication between the devices for long? There should be a QoS NULL keep alive frame sent from ESP32 every 5s, do you see the frames.

tablatronix, 2022-01-06:
Sure let me check those and see if I am not capturing, 

this is a reboot not sleep.

this is a popular issue in arduino , 3 years old hundreds of comments.

tablatronix, 2022-01-06:
Acks are there

[image]

tablatronix, 2022-01-06:
qos and acks are in capture..

[esp32_deauthissue_doublereset.pcapng.zip](https://github.com/espressif/esp-idf/files/7822446/esp32_deauthissue_doublereset.pcapng.zip)

[image]

nishanth-radja, 2022-01-07:
@tablatronix From the capture,We can see that the sta is getting deauthed  because there is a shift in beacon time.I see that there is multiple BSSID and couple of beacons have not been transmitted by the AP after STA comes out of power save.since there is a time shift now the STA will not be able to see the consecutive beacons and eventually will scan again and go for reconnect. scan starts from pkt 24709.Once there is probe-response from AP,STA will send deauth and associate again as expected pkt(26849).
But AP is has not cleared the STA from it's list yet. Hence the ap sends deauth  when sta sends authentication and does not reply to assoc req from STA.

nishanth-radja, 2022-01-07:
[image]

nishanth-radja, 2022-01-07:
Beacon interval in 1024 ms,but we see a 2053 ms difference.To confirm if this is causing the issue,Can you disable the power save on esp32 and check, Also if you have captures during other instance of disconnect ,can you pls share them. we can verify if this is the case every time.

tablatronix, 2022-01-07:
What other instance of disconnect are you interested in? Not sure I understand, this is every time device is reset, well **every other time**, as the ap goes back and forth with the sta state tracking toggle like. I will capture a bunch of stuff and make a new test so you can see it better.

tablatronix, 2022-01-07:
I wrote some new tests and found something that agrees with a timing issue, This only occurs on hardware resets, not soft resets.

tablatronix, 2022-01-07:
Well here is me hardware reseting every 10 seconds, its clear that every second boot fails with auth fail. Hope it helps figure something out.. 

[every_other_hardware_reset.pcapng.zip](https://github.com/espressif/esp-idf/files/7830038/every_other_hardware_reset.pcapng.zip)

tablatronix, 2022-01-07:
Here is a capture from
ESP-IDF version: v3.2.3-14-gd3e56290
connect succeeds on every boot ( fast too 1.3s! ) 
Auth issue not present.

[every_other_hardware_reset_3_2_3.pcapng.zip](https://github.com/espressif/esp-idf/files/7830441/every_other_hardware_reset_3_2_3.pcapng.zip)

and the ino

```cpp

#include <rom/rtc.h>
#include <WiFi.h>
#include <esp_wifi.h>
#include <ESPmDNS.h>

long start = 0;
uint16_t reboot = 10000;

void setup(){
	delay(1000);
	Serial.begin(115200);
	Serial.setDebugOutput(true);
	Serial.println("Starting...");

    // size_t freeHeap = heap_caps_get_free_size(MALLOC_CAP_8BIT);
    // DEBUGGER("Free heap:       ", freeHeap);
    Serial.println("ESP-IDF version: " + (String) esp_get_idf_version());

    // WiFi.mode(WIFI_STA);

	// WiFi.printDiag(Serial);

    // btStop();
    WiFi.setSleep(false);	
	WiFi.setAutoReconnect(false);
	WiFi.begin("leela","Turdtr0n");
	uint8_t res = WiFi.waitForConnectResult();
	Serial.println(res);
	Serial.println(WiFi.status());
	Serial.println((String)(WiFi.status()== WL_CONNECTED)? "CONNECTED" : "NOT CONECTED");
	Serial.println((String)millis());
	start = millis();
}

void loop(){
	if(millis()-start > reboot){ 
	// ESP.restart(); // doesn't work, issue only present on hardware reboots.
	}
	Serial.print(".");
	delay(100);
}
```

nishanth-radja, 2022-01-10:
Hi @tablatronix , what is the latest build on which you are seeing the issue?
From both the captures we were able to figure out the issue,
The issue is that sta sends deauth to AP before starting a fresh connection.(This behavior was introduced lately as a fix for an another issue). 
so the sequence is as follows.
1.Sta sends Deauth to AP before Auth
2.AP Acks the Deauth
3.Sta starts the Auth Process.
4.AP takes time to process the deauth hence is sending the Deauth in between the Auth process. (This issue we are not seen on other Ap's)
5.So Sta is not replying to the further packets from AP.

On the build on which issue is not seen,Deauth is not sent before assoc and hence there is no issue in re connection.
If you can point the build on which the issue is observed,we can port the fix on that build.
Also can you mention the AP model which you are using?

tablatronix, 2022-01-10:
Yup, pretty much what I was seeing hence the title, just wasn't sure how it was causing failures.

I am using a UNIFI Nano HD AP, there are some others in the arduino thread that might have more types.

I will have to do more testing to see what versions, but its been around a while, I have not tested staging yet though

csobrinho, 2022-01-11:
* DevBoard: [ESP32 OLED WIFI Kit 32 0.96 Inch Blue OLED](http://www.hiletgo.com/ProductDetail/2157143.html)

I believe I'm having the same issue. I'm using ESP-IDF directly and I'm synced to 5968bf6b0c6ca5965a7bf09a43274f42a0346c49.

I can see the wifi trying to connect and getting a deauth several times. The connection time went from 2s to 10-20-40s.

Please find attached the [filtered pcap file](https://github.com/espressif/esp-idf/files/7844026/filtered.pcap.zip).

The device is remote so I can only get the logs later tomorrow but the symptoms are similar to these logs over and over:
```
I (23440) wifi:state: init -> auth (b0)
I (23480) wifi:state: auth -> assoc (0)
I (23520) wifi:state: assoc -> run (10)
I (26630) wifi:state: run -> init (fc0)
I (26630) wifi:new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
I (26630) wifi:new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
```

I have a mesh network:
- Nest Wifi Router is at `7c:d9:5c:9a:7a:e0` on channel 11
- Nest Wifi Point is at `7c:d9:5c:9a:7a:e6` on channel 1

My `wifi` code is here:
https://github.com/csobrinho/hydroponics/blob/master/main/network/wifi.c

Sometimes a hard reboot fixes the issue but the device loses wifi later on.

Thanks for the help.

tablatronix, 2022-01-11:
@csobrinho looks like a different issue, you seem to be getting deuths during and after EAPOL which is strange, seems like an unstable connection or a ton of packet loss, whereas this issue occurs every single time with the exact same timing at the same stage of negotiation before assoc.

csobrinho, 2022-01-11:
Portion of the logs:
```
I (10209) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (10209) nwifi: Disconnected from Sobrinho, bssid: c6:f5:47:19:5c:64, reason: 2
I (12279) nwifi: Disconnected from Sobrinho, bssid: 00:00:00:00:00:00, reason: 205
I (14329) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (14339) wifi:state: init -> auth (b0)
I (14339) wifi:state: auth -> assoc (0)
I (14349) wifi:state: assoc -> run (10)
I (17359) wifi:state: run -> init (fc0)
I (17369) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (17369) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (17379) wifi:new:<1,0>, old:<11,0>, ap:<255,255>, sta:<1,0>, prof:1
I (17379) wifi:state: init -> auth (b0)
I (17419) wifi:state: auth -> assoc (0)
I (17419) wifi:state: assoc -> run (10)
I (21429) wifi:state: run -> init (2c0)
I (21429) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (21429) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (21439) nwifi: Disconnected from Sobrinho, bssid: c6:f5:47:19:5c:64, reason: 2
I (23499) nwifi: Disconnected from Sobrinho, bssid: 00:00:00:00:00:00, reason: 205
I (25559) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (25559) wifi:state: init -> auth (b0)
I (25569) wifi:state: auth -> assoc (0)
I (25579) wifi:state: assoc -> run (10)
I (28589) wifi:state: run -> init (fc0)
I (28589) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (28589) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (28599) wifi:new:<1,0>, old:<11,0>, ap:<255,255>, sta:<1,0>, prof:1
I (28609) wifi:state: init -> auth (b0)
I (28669) wifi:state: auth -> assoc (0)
I (28669) wifi:state: assoc -> run (10)
I (28719) wifi:connected with Sobrinho, aid = 1, channel 1, BW20, bssid = c6:f5:47:19:5c:64
I (28719) wifi:security: WPA2-PSK, phy: bgn, rssi: -68
I (28779) wifi:pm start, type: 0
I (28779) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (31799) wifi:state: run -> init (2c0)
I (31799) wifi:pm stop, total sleep time: 0 us / 3014037 us
I (31799) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (31799) nwifi: Disconnected from Sobrinho, bssid: c6:f5:47:19:5c:64, reason: 2
I (33869) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (33869) wifi:state: init -> auth (b0)
I (33879) wifi:state: auth -> assoc (0)
I (33889) wifi:state: assoc -> run (10)
I (35929) wifi:connected with Sobrinho, aid = 4, channel 11, BW20, bssid = 7c:d9:5c:9a:7a:e6
I (35929) wifi:security: WPA2-PSK, phy: bgn, rssi: -67
I (35939) wifi:pm start, type: 0
I (35979) wifi:AP's beacon interval = 102400 us, DTIM period = 2
W (36139) wifi:<ba-add>idx:0 (ifx:0, 7c:d9:5c:9a:7a:e6), tid:0, ssn:0, winSize:64
W (219339) wifi:<ba-add>idx:1 (ifx:0, 7c:d9:5c:9a:7a:e6), tid:6, ssn:4, winSize:64
W (219369) wifi:<ba-del>idx
W (219379) wifi:<ba-add>idx:1 (ifx:0, 7c:d9:5c:9a:7a:e6), tid:6, ssn:5, winSize:64
I (220309) esp_netif_handlers: sta ip: 192.168.86.10, mask: 255.255.255.0, gw: 192.168.86.1

```

tablatronix, 2022-01-11:
I (21439) nwifi: Disconnected from Sobrinho, bssid: c6:f5:47:19:5c:64, reason: 2
I (23499) nwifi: Disconnected from Sobrinho, bssid: 00:00:00:00:00:00, reason: 205

Reasons Are AUTH EXPIRE and WIFI_REASON_CONNECTION_FAIL, more than likely unrelated, have you tried a different router, also your rssi is pretty bad

csobrinho, 2022-01-11:
Hi tablatronix, thanks for the help. Yes, I'm using the `esp_wifi_set_ps(WIFI_PS_NONE)`. I already moved the Nest wifi Point closer to the device but I'm still getting `-68` for one and `-65` for the other. I think the device is trying to connect to both APs on the mesh network and keeps getting bounced around.

AxelLin, 2022-01-12:
> I (21439) nwifi: Disconnected from Sobrinho, bssid: c6:f5:47:19:5c:64, reason: 2 I (23499) nwifi: Disconnected from Sobrinho, bssid: 00:00:00:00:00:00, reason: 205
> 
> Reasons Are AUTH EXPIRE and WIFI_REASON_CONNECTION_FAIL, more than likely unrelated, 

@csobrinho 
Please create a new issue.

AxelLin, 2022-01-17:
> I will have to do more testing to see what versions, but its been around a while, I have not tested staging yet though I think the first capture was on `ESP-IDF version: v3.3.5-1-g85c43024c` But I assume its sometime after 3.1 maybe? Ill see if I can get pio to cooperate and narrow down with some version tagging

Since it's reproducable, you can use "git bisect" to find the first bad commit.

tablatronix, 2022-01-17:
I can but my env is not setup to , it will take a while, its easier for someone that knows the idf to know when this was introduced and why. 

Platformio uses its own repo for esp32 releases, and using upstream doesn't always build

nishanth-radja, 2022-02-08:
Hi @tablatronix Can you pls try this patch.
These files needs to be replaced in the following location.
`$IDF_HOME/components/esp32/lib`
clean the already existing build and rebuild again.
idf.py fullclean
idf.py build

Also pls take the sniffer capture during association using this patch. 

[patch_for_reconnect_issue.zip](https://github.com/espressif/esp-idf/files/8022416/patch_for_reconnect_issue.zip)
.

tablatronix, 2022-02-08:
Sure thing

tablatronix, 2022-02-11:
If I am using PIO can I do a normal clean and compile?
Also what version of IDF do I need to patch from ?

tablatronix, 2022-02-11:
nm, that didnt work, ill just install the IDF cli and do it there..,

[image]

tablatronix, 2022-02-11:
hmm, not seeing any change.. how can I be sure it rebuilt those?

```php
I (626) wifi:wifi firmware version: 4ea405b
I (626) wifi:wifi certification version: v7.0
I (626) wifi:config NVS flash: enabled
I (626) wifi:config nano formating: disabled
I (626) wifi:Init data frame dynamic rx buffer num: 32
I (636) wifi:Init management frame dynamic rx buffer num: 32
I (636) wifi:Init management short buffer num: 32
I (646) wifi:Init dynamic tx buffer num: 32
I (646) wifi:Init static rx buffer size: 1600
I (646) wifi:Init static rx buffer num: 10
I (656) wifi:Init dynamic rx buffer num: 32
[0;32mI (656) wifi_init: rx ba win: 6[0m
[0;32mI (666) wifi_init: tcpip mbox: 32[0m
[0;32mI (666) wifi_init: udp mbox: 6[0m
[0;32mI (666) wifi_init: tcp mbox: 6[0m
[0;32mI (676) wifi_init: tcp tx win: 5744[0m
[0;32mI (676) wifi_init: tcp rx win: 5744[0m
[0;32mI (686) wifi_init: tcp mss: 1440[0m
[0;32mI (686) wifi_init: WiFi IRAM OP enabled[0m
[0;32mI (686) wifi_init: WiFi RX IRAM OP enabled[0m
[0;32mI (696) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07[0m
I (806) wifi:mode : sta (ac:67:b2:df:85:7c)
I (806) wifi:enable tsf
[0;32mI (806) wifi station: wifi_init_sta finished.[0m
I (816) wifi:new:<3,0>, old:<1,0>, ap:<255,255>, sta:<3,0>, prof:1
I (816) wifi:state: init -> auth (b0)
I (826) wifi:state: auth -> assoc (0)
E (836) wifi:Association refused temporarily, comeback time 1048 mSec
I (1886) wifi:state: assoc -> assoc (0)
I (2886) wifi:state: assoc -> init (200)
I (2886) wifi:new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
[0;32mI (2886) wifi station: connect to the AP fail[0m
[0;32mI (2886) wifi station: Failed to connect to SSID:leela, password:Turdtr0n[0m
```

tablatronix, 2022-02-11:
DId you mean

`esp-idf/components/esp_wifi/lib/esp32/` ?

also same errors, in cli IDF

```
.text.esp_wifi_deinit+0x70): multiple definition of `esp_wifi_deinit'; e
```

[output.txt.zip](https://github.com/espressif/esp-idf/files/8046049/output.txt.zip)

nishanth-radja, 2022-02-11:
@tablatronix I am assuming that you are using "ESP-IDF version: v3.3" ,  So pls use this path.
[patch_for_reconnect_issue_3.3.zip](https://github.com/espressif/esp-idf/files/8048477/patch_for_reconnect_issue_3.3.zip)

once the idf v3.3 is cloned,replace the files in the following location "esp-idf/components/esp32/lib" with the patch attached and build.
Yes, clean and build is same even if you use PIO.

tablatronix, 2022-02-11:
```
shawn@Mac-mini station % idf.py --version
ESP-IDF v5.0-dev-1536-g2522277dfe-dirty
```

tablatronix, 2022-02-12:
Well I tried that first set against #v4.4 and it still throws those errors.

I ll try this 3.3. one next, or else I am doing something wrong

tablatronix, 2022-02-12:
Tried v3.3

Still failing, different make errors

[image]

tablatronix, 2022-02-12:
@nishanth-radja 

i am on OSX M1, so 3.x toolchain is a nightmare to get working on this arch, So i give up.

If you can give me a patch for 4.4 and the proper IDF tag or commit to pull, OR better, submodule branch/commit with the patch ??

nishanth-radja, 2022-02-14:
@tablatronix working with the Dev team on this.Will get back to you soon.

nishanth-radja, 2022-02-14:
Hi @tablatronix , pls clone the idf 4.4 , "release/v4.4" and checkout to tag "730ca0ea43 ".
replace the files in this location "/home/esppool/idf/4.4/esp-idf/components/esp_wifi/lib/esp32" with the attached files.
now you will be able to build.
Pls take the sniffer capture with the patch.
[patch_for_reconnect_issue_4.4_esp32.zip](https://github.com/espressif/esp-idf/files/8060101/patch_for_reconnect_issue_4.4_esp32.zip)

tablatronix, 2022-02-15:
Thanks, well that builds just fine now. I do not see any change in result or in packets.
Ill try to grab some to be sure, not sure how to check that the patch is really in there, but it built fine after a fullclean after replacing files

AxelLin, 2022-02-15:
> not sure how to check that the patch is really in there, but it built fine after a fullclean after replacing files so I am assuming it did

With v4.4 branch(v4.4-172-g730ca0ea43) + patch_for_reconnect_issue_4.4_esp32.zip,
you should observe "-dirty" in wifi firmware version.
I (1094) wifi:wifi firmware version: 7679c42-dirty

Withn a clean v4.4 branch tree:
I (1093) wifi:wifi firmware version: 7679c42

nishanth-radja, 2022-02-15:
@tablatronix can you pls share the sniffer capture, we can verify from the capture if the patch is applied.

tablatronix, 2022-02-15:
```
[0;32mI (27) boot: ESP-IDF v4.4-172-g730ca0ea43-dirty 2nd stage bootloader[0m
[0;32mI (27) boot: compile time 22:50:44[0m
[0;32mI (27) boot: chip revision: 1[0m
[0;32mI (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (38) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (43) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (48) boot.esp32: SPI Flash Size : 2MB[0m
[0;32mI (52) boot: Enabling RNG early entropy source...[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   
```
Ah ok, good

[ESP-IDF v4.4-172-g730ca0ea43-dirty.pcapng.zip](https://github.com/espressif/esp-idf/files/8069376/ESP-IDF.v4.4-172-g730ca0ea43-dirty.pcapng.zip)

nishanth-radja, 2022-02-16:
@tablatronix Thanks for the capture, We are working on a better solution for the problem,Will get back to you ASAP.

nishanth-radja, 2022-02-17:
@tablatronix pls use the attached patch and try to associate.Also pls take the sniffer capture.

[patch_for_reconnect_issue_4.4_esp32_20ms.zip](https://github.com/espressif/esp-idf/files/8088079/patch_for_reconnect_issue_4.4_esp32_20ms.zip)

The fix is that there will be a  ~20ms delay between Deauth and Auth sent from the client so that the AP has enf time to process the deauth and will not send the deauth in between association.There is a delay of 18ms in the ref capture which I have tried in my setup.

[image]

Note: pls do a idf.py full clean and then build again.so that the patch gets applied.
pls share the sniffer capture with this patch.

tablatronix, 2022-02-17:
No success

[image]

[ESP-IDF v4.4-172-g730ca0ea43-dirty_PATCH3.pcapng.zip](https://github.com/espressif/esp-idf/files/8089824/ESP-IDF.v4.4-172-g730ca0ea43-dirty_PATCH3.pcapng.zip)

AxelLin, 2022-02-18:
@nishanth-radja 

I'm wondering if https://github.com/espressif/esp32-wifi-lib/commit/5a4e7d21a14522f8b584832ec6b89626b7916b50 is related to this or not?
What was fixed by https://github.com/espressif/esp32-wifi-lib/commit/5a4e7d21a14522f8b584832ec6b89626b7916b50 from user's point of view.

nishanth-radja, 2022-02-18:
Hi @tablatronix,Thanks for the capture and the initial issue of AP not responding to Assoc req is solved.
Capture before patch -
[image]

From the above capture ,we see that the AP sends deauth after auth and does not respond to assoc req.
after the patch ,
[image]

we see that the AP responds with the Assoc resp ,with a come back time of 1 sec.This is expected as PMF is enabled and the STA info is not cleared from the AP. But AP is not sending any SA query packets.  
And the pkt 16407 is an Ack to the STA but the pkt before is not captured.Which could be the Assoc resp. To confirm this can you take the sniffer capture again.
Also can you disable the PMF on the STA and see if it can connect.Once the PMF is disabled AP will accept the connection the second time without delay.

tablatronix, 2022-02-18:
AHHH yes, sorry, I had it off in my test, but not in the IDF example I am now using for these..
I will test again

AxelLin, 2022-02-19:
> Also can you disable the PMF on the STA and see if it can connect.Once the PMF is disabled AP will accept the connection the second time without delay.

@nishanth-radja 

I though below setting on STA should always work.
         .pmf_cfg = {
            .capable = true,
            .required = false
         }

Did I misunderstndd something?

tablatronix, 2022-02-19:
It depends on the AP setting, both AP and STA have disabled, capable, required.
Here is the matrix

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/wifi-security.html#api-usage

nishanth-radja, 2022-02-21:
> It works with STA PMF off
> 
> [ESP-IDF v4.4-172-g730ca0ea43-dirty_PATCH3_pmfcapableFALSE.pcapng.zip](https://github.com/espressif/esp-idf/files/8101322/ESP-IDF.v4.4-172-g730ca0ea43-dirty_PATCH3_pmfcapableFALSE.pcapng.zip)
> 
> Only issue I can think of is that the IDF says that PMF cannot be disabled on the S2. So I am assuming that this WILL work with PMF on and PMF required on ap, and wpa3 ?
> 
> I can test all these things this week.
> 
> Test PMF
> 
>     * [ ]  AP required / STA Optional
> 
>     * [ ]  AP disabled / STA Optional
> 
>     * [ ]  AP WP3

yes,Pls test it with PMF setting on and pls take the sniffer capture while testing them.

nishanth-radja, 2022-02-21:
@tablatronix yes,Pls test with PMF setting on and pls provide the sniffer capture for the same.

AxelLin, 2022-03-04:
@tablatronix 
Do you have update for the tests in https://github.com/espressif/esp-idf/issues/8192#issuecomment-1045605653 ?

AxelLin, 2022-03-04:
> @AxelLin Yes,with that PMF setting It is supposed to work with all the AP's.We need more captures with the PMF enabled scenario on why it failed to connect with PMF enabled.

@nishanth-radja 
I'm wondering if https://github.com/espressif/esp32-wifi-lib/commit/81768e63548385be79e7b35828832a53faba4393 is related to this issue or not? Could you explain a bit about the commit log? Does it fix any known issue? If yes, what was the issue?

tablatronix, 2022-03-04:
Sorry I have not had a chance to do more testing will try soon

AxelLin, 2022-04-18:
@tablatronix Any update?

AxelLin, 2022-05-10:
> Sorry I have not had a chance to do more testing will try soon

@tablatronix 
Any update?
The PMF is always enabled now, so if it failed there must be some bug needs fix.

tablatronix, 2022-06-11:
@nishanth-radja Is this patched in 4.4.1?

Is this this issue?

`Fixed Assoc comeback causing invalid state bug and cleanup in some FTM failure cases`

If so can you tag this issue# in release

nishanth-radja, 2022-06-21:
@tablatronix No ,That is not the patch for the above mentioned issue.

AxelLin, 2022-07-25:
hi @nishanth-radja 
Could you share the commit of this fix? (Since this is marked as resolution done).
I'd like to check if the fix is in v4.3 and v4.4 or not.

jack0c, 2023-06-02:
master: ce1ab731be47fa2d509739b2ce7f0244ce836950
v4.4: 53c186af263ac76fb0a45e6f63037c43a86c3736
v4.3: 53c186af263ac76fb0a45e6f63037c43a86c3736

[2 more replies]

closed 2023-06-02 by jack0c