espressif/esp-idf #8193: Error in running fatfsgen example

MaazSk, 2022-01-04:
## Environment

- Development Kit:      [ESP32-DevKit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32|]
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-939-g952b4d848f
   
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
  
- Operating System:     [Linux]

- Using an IDE?: [VS Code]
- Power Supply:         [USB]

## Problem Description
I am trying fatfsgen example provided in the example section of the [esp-idf here](https://github.com/espressif/esp-idf/tree/master/examples/storage/fatfsgen). I just changed the wear leveling configuration to safety and 512 bytes. But after these settings I am getting the following error

> I (0) cpu_start: Starting scheduler on APP CPU.
> I (349) example: Mounting FAT filesystem
> W (359) vfs_fat_spiflash: f_mount failed (13)
> E (359) example: Failed to mount FATFS (ESP_FAIL)
> 

Thanks in advance

igrr, 2022-01-04:
@MaazSk I'm afraid only 4096 byte WL sector size is supported, for now:
https://github.com/espressif/esp-idf/blob/69757842aab537a5b06dfc699e350ffbaf6330e6/components/fatfs/wl_fatfsgen.py#L70

MaazSk, 2022-01-04:
@igrr, Thank you for the quick update.

Can you just help me with one of the issues I am facing when ever my device gets restarted I am losing all my data stored in fatfs or it is been getting corrupted. So, the reason it should not get corrupted I was using Safe mode but as it is not implemented how can I achieve this?

Thank you in advance

igrr, 2022-01-04:
@MaazSk could you describe what the application is doing before the device is restated? Perhaps you could attach a simplified version of your application which demonstrates the issue?

The "safe" option of wear_levelling with 512 byte sector size is not "safer" than using 4096 byte sector size. The word "safe" refers only to the way 512 byte sectors are emulated on the 4096-byte-sector SPI Flash devices. When wear_levelling is configured with 4096 byte sector size, there is no need for this emulation, and wear_levelling layer should be safe with regards to sudden power off (unless there is a bug we don't know about).

This so far was related just to the wear_levelling layer. The FAT filesystem which is stored inside the wear_levelling "container" isn't safe with regards to the sudden power off. If you turn off the device while FAT library is performing some update to the filesystem, it is possible that the filesystem will get corrupted and won't be fully recovered on the next power on.

If your use case includes the device being randomly powered off by external factors, a different filesystem (like littlefs) could be a better choice. If on the other hand you are restarting the device in a controlled way, you might need to check that fatfs partition is deinitialized before restart.

We are also considering adding an additional journalling layer between FAT and wear_levelling. This should improve robustness of FAT filesystem with respect to sudden power off. However I can't give an estimate yet when this feature will be implemented.

MaazSk, 2022-01-05:
@igrr, The reason behind the restart was a bug which is been resolved for now.

> Perhaps you could attach a simplified version of your application which demonstrates the issue?
I am using fatfs to log data to the file using `esp_log_set_vprintf` function

### Code snippet
**Initializing FATFS**
```
 ESP_LOGI(TAG, "Initializing FATFS");
    // char base_path_var[] = "/spiflash";
    char partition_label_var[] = "cert";
    ESP_LOGI(TAG, "Mounting FAT filesystem");

    // To mount device we need name of device partition, define base_path
    // and allow format partition in case if it is new one and was not formated before
    const esp_vfs_fat_mount_config_t mount_config = {
        .max_files = 6,
        .format_if_mount_failed = false,
        .allocation_unit_size = CONFIG_WL_SECTOR_SIZE};
    esp_err_t err = esp_vfs_fat_spiflash_mount(BASE_PATH_VAR, partition_label_var, &mount_config, &s_wl_handle);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "Failed to mount FATFS (%s)", esp_err_to_name(err));
        return err;
    }
    return err;
```

**Logging Functionality**
```
int log_to_file(const char *fmt, va_list args)
{
    char buffer[100] = {0};
    vsprintf(buffer, fmt, args);

    //for debugging purpose
    // printf("rec str: %s",buffer);

    if (buffer[1] == '[')
    {
        // printf("buffer value is: %c",buffer[0]);
        char *sub = NULL;
        if (buffer[7] == 'E')
        {
            sub = "[0;31m";
        }
        else if (buffer[7] == 'I')
        {
            sub = "[0;32m";
        }
        else if (buffer[7] == 'W')
        {
            sub = "[0;33m";
        }
        char *sub2 = "[0m";
        char *final_string = strremove(buffer, sub);
        final_string = strremove(final_string, sub2);
        strcat(final_string, "\n");
        print_to_file(final_string);
        memset(buffer, 0, sizeof(buffer));
    }
    else
    {
        print_to_file(buffer);
    }
    // ALSO Write to stdout!
    // return vprintf(fmt, args);
    vprintf(fmt, args);
    return 0;
}
```
There was a bug in ```log_to_file``` due to which the system used to get restarted. But then also there are chances that my system may shutdown while a writing operation is been performed e.g. No electricity when the operation is been performed.

> We are also considering adding an additional journalling layer between FAT and wear_levelling. This should improve robustness of FAT filesystem with respect to sudden power off. However, I can't give an estimate yet when this feature will be implemented.
Ok, so can we just keep this issue open so whenever this functionality is done I will come to know

Thank you!

igrr, 2022-01-05:
@MaazSk for the case of forwarding logs to a file, I would recommend collecting multiple log entries into an intermediate buffer, then flushing the buffer to the file when the buffer becomes full. You can do this flushing from a separate task.

This solution helps in two ways:

1. Avoids reentrancy issues if ESP_LOG is called somewhere from filesystem driver. See the discussion in https://github.com/espressif/esp-idf/issues/1693.
2. Write operations are only done from one task, so you are less likely to overflow the stack of one of the other tasks. Usually writing to a file requires more stack space than logging to UART, so some of the default stack sizes might not be sufficient if you want to redirect logs to a file. Moving write operations into a separate task largely helps with this.

> Does esp-idf support little-fs if yes which library would you suggest to use

With IDF 4.4 and later you can install this third party component into your project using IDF Component Manager: https://components.espressif.com/component/joltwallet/littlefs.
For older versions of ESP-IDF, clone the repository into the components directory of your project (or add it as a submodule): https://github.com/joltwallet/esp_littlefs.

MaazSk, 2022-01-05:
> @MaazSk for the case of forwarding logs to a file, I would recommend collecting multiple log entries into an intermediate buffer, then flushing the buffer to the file when the buffer becomes full. You can do this flushing from a separate task.

@igrr, I will implement this in my code, Thank you for your valuable response

> For older versions of ESP-IDF, clone the repository into the components directory of your project (or add it as a submodule): https://github.com/joltwallet/esp_littlefs.

I will check it thank you.

> Ok, so can we just keep this issue open so whenever this functionality is done I will come to know

Can you please let me know about this?

Thank you

igrr, 2022-01-05:
@MaazSk I've created a task to track this — https://github.com/espressif/esp-idf/issues/8197. Please follow the link and click "Subscribe" button on the right to get notified of the updates.

closed 2022-01-05 by igrr