espressif/esp-idf #8194: Ability to regenerate sdkconfig.defaults programmatically

greenaddress, 2022-01-04:
**Is your feature request related to a problem? Please describe.**

I would like to regenerate some sdkconfig.defaults file.
Currently I remove sdkconfig. copy the right sdkconfig.defaults file, idf.py menuconfig and press `d` to save the new regenerated default.
Maybe I can automate it with `expect`. Or maybe there is already a way and I don't know about it.

**Describe the solution you'd like**

idf.py refresh_default_config

**Describe alternatives you've considered**

I could try with using expect and control idf.py menuconfig programmatically that way but it seems a bit of a hack
Alternatively it would be nice if in the `build` directory there was a generated default file anytime idf.py build or idf.py app is called.

igrr, 2022-01-04:
I think this was recently implemented by @dobairoland in https://github.com/espressif/esp-idf/commit/5118dd7cf39041eb0aabd1cdfceb5726cedac137. Could you please give `idf.py save-defconfig` a try, if you are using the recent `master` branch?

greenaddress, 2022-01-05:
@igrr this looks exactly like what I need!

Unfortunately we only use stable releases, right now we are on 4.3.2 waiting for 4.4 - I guess we'll have to wait for a later release that will include this but that's fine! thank you very much for pointing this to me! Shall I close the issue or? cheers

dobairoland, 2022-01-06:
Yes @greenaddress. This will be available starting with v5.0. However, you can cherry-pick the commit into your ESP-IDF copy and use it. It should not break your stable release.

closed 2022-01-06 by dobairoland