espressif/esp-idf #8197: Filesystem reliability in power-off scenarios

igrr, 2022-01-05:
**Is your feature request related to a problem? Please describe.**

FAT and SPIFFS filesystems provided in ESP-IDF may become corrupted if the chip is powered off or reset during a filesystem operation. This is a tracking issue for improvements to filesystem reliability.

**Describe the solution you'd like**

* Add an optional journalling layer between fatfs and wear_levelling
* Document filesystem reliability considerations in the Programming Guide.

**Describe alternatives you've considered**

* Consider adding an officially supported port of littlefs, while deprecating the usage of spiffs and fatfs. (Fatfs in external storage would still be supported.)

**Additional context**

- https://github.com/espressif/esp-idf/issues/2730
- https://github.com/espressif/esp-idf/issues/3311
- https://github.com/espressif/esp-idf/issues/3355
- https://github.com/espressif/esp-idf/issues/6405
- https://github.com/espressif/esp-idf/issues/8193

filzek, 2022-03-07:
Hi, did this problem was fixed already?

We use fatfs and wear leaveling but the problem happen when the partition is saving and the power failure happen, so, it's a caos as it's corrupt all partition so all files became useless.

Is there any patch avaliable?

MaazSk, 2022-03-07:
Hi @filzek, 
I don't think the issue is been still resolved. you can use other libraries if you are using fatfs which are mentioned [here](https://github.com/espressif/esp-idf/issues/8193#issuecomment-1005510803) 

> With IDF 4.4 and later you can install this third party component into your project using IDF Component Manager: https://components.espressif.com/component/joltwallet/littlefs.
> For older versions of ESP-IDF, clone the repository into the components directory of your project (or add it as a submodule): https://github.com/joltwallet/esp_littlefs.

filzek, 2022-03-07:
Can a fatfs partition be mounted on littlefs?

For devices in field what should be the options to adjust it without
loosing data?

MaazSk, 2022-03-14:
Hi @filzek, 
Sorry for the delay in response. I don't know whether we can achieve it or not as I am not using littlefs now and I have not tried it also.

filzek, 2022-04-29:
@igrr to update devices using FAT partition in field, as   data, Fat, encrypted, could we run it using LittleFS on it?

AxelLin, 2023-11-29:
Hi @igrr 

How is the status of this topic?
Any improvement for FAT/SPIFFS in power-off scenarios?

I recently has a device using littlefs that hit below error.
E (70265) esp_littlefs: /IDF-dev/myproj/managed_components/joltwallet__littlefs/src/littlefs/lfs.c:1346:error: Corrupted dir pair at {0x1, 0x0}

E (70270) esp_littlefs: mount failed, (-84)

I'm not sure if it's related to power-off scenarios or not,
but I'm wondering if littlefs is stable enough to replace spiffs ?

igrr, 2023-11-29:
Yes, there is a journalling layer in development now. It will sit between wear_levelling and fatfs and provide "transaction-like" behavior to file operations. This will be available as a standalone component usable along with FAT FS.

We haven't looked into improvements to SPIFFS or LittleFS, though. For LittleFS, I would recommend raising the issue upstream, I think the maintainers address the issues quite actively there.

AxelLin, 2024-03-25:
> Yes, there is a journalling layer in development now. It will sit between wear_levelling and fatfs and provide "transaction-like" behavior to file operations. This will be available as a standalone component usable along with FAT FS.

Is this available now?

filzek, 2024-03-27:
Right now we are working in the SDK 5.2.1  and we are creating a converse transition on the fly to the change the partition table and be able to save in realtime all files that are in the partition to littlefs, next week this part will be ready and I will get back with the feedback.

Currently in 5,2,1 when concurrence happen in ISR, GPTIMER, Console or UART exchange information and File Read/Write the partition get full corrupted, we will also open a ticket for it.

pacucha42, 2024-03-27:
> > Yes, there is a journalling layer in development now. It will sit between wear_levelling and fatfs and provide "transaction-like" behavior to file operations. This will be available as a standalone component usable along with FAT FS.
> 
> Is this available now?

The component is under internal testing now and shall be published soon

filzek, 2024-03-27:
> The component is under internal testing now and shall be published soon

when gptimer, isr dispatch are simultaneous act the fatfs corrupt in sdk v5, so, its quite important to test in this scenario, if you want us to previous test, we can have full usage scenario to give you a real feedback.

AxelLin, 2024-05-25:
> The component is under internal testing now and shall be published soon

@pacucha42 But I still cannot find it in esp-idf?

pacucha42, 2024-05-25:
> @pacucha42 But I still cannot find it in esp-idf?

Hi @AxelLin,
the esp_fs_journal component is in internal testing phase, as its role in the file-system component stack is quite critical and we need to be sure there are no hidden issues possibly jeopardizing user data.
The development got prolonged, sorry - I just hope we haven't caused to much of trouble to you.
If everything goes well, the first public version should be available in few weeks time. Please, keep watching this tracker for the updates.

By the way, the journaling is implemented as a stand-alone component, not bound to a specific IDF version (only minimum version is defined).
More news coming soon, thanks for the patience.

AxelLin, 2024-11-06:
> If everything goes well, the first public version should be available in few weeks time. Please, keep watching this tracker for the updates.

@pacucha42 It has been several months, how is the status now?

filzek, 2024-11-06:
Sdk 5.3.1 has some improvement on failures, but still happen and also has another bug that we will open a ticket tomorrow, in some very heavy cpu use the power failure can cause a new kind of problem that corrupts the boot partition data making the fw to be switched to the previous one. This is very bad and we had about 8 cases from 800 devices in the last week, so 1% doing this odd behavior.

The corruption in fatfs that is currently being writed and cause a failure on its data, but is not all the time it occurs.

pacucha42, 2024-11-06:
Hi @AxelLin - sorry, I have forgotten to update this tracker! So, the first version of esp_jrnl component had been released few months ago, but it had couple of bugs and was not good enough for public use. Known bugs were fixed, the tests were extended and I'm currently preparing another update. I'm letting you know during next week (you are in my calendar now to avoid forgetting again :-) )

@filzek - sorry to hear about your troubles. Please, open the ticket and share the details of your test case, we shall try to find a solution for you.

Regarding FatFS corruption: it mostly happens on power failures during f_sync() calls, as that's where the file system usually gets updated. esp_jrnl significantly mitigates the problem but it cannot guarantee it never happen s again (specifically during unstable powering conditions, which can cause corruption at physical layer of the flash memory. Like reading flipped bits that are actually stored correctly etc)

filzek, 2024-11-06:
We create a recovery service upon files or partition crashes, so, we have made a workaround for each file and partition.

Also we have create a routine to update on the fly partition table reallocation and fixes, so we can change partition type and alignment to fix the problems with olders sdk.

pacucha42, 2024-11-17:
Hi. Here goes the update mentioned previously:
- the component works fine (based on test.apps results)
- currently I'm working on the documentation and examples, the last steps needed before publishing
- I'll post you here when the whole package is ready, which shouldn't take long

pacucha42, 2024-12-30:
Hi again. Public version of FatFS journaling component available at https://components.espressif.com/components/espressif/esp_jrnl/versions/1.0.1, including example and all the standard links (repo, issues, dependencies).

Any sort of testing, comments or reviews are very appreciated.

Happy New Year 2025

closed 2024-12-30 by pacucha42

pacucha42, 2025-07-24:
Hi @nebkat 

> [@pacucha42](https://github.com/pacucha42) Could you just comment on the decision of using of `esp_jrnl` + FatFS vs LittleFS? Is there any benefit to using one over the other or do they both essentially achieve the same thing? 

They both do pretty much the same (cache write data to avoid file-system corruption due to sudden power-off), it's hard to say which one would be more reliable. Need to do a benchmarking study, thanks for the reminder. For now, pick whatever suits better to your application design

> Is it possible to migrate an existing FatFS partition to use `esp_jrnl`?

Yes, the FatFS partition lives without any knowledge of being journaled (and wear-levelled) - you just need to add sufficient space for the journaling store (cache) and the wear-levelling mechanism. See the 'Implementation details' chapter in https://components.espressif.com/components/espressif/esp_jrnl/versions/1.0.1 README. I'll try to provide some tooling for such cases, though it may take a while due to other projects development.

Feel free to ping me here, if you have other questions or proposals.