espressif/esp-idf #8199: WIFI fails to connect in station mode S3 DevKitC using Station example. It fails to set connection bit?

wreyford, 2022-01-06:
## Environment

- Development Kit:      [ESP32-S3-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-S3]
- IDF version (run ``git describe --tags`` to find it):
    v4.4-beta1-284-gd83021a6e8
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [Yes VSCode Insiders, Version: 1.64.0-insider (user setup)
Commit: 9afbea7caab9dd06592ddd98aa4161b43b126d3f
Date: 2022-01-06T05:15:07.481Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.19044]
- Power Supply:         [USB]

## Problem Description

Wifi no longer connecting to router. Password and SSID correctly set.

In the terminal I get the following output, and then it just doesn't return to main app, ie it appears that the connection bits are not set.

### Expected Behavior
Should connect and then set the connection bits, and return to main program execution.

### Actual Behavior
Negotiates the connection with the Mikrotik router, and then just appears to be waiting for the connection bits to be set, but does not progress any further. After preset time it will try and reconnect. It says connected and displays the correct BSSID MAC, but it has not been assigned a IP by the router. It's as if it nearly gets done, then fails for some reason.

### Steps to reproduce

Using the ESP-IDF/examples/wifi/getting_started/station code from github, as an example in VSCode.

### Code to reproduce this issue

Using the ESP-IDF/examples/wifi/getting_started/station code from github, as an example in VSCode.

## Debug Logs

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1664
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f58
entry 0x403b6248
I (24) boot: ESP-IDF v4.4-beta1-284-gd83021a6e8-dirt 2nd stage bootloader
I (25) boot: compile time 16:33:24
I (25) boot: chip revision: 0
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (38) boot.esp32s3: SPI Flash Size : 8MB
I (43) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length  
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00200000
I (81) boot:  3 littlefs         Unknown data     01 82 00210000 00500000
I (89) boot: End of partition table
I (93) esp_image: segment 0: paddr=00010020 vaddr=3c0f0020 size=6f4c8h (455880) map
I (183) esp_image: segment 1: paddr=0007f4f0 vaddr=3fc99c10 size=00b28h (  2856) load
I (184) esp_image: segment 2: paddr=00080020 vaddr=42000020 size=e0a6ch (920172) map
I (354) esp_image: segment 3: paddr=00160a94 vaddr=3fc9a738 size=04f40h ( 20288) load
I (358) esp_image: segment 4: paddr=001659dc vaddr=40374000 size=15c0ch ( 89100) load
I (379) esp_image: segment 5: paddr=0017b5f0 vaddr=50000000 size=00014h (    20) load
I (380) esp_image: segment 6: paddr=0017b60c vaddr=600fe000 size=00028h (    40) load
I (394) boot: Loaded app from partition at offset 0x10000
I (394) boot: Disabling RNG early entropy source...
I (408) cpu_start: Pro cpu up.
I (409) cpu_start: Starting app cpu, entry point is 0x40375454
0x40375454: call_start_cpu1 at C:/esp/esp-idf/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (441) cpu_start: Pro cpu start user code
I (441) cpu_start: cpu freq: 240000000

I (1213) pp: pp rom version: e7ae62f
I (1213) net80211: net80211 rom version: e7ae62f
I (1233) wifi:wifi driver task: 3fce974c, prio:23, stack:6656, core=0
I (1233) system_api: Base MAC address is not set
I (1233) system_api: read default base MAC address from EFUSE
I (1253) wifi:wifi firmware version: 7e67c79
I (1253) wifi:wifi certification version: v7.0
I (1253) wifi:config NVS flash: enabled
I (1253) wifi:config nano formating: disabled
I (1253) wifi:Init data frame dynamic rx buffer num: 32
I (1263) wifi:Init management frame dynamic rx buffer num: 32
I (1263) wifi:Init management short buffer num: 32
I (1273) wifi:Init dynamic tx buffer num: 32
I (1273) wifi:Init static tx FG buffer num: 2
I (1283) wifi:Init static rx buffer size: 1600
I (1283) wifi:Init static rx buffer num: 10
I (1283) wifi:Init dynamic rx buffer num: 32
I (1293) Touch pad: touch pad [9] base 26948, thresh 5389
I (1293) wifi_init: rx ba win: 6
I (1303) wifi_init: tcpip mbox: 32
I (1303) wifi_init: udp mbox: 6
I (1313) wifi_init: tcp mbox: 6
I (1313) wifi_init: tcp tx win: 5744
I (1313) wifi_init: tcp rx win: 5744
I (1323) wifi_init: tcp mss: 1440
I (1323) wifi_init: WiFi IRAM OP enabled
I (1333) wifi_init: WiFi RX IRAM OP enabled
I (1333) Loadassist Wifi: Starting event handlers for managing WIFI /IP etc events.
I (1343) Loadassist Wifi: ESP_WIFI_MODE_STATION
I (1353) Loadassist Wifi: connecteding to ap with SSID:MYSTATION password:MYPASSWORD
I (1363) phy_init: phy_version 302,3ddfab3,Dec  3 2021,14:39:12
I (1413) wifi:mode : sta (S3 MacAddress reported correctly)
I (1413) wifi:enable tsf
I (1413) Loadassist Wifi: wifi_init_sta finished. ...Waiting for wifi_event_group Bits to be set
I (1413) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (1423) wifi:state: init -> auth (b0)
I (1423) wifi:state: auth -> assoc (0)
I (1433) wifi:state: assoc -> run (10)
I (1453) wifi:connected with MYSTATION, aid = 3, channel 1, BW20, bssid = Mikrotik Mac Reported Correctly
I (1453) wifi:security: WPA2-PSK, phy: bgn, rssi: -31
I (1463) wifi:pm start, type: 1

I (1463) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (1483) wifi:BcnInt:102400, DTIM:1
W (2313) wifi:<ba-add>idx:0 (ifx:0, e4:8d:8c:0b:0e:bb), tid:0, ssn:3, winSize:64
```
And it gets no further than 2313....

So Wifi is currently not usable.

mhdong, 2022-01-09:
hi @wreyford thanks for reporting this issue. According to the log, we know that your IDF version does not correspond to your lib version.
I (24) boot: ESP-IDF v4.4-beta1-284-gd83021a6e8-dirt 2nd stage bootloader
I (1253) wifi:wifi firmware version: 7e67c79
wifi firmware version should be 0d44bbe8

 Please use the following commend to update lib: 
git submodule update -f

wreyford, 2022-01-09:
hi @mhdong
Sorry, I did not pick that up, and you saved the day. Thus there never was an issue with the WIFI, other than outdated libs on my part.
I confirm that wifi working as always again.

closed 2022-01-09 by wreyford