espressif/esp-idf #8202: 环境变量问题

We-Sky, 2022-01-07:
我是用WSL1设置开发环境，在使用`. $HOME/esp-idf/export.sh ` 添加环境变量后，每次重启都需要重新添加环境变量，是什么原因呢

We-Sky, 2022-01-07:
现在输入`. $HOME/esp-idf/export.sh` 后提示这个
IDF_PATH is set to '/home/zhongtian/esp/esp-idf', but it is not a valid directory.
If you have set IDF_PATH manually, check if the path is correct.

o-marshmallow, 2022-01-07:
Hi @We-Sky ,
This is how environment variables work, they are only active for the current terminal session. You can make the export automatic when you launch bash by adding the line `. /home/zhongtian/esp-idf/export.sh` to your `~/.bashrc` file.
```
echo ". /home/zhongtian/esp-idf/export.sh" >> ~/.bashrc
```

We-Sky, 2022-01-07:
> Hi @We-Sky , This is how environment variables work, they are only active for the current terminal session. You can make the export automatic when you launch bash by adding the line `. /home/zhongtian/esp-idf/export.sh` to your `~/.bashrc` file. You can do this with the command:
> 
> ```
> echo ". /home/zhongtian/esp-idf/export.sh" >> ~/.bashrc
> ```

但是执行后现在提示这个是什么原因呢：
IDF_PATH is set to '/home/zhongtian/esp/esp-idf', but it is not a valid directory.
If you have set IDF_PATH manually, check if the path is correct.

o-marshmallow, 2022-01-07:
@We-Sky 你可以手动定义IDF_PATH变量。请你试一下：
```
export IDF_PATH=/home/zhongtian/esp-idf
. $IDF_PATH/export.sh
```

We-Sky, 2022-01-07:
> @We-Sky 你可以手动定义IDF_PATH变量。请你试一下：
> 
> ```
> export IDF_PATH=/home/zhongtian/esp-idf
> . $IDF_PATH/export.sh
> ```
我刚才执行'echo ". /home/zhongtian/esp-idf/export.sh" >> ~/.bashrc' 后环境变量被添加到了这里，路径不知道怎么多了个esp,这个怎么修改掉呀？
[image]

o-marshmallow, 2022-01-07:
@We-Sky 你可以用`nano`编辑这个文件：
```
nano ~/.bashrc
```
你改文件的最后一行：
```
. /home/zhongtian/esp-idf/export.sh
```
为
```
export IDF_PATH=/home/zhongtian/esp-idf
. $IDF_PATH/export.sh
```
写好了就按Ctrl-X推出`nano`程序

Caffreyfans, 2022-01-07:
你可以在你的 .bashrc 文件最后一行直接添加就行了，你每次设置都是临时的环境变量，所以每次重新登录后都需要重新设置。懂了啵

```
source /home/zhongtime/esp-idf/export.sh
````

We-Sky, 2022-01-07:
谢谢，已经成功啦

Alvin1Zhang, 2022-01-07:
感谢反馈。

closed 2022-01-07 by Alvin1Zhang

igrr, 2022-01-07:
Please note that while you can add the export script into your profile script, this is not recommend. Quoting https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#get-started-export:

> Technically, you can add export.sh to your shell’s profile directly; however, it is not recommended. Doing so activates IDF virtual environment in every terminal session (including those where IDF is not needed), defeating the purpose of the virtual environment and likely affecting other software.