espressif/esp-idf #8204: `--openocd-bin` option and `OPENOCD_BIN` environment variable

trombik, 2022-01-07:
**Is your feature request related to a problem? Please describe.**

`idf.py` assumes `openocd` executable is `openocd`. this conflicts when you have another version of `openocd`.

major distributions provides `openocd` as a package. as espressif maintains a fork of `openocd`, the `openocd` package does not have necessary bits for `esp32` family. when you want use both versions of `openocd`, you have to remove one.

**Describe the solution you'd like**

the best solution would be to upstream all the necessary bits. however, espressif keeps maintaining its own forks. there must have been reasons we do not know.

another possible solution is to add `--openocd-bin` option and `OPENOCD_BIN` environment variable to `tools/idf_py_actions/debug_ext.py`. users would be able to switch `openocd` with the option, or the environment variable.

**Describe alternatives you've considered**

see above.

**Additional context**

n/a

dobairoland, 2022-01-07:
I assume from the request and described solution that the conflict occurs during invoking `idf.py` and not during the install or setting up the environment.

So if you have `openocd` in the PATH and then switch to an ESP-IDF environment then the path to the appropriate Espressif fork of `openocd` is getting added to the beginning of the PATH string and therefore, the right version of `openocd` will be invoked in the ESP-IDF terminal environment. When you leave the environment then the previously available `openocd` will be available again.

What am I missing? What is wrong with this approach?

You are not supposed to add path to ESP-IDF's `openocd` into the `PATH` environment permanently.

trombik, 2022-01-07:
@dobairoland yes, that's another solution. would you clarify why espressif do not upstream patches (not only `binutils`, but other tools)? of course, I can guess many reasons why you don't, but an official statement is better.

dobairoland, 2022-01-07:
I'm sorry but I cannot give any official statement regarding this because my guess is just as good as yours. I handle only integration into the environment and not openocd development.

trombik, 2022-01-07:
@dobairoland okay, thanks.

closed 2022-01-07 by trombik

igrr, 2022-01-07:
@trombik I can answer about upstreaming our OpenOCD changes.

Initially we didn't build our development process for OpenOCD around the upstream repository for two reasons:

When the development of openocd for ESP32 was started, we were still learning about the Xtensa debug module, experimenting with ways to support multi-core RTOS aware debugging, tracing, and so on. Our code was changing quite often and going through the upstream repository for each change would be pretty slow. At the same time, OpenOCD project had a period when new contributions were pending without review for a long time.

At the same time, with a very small team working on toolchains and debugging tools, it was hard to justify putting more effort on upstreaming when there are bugs that need to be fixed, new chips which need to be supported, test infrastructure which should be improved. The practical need of releasing a feature-complete and stable version of OpenOCD for our chips had higher priority than the effort required to upstream our changes.

Recently, the basic part of our Xtensa port in OpenOCD has become relatively stable, to the point that we believe it's close to the state where it could be upstreamed. The team working on debugging tools has also expanded, so now it became viable to start working on upstreaming while simultaneously handling the maintenance and chip support related tasks. As @gerekon has mentioned in https://github.com/espressif/openocd-esp32/issues/23, we plan to start working on cleaning up our fork and submitting the first batch of patches in the first half of 2022.

trombik, 2022-01-07:
@igrr you've been great, and the explanation is another proof of it. what you described is not far from my guess, and is understandable. thanks for being so open.