espressif/esp-idf #8205: Modbus user-defined function codes

fredj0207, 2022-01-07:
**Is your feature request related to a problem? Please describe.**

We need to support user-defined function codes as defined in Modbus protocol.
The freeModbus library allows to do it through the eMBRegisterCB however the IDF freeModbus component does not give access to this function (no public API) and the total number of supported functions is very limited (16).

**Describe the solution you'd like**

- Have a public API to register CB in freemodbus component
- Have a way to support more function codes than 16 in the component

alisitsyn, 2022-01-11:
@fredj0207 ,

Thank you for your feature request to add this functionality. The ESP_Modbus supports just common read/write functions for now. Registration of custom function codes and handlers is planned to be supported after ESP-IDF v5.0.
For now, I would propose you to modify the component CMakeLists.txt file and expose the private include paths for your project.

Alvin1Zhang, 2025-02-25:
Thanks for reporting and sorry for slow turnaround, let us close this one and track in https://github.com/espressif/esp-modbus/issues/34 and https://github.com/espressif/esp-modbus/issues/83.

closed 2025-02-25 by Alvin1Zhang