espressif/esp-idf #8209: esp_restart() issue

fpgamaster, 2022-01-07:
## Environment

- Development Kit:  none
- Module or chip used: ESP32-WROOM-32U
- IDF version:  v5.0-dev-382-g098dfe8a08
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- Power Supply: external

## Problem Description
LoadProhibited exception on boot after calling esp_restart()

....
W (4458) device: Factory reset
I (5118) wifi:state: run -> init (0)
I (5118) wifi:pm stop, total sleep time: 2392006 us / 3791625 us

W (5118) wifi:<ba-del>idx
I (5118) wifi:new:<9,0>, old:<9,1>, ap:<255,255>, sta:<9,1>, prof:1
I (5138) wifi:new:<9,1>, old:<9,0>, ap:<255,255>, sta:<9,1>, prof:1
I (5138) wifi:state: init -> auth (b0)
I (5138) wifi:state: auth -> assoc (0)
I (5148) wifi:state: assoc -> run (10)
I (5158) wifi:state: run -> init (0)
I (5158) wifi:new:<9,0>, old:<9,1>, ap:<255,255>, sta:<9,1>, prof:1
I (5218) wifi:flush txq
I (5218) wifi:stop sw txq
I (5218) wifi:lmac stop hw txq
I (5238) wifi:Deinit lldesc rx mblock:4
W (5298) device: Factory reset done. Rebooting ...
ets Jun  8 2016 00:22:57

**rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0038,len:9612
ho 0 tail 12 room 4
load:0x40078000,len:24908
load:0x40080400,len:3464
0x40080400: _init at ??:?

entry 0x40080638
Fatal exception (28): LoadProhibited
epc1=0x4007bf1b, epc2=0x00000000, epc3=0x00000000, excvaddr=0xa34f0000, depc=0x00000000
ets Jun  8 2016 00:22:57**

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0038,len:9612
ho 0 tail 12 room 4
load:0x40078000,len:24908
load:0x40080400,len:3464
0x40080400: _init at ??:?

entry 0x40080638
I (832) cpu_start: Pro cpu up.
I (832) cpu_start: Starting app cpu, entry point is 0x400816bc

o-marshmallow, 2022-01-11:
Hi @fpgamaster ,
I tried to reproduce the issue, with the same issue and the same board and same commit ID, but I was unable to. Can you provide a minimal example to reproduce the issue?

fpgamaster, 2022-01-21:
Hi @o-marshmallow,
Unfortunately it is not a small project and it is not easy to extract something as a minimal example :(
The project was moved from  v4.2 where this issue had never happened.
I saw that the esp_restart() function is refactored in v5 but I have not looked too deep in it.

Best regards

AxelLin, 2022-02-13:
@fpgamaster 

Fatal exception (28): LoadProhibited happens before cpu_start: Pro cpu up.
Can you provide the sdkconfig?

fpgamaster, 2022-02-14:
Hi @AxelLin,

All diffs of the sdkcondid are in sdkconfig.default

CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP=y
#
CONFIG_BOOTLOADER_LOG_LEVEL_WARN=y
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOTLOADER_REFLASHABLE=y
CONFIG_SECURE_FLASH_ENC_ENABLED=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="parts_ota.csv"
CONFIG_PARTITION_TABLE_OFFSET=0x10000
CONFIG_COMPILER_OPTIMIZATION_SIZE=y
CONFIG_BT_ENABLED=y
CONFIG_BTDM_CTRL_BLE_MAX_CONN=4
# CONFIG_BTDM_CTRL_FULL_SCAN_SUPPORTED is not set
CONFIG_BT_NIMBLE_ENABLED=y
# CONFIG_SPI_MASTER_ISR_IN_IRAM is not set
# CONFIG_SPI_SLAVE_ISR_IN_IRAM is not set
CONFIG_ESP32_REV_MIN_1=y
CONFIG_ESP32_ULP_COPROC_ENABLED=y
CONFIG_ESP32_ULP_COPROC_RESERVE_MEM=7936
CONFIG_ESP32_RTC_CLK_SRC_EXT_OSC=y
CONFIG_ESP32_RTC_CLK_CAL_CYCLES=1024
# CONFIG_ETH_USE_ESP32_EMAC is not set
# CONFIG_ETH_USE_SPI_ETHERNET is not set
# CONFIG_ESP_HTTP_CLIENT_ENABLE_HTTPS is not set
CONFIG_ESP32_UNIVERSAL_MAC_ADDRESSES_TWO=y
# CONFIG_ESP_NETIF_TCPIP_ADAPTER_COMPATIBLE_LAYER is not set
CONFIG_ESP_PHY_INIT_DATA_IN_PARTITION=y
CONFIG_PM_ENABLE=y
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=4
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=8
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=16
CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS=y
CONFIG_FREERTOS_USE_TICKLESS_IDLE=y
CONFIG_LWIP_SO_RCVBUF=y
# CONFIG_LWIP_DHCP_DOES_ARP_CHECK is not set
CONFIG_LWIP_TCP_SYNMAXRTX=6
CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN=4096
CONFIG_MBEDTLS_PSK_MODES=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y
# CONFIG_MBEDTLS_SSL_PROTO_TLS1 is not set
# CONFIG_MBEDTLS_SSL_PROTO_TLS1_1 is not set
CONFIG_MBEDTLS_SSL_PROTO_DTLS=y
CONFIG_MBEDTLS_DHM_C=y
# CONFIG_MQTT_PROTOCOL_311 is not set
# CONFIG_MQTT_TRANSPORT_SSL is not set
# CONFIG_MQTT_TRANSPORT_WEBSOCKET is not set
# CONFIG_NVS_ENCRYPTION is not set

fpgamaster, 2022-02-14:
It seems that sometimes esp_restart() does not trigger this exception, mostly after the first power on.
If someone wants to reproduce this let's wait for a few restarts.

Also, I am observing some issues with the ULP code running after a reboot using esp_restart().
It looks like the hardware is not in the same state as it is after a power on.
I get strange readings from ADC1 which may be related to GPIO state or ADC registers state or something else.

AxelLin, 2022-02-14:
Does this issue happen when ULP is disabled?

ESP-Marius, 2024-11-15:
Marking this one as closed as with the current information it wasn't possible to reproduce or give much debug feedback.

If you are still experiencing this issue and want to continue debugging it please re-open the issue and ill take a look at it

closed 2024-11-15 by ESP-Marius