espressif/esp-idf #8210: eth2ap example throws error ESP_ERR_WIFI_NOT_ASSOC (12309) error

sauttefk, 2022-01-08:
## Environment

- Module or chip used:  ESP32-WROVER-I
- IDF version: v5.0-dev-1077-g397639da7d
- Build System: idf.py
- Compiler version: (crosstool-NG esp-2021r2) 8.4.0
- Operating System: macOS
- Using an IDE?: VScode
- Power Supply: USB + PoE
- Ethernet PHY: LAN8710
- EMAC-CLK: ETH_CLOCK_GPIO0_OUT + 74AC04 inverter

## Problem Description

eth2ap example throws error `ESP_ERR_WIFI_NOT_ASSOC` (12309) error
```
E (326848) eth_example: WiFi send packet failed: 12309
```

### Expected Behavior
Bridge between Ethernet and WiFi

### Actual Behavior
* Ethernet connects properly
* Client can establish a connection to the ESP32 WiFi AP
* ESP32 throws error "The WiFi connection is not associated"

### Code to reproduce this issue
just using the example code with the necessary changes reflecting the hardware setup.

## Debug Logs

```
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7100
ho 0 tail 12 room 4
load:0x40078000,len:15528
ho 0 tail 12 room 4
load:0x40080400,len:4488
0x40080400: _init at ??:?

entry 0x400806b4
I (31) boot: ESP-IDF v5.0-dev-1077-g397639da7d 2nd stage bootloader
I (31) boot: compile time 23:59:33
I (31) boot: chip revision: 3
I (35) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (42) qio_mode: Enabling default flash chip QIO
I (48) boot.esp32: SPI Speed      : 80MHz
I (52) boot.esp32: SPI Mode       : QOUT
I (57) boot.esp32: SPI Flash Size : 16MB
I (62) boot: Enabling RNG early entropy source...
I (67) boot: Partition Table:
I (71) boot: ## Label            Usage          Type ST Offset   Length
I (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  2 factory          factory app      00 00 00010000 00100000
I (100) boot: End of partition table
I (105) boot_comm: chip revision: 3, min. application chip revision: 0
I (112) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=163c0h ( 91072) map
I (146) esp_image: segment 1: paddr=000263e8 vaddr=3ffb0000 size=03998h ( 14744) load
I (151) esp_image: segment 2: paddr=00029d88 vaddr=40080000 size=06290h ( 25232) load
I (160) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=61f10h (401168) map
I (274) esp_image: segment 4: paddr=00091f38 vaddr=40086290 size=126d8h ( 75480) load
I (300) esp_image: segment 5: paddr=000a4618 vaddr=50000000 size=00010h (    16) load
I (312) boot: Loaded app from partition at offset 0x10000
I (312) boot: Disabling RNG early entropy source...
I (323) psram: This chip is ESP32-D0WD
I (324) spiram: Found 64MBit SPI RAM device
I (324) spiram: SPI RAM mode: flash 80m sram 80m
I (329) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (336) cpu_start: Pro cpu up.
I (340) cpu_start: Starting app cpu, entry point is 0x4008145c
0x4008145c: call_start_cpu1 at [...]/components/esp_system/port/cpu_start.c:152

I (0) cpu_start: App cpu up.
I (842) spiram: SPI SRAM memory test OK
I (850) cpu_start: Pro cpu start user code
I (850) cpu_start: cpu freq: 240000000 Hz
I (850) cpu_start: Application information:
I (853) cpu_start: Project name:     eth2ap
I (858) cpu_start: App version:      v5.0-dev-1077-g397639da7d
I (865) cpu_start: Compile time:     Jan  8 2022 00:36:52
I (871) cpu_start: ELF file SHA256:  db7e4d79dfadc5a1...
I (877) cpu_start: ESP-IDF:          v5.0-dev-1077-g397639da7d
I (883) heap_init: Initializing. RAM available for dynamic allocation:
I (890) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (897) heap_init: At 3FFB6720 len 000298E0 (166 KiB): DRAM
I (903) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (909) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (916) heap_init: At 40098968 len 00007698 (29 KiB): IRAM
I (922) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (930) spi_flash: detected chip: gd
I (934) spi_flash: flash io: qout
I (939) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (948) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (978) wifi:wifi driver task: 3ffc6bc4, prio:23, stack:6656, core=0
I (978) system_api: Base MAC address is not set
I (978) system_api: read default base MAC address from EFUSE
I (978) wifi:wifi firmware version: 7679c42
I (988) wifi:wifi certification version: v7.0
I (988) wifi:config NVS flash: enabled
I (988) wifi:config nano formating: disabled
I (998) wifi:Init data frame dynamic rx buffer num: 32
I (998) wifi:Init management frame dynamic rx buffer num: 32
I (1008) wifi:Init management short buffer num: 32
I (1008) wifi:Init static tx buffer num: 16
I (1018) wifi:Init tx cache buffer num: 32
I (1018) wifi:Init static rx buffer size: 1600
I (1018) wifi:Init static rx buffer num: 10
I (1028) wifi:Init dynamic rx buffer num: 32
I (1028) wifi_init: rx ba win: 6
I (1038) wifi_init: tcpip mbox: 32
I (1038) wifi_init: udp mbox: 6
I (1038) wifi_init: tcp mbox: 6
I (1048) wifi_init: tcp tx win: 5744
I (1048) wifi_init: tcp rx win: 5744
I (1058) wifi_init: tcp mss: 1440
I (1058) wifi_init: WiFi/LWIP prefer SPIRAM
I (1058) wifi_init: WiFi IRAM OP enabled
I (1068) wifi_init: WiFi RX IRAM OP enabled
I (3618) eth_example: Ethernet Started
I (3618) eth_example: Ethernet Link Up
I (3618) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (3708) wifi:mode : softAP (ac:67:b2:75:4d:23)
I (3718) wifi:Total power save buffer number: 8
I (3718) wifi:Init max length of beacon: 752/752
I (3718) wifi:Init max length of beacon: 752/752
I (19808) wifi:new:<9,0>, old:<9,1>, ap:<9,1>, sta:<255,255>, prof:9
I (19808) wifi:station: 10:f1:f2:41:17:75 join, AID=1, bgn, 20
I (19828) eth_example: Wi-Fi AP got a station connected
W (21898) wifi:<ba-add>idx:2 (ifx:1, 10:f1:f2:41:17:75), tid:6, ssn:2, winSize:64
W (21908) wifi:<ba-add>idx:3 (ifx:1, 10:f1:f2:41:17:75), tid:0, ssn:10, winSize:64
E (23388) eth_example: WiFi send packet failed: 12309
E (25638) eth_example: WiFi send packet failed: 12309
E (25748) eth_example: send flow control message failed or timeout
E (27898) eth_example: WiFi send packet failed: 12309
I (29208) wifi:station: 10:f1:f2:41:17:75 leave, AID = 1, bss_flags is 658547, bss:0x3f80f31c
I (29208) wifi:new:<9,0>, old:<9,0>, ap:<9,1>, sta:<255,255>, prof:9
W (29208) wifi:<ba-del>idx
W (29218) wifi:<ba-del>idx
I (29218) eth_example: Wi-Fi AP got a station disconnected
E (30368) eth_example: WiFi send packet failed: 12309
```

xueyunfei998, 2022-01-18:
hi @sauttefk 

E (27898) eth_example: WiFi send packet failed: 12309

This error occurs because there is a problem with the link between sta and ap, which leads to an error in sending data.

I (29218) eth_example: Wi-Fi AP got a station disconnected

AxelLin, 2022-02-13:
@sauttefk 
Looks like you are running eth2ap example code.

E (25638) eth_example: WiFi send packet failed: 12309
E (25748) eth_example: send flow control message failed or timeout

Please take a look at https://github.com/espressif/esp-idf/tree/master/examples/ethernet/eth2ap#troubleshooting

closed 2022-03-16 by Alvin1Zhang