espressif/esp-idf #8211: MODBUS TCP slave crash on start after stop

oachkatzalschwoaf, 2022-01-08:
## Environment

- Development Kit:  ESP32 NodeMCU + custom ESP32-S2 board
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.3.2
- Operating System: Windows
- Using an IDE: Visual Studio Code - Espressif IDF

## Problem Description

Crash/Guru Mediation when trying to initialize modbus-tcp after it got stopped.  
Discovered on custom ESP32-S2, reproduced with example code on ESP32

### Steps to reproduce

initialize modbus tcp as described in documentation: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/modbus.html?highlight=modbus#modbus-slave-api-overview

1. start/connect to wifi
2. mbc_slave_init_tcp()
3. mbc_slave_setup()
4. mbc_slave_set_descriptor()
5. mbc_slave_start()
	- modbus-tcp is running
	
6. mbc_slave_destroy()

7. goto steps 2-5  
	 - Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled. 

### Code to reproduce this issue

Based on example mb_tcp_slave - changes:  
Configured CONFIG_EXAMPLE_WIFI_SSID and CONFIG_EXAMPLE_WIFI_PASSWORD

modified app_main:

```cpp
void app_main(void)
{
    ESP_ERROR_CHECK(init_services());

    // Set UART log level
    esp_log_level_set(SLAVE_TAG, ESP_LOG_INFO);

    mb_communication_info_t comm_info = { 0 };

    comm_info.ip_addr_type = MB_IPV4;
    comm_info.ip_mode = MB_MODE_TCP;

    comm_info.ip_port = MB_TCP_PORT_NUMBER;
    ESP_ERROR_CHECK(slave_init(&comm_info));

    // The Modbus slave logic is located in this function (user handling of Modbus)
    // slave_operation_func(NULL);

    while(1){
        vTaskDelay(2000);
        ESP_LOGW(SLAVE_TAG, "Close Modbus...");
        slave_destroy();
        vTaskDelay(2000);
        ESP_LOGW(SLAVE_TAG, "Start Modbus...");
        slave_init(&comm_info);					// < CRASH!!!!
    }

    ESP_ERROR_CHECK(slave_destroy());
    ESP_ERROR_CHECK(destroy_services());
}

```

## Debug Logs

```
...
I (6938) example_connect: - IPv4 address: 192.168.1.22
I (6948) wifi:Set ps type: 0

D (6948) MB_PORT_COMMON: vMBPortSetMode: Port enter critical.
D (6958) MB_PORT_COMMON: vMBPortSetMode: Port exit critical
I (6968) MB_TCP_SLAVE_PORT: Protocol stack initialized.
I (6968) MB_TCP_SLAVE_PORT: Socket (#54), listener  on port: 502, errno=0
W (27028) SLAVE_TEST: Close Modbus...
D (27028) MB_PORT_COMMON: vMBPortSetMode: Port enter critical.
D (27028) MB_PORT_COMMON: vMBPortSetMode: Port exit critical
W (47028) SLAVE_TEST: Start Modbus...
D (47028) MB_PORT_COMMON: vMBPortSetMode: Port enter critical.
D (47028) MB_PORT_COMMON: vMBPortSetMode: Port exit critical
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008d8ee  PS      : 0x00060f33  A0      : 0x8008e39a  A1      : 0x3ffbad90
A2      : 0x3ffb828c  A3      : 0x00000024  A4      : 0x00060b20  A5      : 0x00060923
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x3ffc6f88  A9      : 0x00000019
A10     : 0x3ffb828c  A11     : 0x00000000  A12     : 0x00000003  A13     : 0x00060f23
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000017  EXCCAUSE: 0x0000001d
EXCVADDR: 0x0000000f  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd

Backtrace:0x4008d8eb:0x3ffbad90 0x4008e397:0x3ffbadb0 0x400817e5:0x3ffbadd0 0x40081819:0x3ffbadf0 0x4008fb41:0x3ffbae10 0x4008ba9e:0x3ffbae30 0x400d7523:0x3ffbae50 0x400d73ba:0x3ffbae80 0x400d669f:0x3ffbaeb0 0x400d6919:0x3ffbaf00 0x401438d5:0x3ffbaf40 0x4008c239:0x3ffbaf60

```

alisitsyn, 2022-01-10:
Hi @oachkatzalschwoaf ,

Thank you for reporting this issue. The issue is supposed to be fixed in commit:
5a0b570702d83e12b10c4b2bb1bec7583fa5e3c2

I will check this ASAP. Could you check this commit on your side? Thanks.

alisitsyn, 2022-01-10:
@oachkatzalschwoaf ,

The issue is confirmed. Please do the following to fix the issue:

1. The `components/freemodbus/tcp_slave/modbus_controller/mbc_tcp_slave.c::mbc_tcp_slave_destroy()` function is missing the initialization of mbs_interface_ptr=NULL.
[The patch to apply: `git apply 0001_fix_tcp_slave_destroy.patch.log`](https://github.com/espressif/esp-idf/files/7837242/0001_fix_tcp_slave_destroy.patch.log)
2. Please use the command `git cherry-pick 61f3af0` which applies the fixes to correctly cleanup lwip internals.

The log after the fix is below:
[tcp_slave_fix_destroy.log](https://github.com/espressif/esp-idf/files/7837256/tcp_slave_fix_destroy.log)

Please try the above and let me know the result. Thanks.

oachkatzalschwoaf, 2022-01-10:
@alisitsyn 

now its working but somewehere might be a memory leak  
each init/destroy-cycle 16-36bytes are lost:

[tcp_slave_memory_leak.log](https://github.com/espressif/esp-idf/files/7838431/tcp_slave_memory_leak.log)

```
    while(1){
        vTaskDelay(1000/portTICK_PERIOD_MS);
        ESP_LOGW(SLAVE_TAG, "Close Modbus...");
        slave_destroy();
        ESP_LOGI("RTOS", "    total_free_bytes:      destroyed: %d", heap_caps_get_free_size(MALLOC_CAP_INTERNAL));
        vTaskDelay(100/portTICK_PERIOD_MS);
        ESP_LOGW(SLAVE_TAG, "Start Modbus...");
        slave_init(&comm_info);
        ESP_LOGI("RTOS", "    total_free_bytes:     init:  %d", heap_caps_get_free_size(MALLOC_CAP_INTERNAL));
    }
```

alisitsyn, 2022-01-11:
Hi @oachkatzalschwoaf,

I think what you are measuring here is the cost of heap fragmentation.
I checked again the Modbus stack allocations. My check shows that all allocated blocks for structures/buffers are freed at destroying.
The Modbus stack uses WIFI and the LWIP and intensively allocates and frees memory. The heap allocator has the metadata headers for each allocated block = 4 bytes. The more fragmentation, the more total free heap size is also lost due to lots of non-contiguous free blocks that each needs their own 4-byte header.

oachkatzalschwoaf, 2022-01-11:
Thanks @alisitsyn 

I've tested Heap Tracing feature for the first time last night - wasn't sure about the result, as i just got 7 entries, 6 of them: `0x400e5358: mem_malloc at C:/esp/esp-idf/components/lwip/lwip/src/core/mem.c:237`

and a single xMBTCPPortInit 

```
7 allocations trace (1000 entry buffer)
24 bytes (@ 0x3ffdecd8) allocated CPU 0 ccount 0x4c28cf28 caller 0x400d8c06:0x400d7b39:0x400d7981:0x400d77bf:0x400d6c69:0x400d6816:0x400d692
0x400d8c06: xMBTCPPortInit at C:/esp/esp-idf/components/freemodbus/tcp_slave/port/port_tcp_slave.c:130
```

but each time 1 missed free with 120byte leaked
```
120 bytes 'leaked' in trace (7 allocations)
total allocations 90 total frees 89
```

but after adding 
`free(xConfig.pxMbClientInfo); // <<<<<<<< Free array of slave configuration pointers`

I also get the same result. 

With these changes the issue is fixed for me.

alisitsyn, 2022-01-11:
@oachkatzalschwoaf ,

Thank you for update. I guess we use different commits. The leak was in `0x400d8c06: xMBTCPPortInit components/freemodbus/tcp_slave/port/port_tcp_slave.c:138 ` on my side. The fix above helps to free the xConfig.pxMbClientInfo and the leak is gone as per log on my side.

Alvin1Zhang, 2022-03-08:
Thanks for reporting and sorry for slow turnaround, fix on master branch is available at https://github.com/espressif/esp-idf/commit/dcaac79380bc215d91e36ceb61c1b0cf03f75b81.

Alvin1Zhang, 2022-03-17:
Thanks for reporting, fixes on release/4.4 and release/4.3 are available, feel free to reopen if the issue still happens. Thanks.

closed 2022-03-17 by Alvin1Zhang

alisitsyn, 2022-04-06:
Hello @dabbinavo, 

This issue has been corrected. I will check it soon.

dabbinavo, 2022-04-07:
Hello @alisitsyn,

ok, thanks for the hint! Now i found the other commits that fix the issue on the release/v4.4 branch:
- ba9c0c29a9fe3c079b20a626723e848bf2158213
- d7fafdc916eac8e65b190e19eab25f8a5826f7c1

Switching to the latest commit on release/v4.4 instead of using the tag v4.4 solves the issue.

Seems that we have to wait for v4.4.1 to have the fix in an official release.

Alvin1Zhang, 2022-04-07:
Thanks for reporting and sharing the feedback, will update once IDF v4.4.1 releases, thanks.

mjgci, 2024-09-24:
I am still having this issue on my application. 

My application involves init() of Modbus TCP Slave after the Wifi IP Address is acquired. If I change the Wifi configuration, 
I would like to stop the Modbus Slave and restart after the Wifi network is reconnected and the ESP32-S3 crashes.

The logs on my end are while calling mbc_slave_destroy().

IDF version 5.2

alisitsyn, 2024-09-25:
@mjgci,

Sorry for this mistake. There was the mistake when merging the MR. It is already fixed in the esp-modbus master with commit [2601e34e21238787d48b565867575fa825afad0a](https://github.com/espressif/esp-modbus/commit/2601e34e21238787d48b565867575fa825afad0a). Could you check it in your project and report before the component release?

mjgci, 2024-09-25:
@alisitsyn 

Thank you for pointing in the right direction. I have retested my project and it works correctly as expected.
I am able to use mbc_slave_destroy() and re-init the Modbus TCP Slave without crashes repeatedly.
Appreciate your help!

alisitsyn, 2024-09-25:
@mjgci,

Thank you for update. My pleasure to help. The fix will be included into new component release.
Good luck!