espressif/esp-idf #8212: Flickers when using double-buffered i8080 lcd on ESP32-S3 while Wi-Fi performs time-consuming tasks

wangyz1997, 2022-01-09:
## Environment

- Development Kit: Custom ESP32-S3 board.
- Module or chip used: `ESP32-S3-WROOM-1` (D3N8R2).
- IDF version (run ``git describe --tags`` to find it): `v4.4-beta1-284-gd83021a6e8`
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
```
xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
- Operating System: Windows10 21H2.
- (Windows only) environment type: PowerShell.
- Using an IDE?: No, VS Code as editor.
- Power Supply: USB

## Problem Description

Hi. I am using the lateet ESP32-S3 to drive a Intel 8080 LCD Panel. Everything seems quite normal when I port the driver without enabling Wi-Fi, with or without lvgl. But after I enable Wi-Fi, I can see flickers on screen frequently when Wi-Fi performs time-consuming tasks (connecting, https request, etc). The display buffer is correctly flushed to screen, but on the wrong place (which can be seen in images below).

Lower the PCLK helps reducing the frequency of the bug, but the bug still happen.

I guess this is due to interrupt not handled timely. Wi-Fi functions have many critical sections, and the LCD DMA uses interrupt to notify the driver when transfer is done. When Wi-Fi stack enters critical section, if LCD DMA generates an interrupt, the interrupt will not be handled timely. This may cause mistakes in command transferring, so the buffer is flushed to the wrong place.

### Expected Behavior

The LCD displays correct content.

### Actual Behavior

With LVGL:
[image]

[image]

With the code provided:

https://user-images.githubusercontent.com/29337599/148683367-fe6762ca-67b2-453f-8b6c-eea0fde5ae45.mp4

### Steps to reproduce

1. Write a screen refreshing task on core 1, which flushes 8 colored blocks to screen. Each block is transferred with seprate transaction. Double buffer is used.
2. Write a Wi-Fi task on core 0, and the Wi-Fi stack is also on core 0. Connect to a wrong Wi-Fi to simulate time-consuming tasks.
3. The bug occurs.

### Code to reproduce this issue

Minimal code to reproduce this issue is uploaded below.

https://gist.github.com/wangyz1997/a0fde61e4a212d72dcd964ec9346a9fc

## Debug Logs

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x16e4
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f90
entry 0x403b6248
I (24) boot: ESP-IDF v5.0-dev-1082-g9f303290d8-dirty 2nd stage bootloader
I (25) boot: compile time 20:31:04
I (25) boot: chip revision: 0
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
[... 86 lines cut ...]
I (31296) example_connect: Wi-Fi disconnected, trying to reconnect...
I (33346) example_connect: Wi-Fi disconnected, trying to reconnect...
I (35396) example_connect: Wi-Fi disconnected, trying to reconnect...
I (37446) example_connect: Wi-Fi disconnected, trying to reconnect...
I (39496) example_connect: Wi-Fi disconnected, trying to reconnect...
I (41536) example_connect: Wi-Fi disconnected, trying to reconnect...
I (43586) example_connect: Wi-Fi disconnected, trying to reconnect...
I (45636) example_connect: Wi-Fi disconnected, trying to reconnect...
I (47686) example_connect: Wi-Fi disconnected, trying to reconnect...
I (49726) example_connect: Wi-Fi disconnected, trying to reconnect...
I (51776) example_connect: Wi-Fi disconnected, trying to reconnect...
I (53826) example_connect: Wi-Fi disconnected, trying to reconnect...
I (55876) example_connect: Wi-Fi disconnected, trying to reconnect...
I (57926) example_connect: Wi-Fi disconnected, trying to reconnect...
I (59966) example_connect: Wi-Fi disconnected, trying to reconnect...
I (62016) example_connect: Wi-Fi disconnected, trying to reconnect...
I (64066) example_connect: Wi-Fi disconnected, trying to reconnect...
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
    everything is left default, and wifi ssid and password is set to a wrong one.
- [x] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
    [esp32s3-8080lcd-bugreport.elf.zip](https://github.com/espressif/esp-idf/files/7835018/esp32s3-8080lcd-bugreport.elf.zip)

suda-morris, 2022-01-10:
The `esp_lcd_panel_io_tx_color` is an async API, which in internal will set up a transaction and send it to a task queue, and then that function will return immediately. So before that transaction got transmitted finally, you have to ensure the associated frame buffer won't be overwritten.

wangyz1997, 2022-01-10:
Yes, I have noticed that already. But everything works fine when Wi-Fi is not turn on. This transmission issue only happens when Wi-Fi is performing time-consuming jobs, which is confusing.

wangyz1997, 2022-01-10:
I patched the code with mutex. The same issue still happen. https://gist.github.com/wangyz1997/0cc05904443d0bd7f16321044d1d42e0

suda-morris, 2022-03-10:
@wangyz1997 Hi, sorry for the late response,  we can reproduce this issue now. Thanks for your detailed report and your code that helps us a lot. 
For now, a quick fix could be adding a `esp_rom_delay_us(1)` after [this line](https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/src/esp_lcd_panel_io_i80.c#L562)

The issue is that, after we start the DMA engine, DMA speed can't match the peripheral requirement, thus it can't fill itself with transfer data immediately, so the peripheral's FIFO might contain garbage data. This is also explained why slowing down the PCLK frequency can somehow do the help.

wangyz1997, 2022-03-11:
Thank you for your detailed explaination.

closed 2022-03-18 by espressif-bot