espressif/esp-idf #8213: GPIO interrupt type is set to rising edge, but the falling edge also triggers the interrupt

ForgetCSX, 2022-01-10:
## Environment

Development Kit: NULL
Kit version : master
Module or chip used: ESP32module and esp32s2module
IDF version : ESP-IDF v5.0-dev-961-g733fbd9ecc
Build System: idf.py
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
Operating System: Linux
Power Supply: USB

## Problem Description

The GPIO interrupt type is set to rising edge, but the falling edge also triggers the interrupt

### Code to reproduce this issue

static void IRAM_ATTR gpio_irq_handler(void* arg)
{
    if (gpio_get_level(GPIO_INPUT_IO_ZERO_ISR))
         HEAT_OUT_HIGH();
     else
        HEAT_OUT_LOW();
}

static void heat_out_init(void)
{
    //zero-initialize the config structure.
    gpio_config_t io_conf = {};
    io_conf.pin_bit_mask = GPIO_INPUT_PIN_ZERO_ISR_SEL;
    io_conf.mode = GPIO_MODE_INPUT;
    io_conf.intr_type = GPIO_INTR_POSEDGE;
    io_conf.pull_down_en = 0;
    io_conf.pull_up_en = 0;
    gpio_config(&io_conf);

    //install gpio isr service
    gpio_install_isr_service(ESP_INTR_FLAG_LEVEL1);
    //hook isr handler for specific gpio pin
    gpio_isr_handler_add(GPIO_INPUT_IO_ZERO_ISR, gpio_irq_handler, (void*) GPIO_INPUT_IO_ZERO_ISR);
    gpio_intr_disable(GPIO_INPUT_IO_ZERO_ISR);
}

## IO waveform
The signal detected：
[image]
Input trigger signal：
[image]
The interrupt handler outputs a signal：
[image]

negativekelvin, 2022-01-10:
https://github.com/espressif/esp-idf/issues/7602

SHristov92, 2023-06-13:
Hi everybody,
It's old issue, but still stands! I have hit the same bug with interrupt based on signal from optocoupler.
Relatively slow falling/rising edges - around 500uS

I needed to get the falling edge so, to get around it I got to sample the logic level of the pin 1000 times and if the sum of all samples is more that 10 I ignore the interrupt. It's really dirty way to do it but it seems to work fine.

`
void ISR(void *arg)
{
	uint16_t count = 0;

	for(uint16_t i=0;i<1000;i++)
	{
		count += gpio_get_level(instance->zcPin);
	}

	if(count<10)
	{
              // valid interrupt
        }
`

ForgetCSX, 2023-08-25:
时间太过久远，关闭问题

closed 2023-08-25 by ForgetCSX

GCTechnology, 2025-02-18:
该问题依然存在，设置上升沿/下降沿触发，但是两个边沿均会触发中断函数。
The problem persists, set the rising edge/falling edge trigger, but both edges trigger the interrupt function.

songruo, 2025-02-20:
@GCTechnology My assumption of your wrong interrupt trigger was due to slow rise/fall time. I have summarized the cause and what you can do in this [thread](https://github.com/espressif/esp-idf/issues/7602#issuecomment-931306892) (the first section). Please try to confirm whether this is the issue, or prove that your problem is not caused by the slow slew rate.