espressif/esp-idf #8215: Adding Matter (CHIP) as an official component in ESP-IDF for future use

oussemagaied, 2022-01-10:
With the introduction of Matter protocol today, I tried to build the lighting example on an ESP32 Dev Kit, It worked perfectly but the handling of the matter component is only accesible from the Matter (CHIP) project.
I tried adding the component in the Platform-IO but there are alot of complicated dependencies in the CMake / Make component list.

This implementation will help developers discover matter and try to integrate it in several applications including the implementation of matter on top the ESP WIFI mesh already provided.

Is there any implementation or ideas working currently in this context?

igrr, 2022-01-10:
@oussemagaied Adding Matter implementation to ESP-IDF is not planned currently, but a matter-specific SDK will be released as a separate project.
Currently we are trying to reduce the number of components in ESP-IDF to make it more maintainable, so new optional features will generally be added via the [IDF component registry](https://components.espressif.com/) or as a standalone projects.

chshu, 2022-01-10:
@oussemagaied Matter is compiled with ESP-IDF as an external component, you can refer to [all-clusters-app example](https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app) and [matter cmake](https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/CMakeLists.txt).

It's straightforward for you to integrate other existing features in ESP-IDF, let us know if you encounter any issue during the integration.

oussemagaied, 2022-01-10:
@chshu Creating an empty ESP-IDF project using Platform IO (Plug-in VS Code) and trying to add the Matter (CHIP) component is not straightforward, What I can see in the all clusters app is that the Matter project is using the ESP-IDF and its toolchain to create a customized matter component and not the opposite,

So the question is : 1-  Can we create the Matter (CHIP) component using the ESP-IDF running on the Platform IO (VS code plugin)
the registration of the component is complicated looking at the CMake list that the component requires
2- Is this implementation is supported on Windows ?

chshu, 2022-01-10:
@oussemagaied There will be corresponding vscode plug-in support with our matter-specific SDK, but it's not ready yet. For now, you will need to use the shell environment for matter development.

The current Matter build system doesn't support Windows, the following platforms are recommended:
- macOS 10.15+
- Debian 10
- Ubuntu 20.04 LTS

igrr, 2022-01-18:
I'm going to close the issue since the question has been answered.

closed 2022-01-18 by igrr

Diegorro98, 2022-03-17:
There's any place to follow updates about the matter-specific SDK (mentioned by @igrr and @chshu)? I'm hyped with matter and I can't wait to start building and testing devices

chshu, 2022-03-18:
@Diegorro98 Please tune in our Matter blogs: https://blog.espressif.com/matter-38ccf1d60bcd, we will have announce there when ESP's matter SDK is public.
You could also start testing with the Matter upstream SDK: https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/esp32