espressif/esp-idf #8217: where to look up the meaning of ESP32 BLE 4.2 function error code?

Rita858, 2022-01-10:
I got an error code(0x51) when doing ble gap security authentication.
I wonder what is the meaning of this code but there is no way to look up. 
So can you tell me where to look up the meaning of ESP32 BLE 4.2 function error code.
[image]
[image]

nopnop2002, 2022-01-14:
I think this can be used.
esp_err_to_name()

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_err.html

Rita858, 2022-01-20:
> I think this can be used. esp_err_to_name()
> 
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_err.html

I'll make a try, thank you.

Alvin1Zhang, 2022-10-27:
Thanks for reporting, would you please help share if any updates for the issue? Thanks.

AxelLin, 2023-05-05:
> I think this can be used. esp_err_to_name()
> 
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_err.html

No, the  ble_security.auth_cmpl.fail_reason is nothing to do with esp_err_t.
esp_ble_auth_cmpl_t:
uint8_t               fail_reason;           /*!< The HCI reason/error code for when success=FALSE */

brendomaciel, 2024-06-23:
I don't know if this is still a problem for the author of this issue, but for anyone who is struggling with it, look up for these kind of logs:

[image]

Since `rsn` actually matches an entry in the reference table, I'm assuming this is the value that is supposed to be in `fail_reason` as well, but for some reason they're different.

By the way, there are a few reference tables out there. Take a look:
- https://www.lisha.ufsc.br/teaching/shi/ine5346-2003-1/work/bluetooth/hci_commands.html
- https://software-dl.ti.com/simplelink/esd/simplelink_cc13x0_sdk/4.20.02.07/exports/docs/blestack/blestack-api/group__HCI__Error__Codes.html
- https://docs.silabs.com/bluetooth/2.11/error-codes