espressif/esp-idf #8218: Core 0 panic'ed (StoreProhibited) during OTA update

permamentydebet, 2022-01-10:
## Environment
- Module or chip used: ESP32-­WROOM­-32E
- IDF version: v4.3.1
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description
During OTA "Core  0 panic'ed (StoreProhibited)" occurred. 

Program received chunk of data (192 bytes) and flash them using  **esp_ota_write_with_offset**. After flashing received chunk data are freed. If some chunks are lost, program ask Modbus master device to retransmit it. After received lost chunk data, program flash them normally using **esp_ota_write_with_offset** function with correct offset. 

Error have been occurred when program try freed retransmitted chunk and **when CLI task is turned on.** Without CLI, whole OTA process works well. 

Used components: 
-  Modbus (RTU mode, RS485)
-  OTA (without flash encryption)
-  CLI 

Task stack size: 
- modbus (2048 bytes)
- firmware update (3072 bytes)
- CLI (default 4096 bytes)

I try change idf version (v4.3, v4.3.2) but error still occurred. 
Fix from: [09db884](https://github.com/espressif/esp-idf/commit/09db8845c692ee5002810744fd1ce889d4d69c9d) doesn't work.

## Debug Logs

```
INF (FWU): Received chunk 880/880 size 32. Chunk data address: 0x3ffbf0fc
INF (FWU): Image size=168224
INF (FWU): free heap=260208
INF (MODBUS): Timestamp: (64501825 us), ADDR:10000, TYPE:64, INST_ADDR:0x3ffb0fbc, SIZE:1
INF (MODBUS): Timestamp: (65519295 us), ADDR:60104, TYPE:1, INST_ADDR:0x3ffb0fba, SIZE:1
INF (FWU): Verify chunk cmd!
INF (FWU): Need to retransmit 4 chunk.
INF (FWU): free heap=261080
INF (MODBUS): Timestamp: (66536504 us), ADDR:60104, TYPE:1, INST_ADDR:0x3ffb0fba, SIZE:1
INF (FWU): Retransmit cmd!
INF (FWU): Retransmit idx done!
INF (FWU): free heap=261080
INF (MODBUS): Timestamp: (66552447 us), ADDR:60110, TYPE:2, INST_ADDR:0x3ffb0fbd, SIZE:3
INF (MODBUS): Timestamp: (66589960 us), ADDR:60000, TYPE:1, INST_ADDR:0x3ffb0ef4, SIZE:99
INF (MODBUS_FWU): Data address: 0x3ffb3d0c
INF (FWU): Flash cmd!
INF (FWU): Received chunk 3/880 size 192. Chunk data address: 0x3ffbf10c
INF (FWU): free heap=261080
INF (MODBUS): Timestamp: (66605987 us), ADDR:10000, TYPE:64, INST_ADDR:0x3ffb0fbc, SIZE:1
INF (MODBUS): Timestamp: (66622120 us), ADDR:60104, TYPE:1, INST_ADDR:0x3ffb0fba, SIZE:1
INF (FWU): Retransmit cmd!
INF (FWU): Retransmit idx done!
INF (FWU): free heap=261080
INF (MODBUS): Timestamp: (66637952 us), ADDR:60110, TYPE:2, INST_ADDR:0x3ffb0fbd, SIZE:3
INF (MODBUS): Timestamp: (66675386 us), ADDR:60000, TYPE:1, INST_ADDR:0x3ffb0ef4, SIZE:99
INF (MODBUS_FWU): Data address: 0x3ffb3d0c
INF (FWU): Flash cmd!
INF (FWU): Received chunk 15/880 size 192. Chunk data address: 0x3ffbf10c
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008bcc8  PS      : 0x00060633  A0      : 0x8008c2ff  A1      : 0x3ffbebe0  
0x4008bcc8: insert_free_block at /workspaces/charger-sw/build/../esp-idf-v4.3.1/components/heap/heap_tlsf.c:243
 (inlined by) block_insert at /workspaces/charger-sw/build/../esp-idf-v4.3.1/components/heap/heap_tlsf.c:269
 (inlined by) tlsf_free at /workspaces/charger-sw/build/../esp-idf-v4.3.1/components/heap/heap_tlsf.c:870

A2      : 0x3ffb3d0c  A3      : 0xa5a5a5a5  A4      : 0x000000c0  A5      : 0x000000c0  
A6      : 0x00000000  A7      : 0x00110b40  A8      : 0x3ffbf104  A9      : 0x00000012  
A10     : 0x00000000  A11     : 0x3ffb464c  A12     : 0x3ffbf10c  A13     : 0x3ffb3d0c  
A14     : 0x00000012  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xa5a5a5b1  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4008bcc5:0x3ffbebe0 0x4008c2fc:0x3ffbec00 0x4008182a:0x3ffbec20 0x4008c609:0x3ffbec40 0x400d8889:0x3ffbec60 0x400d8b2e:0x3ffbed40 0x4008a129:0x3ffbee30
0x4008bcc5: insert_free_block at /workspaces/charger-sw/build/../esp-idf-v4.3.1/components/heap/heap_tlsf.c:242
 (inlined by) block_insert at /workspaces/charger-sw/build/../esp-idf-v4.3.1/components/heap/heap_tlsf.c:269
 (inlined by) tlsf_free at /workspaces/charger-sw/build/../esp-idf-v4.3.1/components/heap/heap_tlsf.c:870

0x4008c2fc: multi_heap_free_impl at /workspaces/charger-sw/build/../esp-idf-v4.3.1/components/heap/multi_heap.c:220

0x4008182a: heap_caps_free at /workspaces/charger-sw/build/../esp-idf-v4.3.1/components/heap/heap_caps.c:305

0x4008c609: free at /workspaces/charger-sw/build/../esp-idf-v4.3.1/components/newlib/heap.c:46

0x400d8889: FwuUnpackAndFlashChunk at /workspaces/charger-sw/build/../src/fwu/fwu.c:136

0x400d8b2e: FwuTask at /workspaces/charger-sw/build/../src/fwu/fwu.c:260

0x4008a129: vPortTaskWrapper at /workspaces/charger-sw/build/../esp-idf-v4.3.1/components/freertos/port/xtensa/port.c:168
```

ESP-YJM, 2022-02-08:
@permamentydebet Could you please help offer a simple code to reproduce it.

Alvin1Zhang, 2022-02-17:
@permamentydebet Thanks for reporting, would you please help share any simple code for us to reproduce? Thanks.

permamentydebet, 2022-02-23:
Hi, I'm almost ready with simple code. Sory to keep you waitning. 

In preparation I have seen somethings interesting. 
I execute a simple allocation in loop inside a function:
`uint8_t* pointer = malloc(sizeof(uint8_t) * 16);`
and I don't use a free before exit the function.

Should I always get the same address in pointer?

igrr, 2022-02-23:
In this case I think the address returned by `malloc` will be different in every loop iteration.

Another thing, not sure if it will help you — in the original issue report it can be seen that the address which was passed to `free` is `0xa5a5a5a5`. This is a value that the stack is filled with before the task starts. So if you see a value 0xa5a5a5a5 in the crash dump, there's a good chance that some uninitialized local variable is involved. Check that in every code path the variable passed to `free` at `FwuUnpackAndFlashChunk at /workspaces/charger-sw/build/../src/fwu/fwu.c:136` is initialized.

permamentydebet, 2022-03-03:
As you mention I got diffrent address in every loop. In my code with FWU I get the same address, because I request to allocate always the same size of memory and any other allocation does not occur.

If I enable heap poisoning I get a information that tail doesn't match.

Here is a repo with code:
https://github.com/permamentydebet/esp32_modbus_fwu

Info:
1. Now I work on esp32s3 and the errors occur too.
2. To upload some data via modbus I use max485 connected to esp (see RS485_init and Modbus_init function)
3. upload.py is a simple python script to upload via modbus. It require: pyserial and pymodbus. 
4. I use usb to rs485 adapter recognised by PID in python script.

If you will have some problems with project, feel free to ask

igrr, 2022-03-03:
If you are enabling heap poisoning and getting the assert that tail doesn't match, you are probably overrunning some buffer you have allocated. You can enable heap tracing to find where the buffer which gets corrupted was allocated. This can help you find the place in the code where you might be writing past the end of a buffer.

permamentydebet, 2022-03-09:
As you suggested I enabled heap tracing. Here is a log of heap_trace_dump function exactly before transmit next chunk and crash.

```
13 allocations trace (2000 entry buffer)
128 bytes (@ 0x3fce7404) allocated CPU 0 ccount 0x061d3968 caller 0x42011c38:0x42011d20
0x42011c38: linenoiseHistoryAdd at ../esp/esp-idf/components/console/linenoise/linenoise.c:1183

0x42011d20: linenoiseEdit at ../esp/esp-idf/components/console/linenoise/linenoise.c:841

1 bytes (@ 0x3fce7488) allocated CPU 0 ccount 0x061d4280 caller 0x40381cc8:
0x40381cc8: _malloc_r at ../esp/esp-idf/components/newlib/heap.c:45

84 bytes (@ 0x3fce7498) allocated CPU 0 ccount 0x068db0b8 caller 0x4037bbdb:0x4037be97
0x4037bbdb: xQueueGenericCreate at ../esp/esp-idf/components/freertos/queue.c:447

0x4037be97: xQueueCreateMutex at ../esp/esp-idf/components/freertos/queue.c:564

84 bytes (@ 0x3fce74f0) allocated CPU 0 ccount 0x068dc000 caller 0x4037bbdb:0x4037be97
0x4037bbdb: xQueueGenericCreate at ../esp/esp-idf/components/freertos/queue.c:447

0x4037be97: xQueueCreateMutex at ../esp/esp-idf/components/freertos/queue.c:564

84 bytes (@ 0x3fce7558) allocated CPU 0 ccount 0xa809b9d0 caller 0x4037bbdb:0x4037be97
0x4037bbdb: xQueueGenericCreate at ../esp/esp-idf/components/freertos/queue.c:447

0x4037be97: xQueueCreateMutex at ../esp/esp-idf/components/freertos/queue.c:564

48 bytes (@ 0x3fce75c8) allocated CPU 0 ccount 0xa8148adc caller 0x4200f0e8:0x4200f23a
0x4200f0e8: load_partitions at ../esp/esp-idf/components/spi_flash/partition.c:214

0x4200f23a: ensure_partitions_loaded at ../esp/esp-idf/components/spi_flash/partition.c:74 (discriminator 15)

48 bytes (@ 0x3fce75fc) allocated CPU 0 ccount 0xa8149bd8 caller 0x4200f0e8:0x4200f23a
0x4200f0e8: load_partitions at ../esp/esp-idf/components/spi_flash/partition.c:214

0x4200f23a: ensure_partitions_loaded at ../esp/esp-idf/components/spi_flash/partition.c:74 (discriminator 15)

48 bytes (@ 0x3fce7630) allocated CPU 0 ccount 0xa814a3a8 caller 0x4200f0e8:0x4200f23a
0x4200f0e8: load_partitions at ../esp/esp-idf/components/spi_flash/partition.c:214

0x4200f23a: ensure_partitions_loaded at ../esp/esp-idf/components/spi_flash/partition.c:74 (discriminator 15)

48 bytes (@ 0x3fce7664) allocated CPU 0 ccount 0xa814aea8 caller 0x4200f0e8:0x4200f23a
0x4200f0e8: load_partitions at ../esp/esp-idf/components/spi_flash/partition.c:214

0x4200f23a: ensure_partitions_loaded at ../esp/esp-idf/components/spi_flash/partition.c:74 (discriminator 15)

48 bytes (@ 0x3fce7698) allocated CPU 0 ccount 0xa814b680 caller 0x4200f0e8:0x4200f23a
0x4200f0e8: load_partitions at ../esp/esp-idf/components/spi_flash/partition.c:214

0x4200f23a: ensure_partitions_loaded at ../esp/esp-idf/components/spi_flash/partition.c:74 (discriminator 15)

48 bytes (@ 0x3fce76cc) allocated CPU 0 ccount 0xa814c170 caller 0x4200f0e8:0x4200f23a
0x4200f0e8: load_partitions at ../esp/esp-idf/components/spi_flash/partition.c:214

0x4200f23a: ensure_partitions_loaded at ../esp/esp-idf/components/spi_flash/partition.c:74 (discriminator 15)

44 bytes (@ 0x3fce7700) allocated CPU 0 ccount 0xafd79a5c caller 0x4200e364:0x420069ce
0x4200e364: esp_ota_begin at ../esp/esp-idf/components/app_update/esp_ota_ops.c:160

0x420069ce: FwuBegin at ../esp/esp32_modbus_fwu/build/../main/fwu/fwu.c:41

12 bytes (@ 0x3fce75b0) allocated CPU 0 ccount 0x6d6d76ec caller 0x42006a3d:0x42006d5d
0x42006a3d: FwuCheckMissingChunk at ../esp/esp32_modbus_fwu/build/../main/fwu/fwu.c:130

0x42006d5d: FwuTask at ../esp/esp32_modbus_fwu/build/../main/fwu/fwu.c:251

725 bytes 'leaked' in trace (13 allocations)
total allocations 4673 total frees 4660
```
1. linenoiseHistoryAdd - is from cli module
2. _malloc_r 
3. xQueueGenericCreate - I use 2 queues but probably some module create another queue
4. load_partitions - occur 6 times? it's okay? above function definition is a comment 
```
// Create linked list of partition_list_item_t structures.
// This function is called only once, with s_partition_list_lock taken. 
```
5. esp_ota_begin
6. FwuCheckMissingChunk - should be freed when all missed chunk will be retransmitted (fwu.c:188)

permamentydebet, 2022-03-22:
As always a problem was save to memory which was freed. 

So problem resolved :)

closed 2022-03-22 by permamentydebet

o-marshmallow, 2022-03-22:
@permamentydebet Thanks for comming back with the solution! :smile: