espressif/esp-idf #8220: idf.py menuconfig crashes

sergioaiello, 2022-01-11:
## Environment
- Module or chip used:  ESP32-WROOM-32E
- IDF version: v4.4-beta1-189-ga79dc75f0a
- Build System:  CMake
- Compiler version: esp-2021r2-8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?:  VSC 1.63.2
- Power Supply:  external 3.3V

## Problem Description

//Detailed problem description goes here.
"idf.py menuconfig" command does not terminate with the following output

[menuconfig.txt](https://github.com/espressif/esp-idf/files/7842649/menuconfig.txt)

### Expected Behavior
menu GUI to navigate with arrows

### Actual Behavior
from IDF command prompt digit "idf.py menuconfig" and the output is as follows
[menuconfig.txt](https://github.com/espressif/esp-idf/files/7842664/menuconfig.txt)

### Steps to reproduce

1. open IDF command window
2. change directory to select the project path
3. digit "idf.py menuconfig"

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

dobairoland, 2022-01-11:
Could you please set your `IDF_PATH` environment variable to `C:\esp\tools-44` instead of `C:/esp/tools-44/`? I think the slash at the end is causing the problem.

BTW, [the vscode plugin](https://github.com/espressif/vscode-esp-idf-extension) has a built-in alternative, SDK Configuration editor, which is the recommended way instead of `idf.py menuconfig`.

sergioaiello, 2022-01-11:
Hi Roland,
thank you for your hint.
The IDF_PATH env var is set to "C:\esp\esp-idf-44"
Changed accordingly to your hint to "C:\esp\tools-44"
Restarted the ESP-IDF command window. I got "IDF_TOOLS_PATH, was not set", change directory to select the project path and the "idf.py menuconfig" command reports the same error. So I set IDF_TOOLS_PATH to "C:\esp\tools-44\.espressif". Restarted the ESP-IDF command window. I got "PYHTONNOUSERSITE, was not set", change directory to select the project path and the "idf.py menuconfig" command reports the same error. So I set PYHTONNOUSERSITE to "C:\esp\tools-44\.espressif\python_env\idf4.4_py3.8_env\Scripts"
Restarted the ESP-IDF command window, change directory to select the project path and the "idf.py menuconfig" command reports the same error.
The first rows for env initializing are as follows:
-----------
Using Python in C:\esp\tools-44\.espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\esp\tools-44\.espressif\tools\idf-git\2.30.1\cmd\
git version 2.30.1.windows.1
Setting IDF_PATH: C:\esp\esp-idf-44

Adding ESP-IDF tools to PATH
WARNING: tool idf-exe version 1.0.2 is installed, but has reported version 1.0.3
ERROR tools idf-exe has no installed version. Please run 'C:\esp\esp-idf-44\install.bat' to install it.
------------
so I run the following two commands
"C:\esp\esp-idf-44\install.bat" 
and
C:\esp\esp-idf-44\export.bat
change directory to select the project path and the "idf.py menuconfig" command was successful.
It is ok, but ESC key does nothing

I did not know about "ESP-IDF:  SDK Configuration editor (menuconfig)"
I'll use it.

Regarding the traditional "menuconfig" I think to have manually replaced the idf.py.exe might give instability. What else could be the error cause?

dobairoland, 2022-01-11:
Great. I'm happy that it got solved.

> It is ok, but ESC key does nothing

There is some issues with special keys (ESC, arrow keys, ...) when run from vscode. That is the GUI based SDK configuration is the preferred way for vscode.

closed 2022-01-11 by dobairoland

sergioaiello, 2022-01-11:
Running menuconfig from ESP-IDF command window (and not from VSCode) I got from "idf.py menuconfig " menu says:
[ESC] Leave menu
Therefore ESC does something.
Just to clarify.
To me is ok to use "ESP-IDF: SDK Configuration editor (menuconfig)"