espressif/esp-idf #8221: how to restart once task wdt is triggered

achao1104, 2022-01-11:
Hello,

I used blow codes before to restart device once task wdt is triggered, we found some times  'wifi' task will be always busy to trigger the task wdt, but once esp_task_wdt_isr_user_handler called esp_restart(), it will cause another panic due to ungracefully shutdown of the wifi.
void esp_task_wdt_isr_user_handler(void)
{
	esp_restart();
}
so I am told to code like below, the handler will send to release a semaphore and then in turn, in another task (named restart task) which is waiting for this semaphore, there are codes to restart device, however, due to the cpu is occupied by the 'wifi' task, I guess the 'restart' task’won't get chance to execute, thus, the device won't be restarted though the wdt handler is called for muliple times. so how to solve it?

void esp_task_wdt_isr_user_handler(void)
{
	BaseType_t xHigherPriorityTaskWoken = pdFALSE;
	xSemaphoreGiveFromISR( restart_semaphore_handle, &xHigherPriorityTaskWoken );
}

E (1753869) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (1753869) task_wdt:  - IDLE (CPU 0)
E (1753869) task_wdt: Tasks currently running:
E (1753869) task_wdt: CPU 0: wifi
E (1753869) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400C3CD6:0x3FFCDFA0 0x40027132:0x3FFCDFC0 0x400DF8ED:0x3FFE0D60 0x400DBA89:0x3FFE0D90 0x400DBF09:0x3FFE0DD0 0x400DC05F:0x3FFE0E00 0x400DC17B:0x3FFE0E30 0x4003A199:0x3FFE0E70
0x400c3cd6: adc_gpio_init at D:/project/esp32/idf-v4.3.1/components/driver/adc_common.c:279 (discriminator 4)

0x40027132: _xt_lowint1 at D:/project/esp32/idf-v4.3.1/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x400df8ed: rcUpdatePhyMode at ??:?

0x400dba89: ic_set_trc at ??:?

0x400dbf09: lmacInit at ??:?

0x400dc05f: lmac_record_txtime at ??:?

0x400dc17b: lmacRetryTxFrame at ??:?

0x4003a199: ppResortTxAMPDU at ??:?

Dazza0, 2022-01-11:
Using `menuconfig`, enable `CONFIG_ESP_TASK_WDT_PANIC` so that the watchdog timer will panic when it triggers. Depending on your panic behavior configuration (e.g., `CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT), a panic will print some debug information then do a restart. If you want to debug further, you can also trigger GDB on a panic using `CONFIG_ESP_SYSTEM_PANIC_GDBSTUB`.

Dazza0, 2022-02-14:
@achao1104 As mentioned above, enabling `CONFIG_ESP_TASK_WDT_PANIC` will reset on a task WDT timeout. Closing this issue. Please feel free to reopen in necessary.

closed 2022-02-14 by Dazza0