espressif/esp-idf #8222: occasionaly, 'wifi' task will trigger task wdt

achao1104, 2022-01-11:
hello,

idf:v4.3.1, module: esp32s2

I connected video camera to the esp32s2, occasionaly the task wdt is triggered like below, how to solve it, thanks.

E (1753869) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (1753869) task_wdt: - IDLE (CPU 0)
E (1753869) task_wdt: Tasks currently running:
E (1753869) task_wdt: CPU 0: wifi
E (1753869) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400C3CD6:0x3FFCDFA0 0x40027132:0x3FFCDFC0 0x400DF8ED:0x3FFE0D60 0x400DBA89:0x3FFE0D90 0x400DBF09:0x3FFE0DD0 0x400DC05F:0x3FFE0E00 0x400DC17B:0x3FFE0E30 0x4003A199:0x3FFE0E70
0x400c3cd6: adc_gpio_init at D:/project/esp32/idf-v4.3.1/components/driver/adc_common.c:279 (discriminator 4)

0x40027132: _xt_lowint1 at D:/project/esp32/idf-v4.3.1/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x400df8ed: rcUpdatePhyMode at ??:?

0x400dba89: ic_set_trc at ??:?

0x400dbf09: lmacInit at ??:?

0x400dc05f: lmac_record_txtime at ??:?

0x400dc17b: lmacRetryTxFrame at ??:?

0x4003a199: ppResortTxAMPDU at ??:?

AxelLin, 2022-03-01:
The backtrace is the same as https://github.com/espressif/esp-idf/issues/7936#issuecomment-1009566804

Please try the suggestion in https://github.com/espressif/esp-idf/issues/7936#issuecomment-1009655876

Alvin1Zhang, 2022-07-11:
Thanks for reporting, and thanks to @AxelLin for sharing the notes. Feel free to reopen.

closed 2022-07-11 by Alvin1Zhang