espressif/esp-idf #8223: "pppos+ip_napt_enable" not work for camera to connect

achao1104, 2022-01-11:
Hello,
My idf is 4.3.1 and module is esp32s2. I am running the pppos sample with the napt enabled. It works well, both my mobile and my PC can connect esp32s2 which actually works as wifi ap.
One third party camera can connect the esp32s2-wifi-ap as well, and even the camera can upload video to cloud smoothly.
If my mobile connects the same esp32s2-wifi-ap, my mobile can operate, open and play the live video of the camera smoothly.
The problem is that if my mobile connects LTE network or another wifi router, my mobile can not connect the camera and play live video, I guess there might be some issues related to NAT or setting of wifi ap, but don't know how to solve.
Please help!
Thanks,
Peter

david-cermak, 2022-02-01:
Hello Peter,

Could you please explain what exactly doesn't work? Maybe a simple description of your network(s)? You're talking about *another* wifi router, *esp32s2-wifi-ap*, your mobile phone, etc.

> The problem is that if my mobile connects LTE network or another wifi router, my mobile can not connect the camera and play live video

If your mobile phone disconnects from the *esp32s2-wifi-ap*, is there any other "route* for it to access the camera (esp32s2-wifi-client)? Is your *another wifi router* connected to the esp32s2 in any way?
Same question for the LTE: if your phone gets connected to the internet over LTE, is there any way for it to access the local network?

achao1104, 2022-02-07:
I guess it might be related to "how esp32 handle hole punching", please refer to https://www.geeksforgeeks.org/nat-hole-punching-in-computer-network/

david-cermak, 2022-02-07:
@achao1104 So, it means there's no route from your *other wifi router* to the esp32s2, and you're trying to access devices behind NAT, correct? I think there are some methods (as you mentioned above, and some others), that should work, but TBH I've never tested any of them.

> I guess it might be related to "how esp32 handle hole punching",

Could you please explain, how exactly you're doing that? You'd have to initiate a (tcp?) connection from your camera-device first, so the port/address gets stored into the internal NAPT descriptors.

closed 2022-03-08 by Alvin1Zhang